lumapu/ahoy #1498: Ahoy DTU sendet ständig Shutdown Befehl. Inverter alarm: Shut down by remote controll

mxgross, 2024-03-14:
### Platform

ESP8266

### Assembly

the DTU was already assembled

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

[image]
[image]
[image]
[image]

### Debug Serial Log output

```shell
Werde ich nachreichen sobald er nochmal den Wechselrichter zum Shutdown bewegt.
```

### Error description

Seit uns vorgestern eine Sicherung mal rausgeflogen war (eine Docking Station vom PC spinnt manchmal), schaltet sich mein Balkonkraftwerk andauernd aus, sobald die Ahoy-DTU auch angeschaltet ist.
Ich habe die aktuellen Einstellungen (JSON Export) mit meinen von vor  einem Monat verglichen und es gibt keinen Unterschied.

Ohne Ahoy-DTU läuft die Anlage problemlos. Ich habe keinerlei Power Control aktiv.
Meine Geokoordinaten sind korrekt, die Systemzeit ist korrekt.
Ich habe absolut keinen Schimmer woran es liegen könnte.
Mit einer anderen Firmware Version war ich zuvor über ein Jahr problemlos in Betrieb.

[image]
[image]
[image]

tictrick, 2024-03-14:
Hallo mxgross,

Hier ein paar Anmerkungen zur Fehleranalyse.

- Die Inverter waren nicht um 13:25 aus, sie wurden bereits um 13:03 abgeschaltet. Das Webserial zeigt diesen Bereich nicht.
- Ahoy versendet nicht ohne Grund Nachrichten an den Inverter. Es muss einen Grund dafür geben. Webinterface, API, MQTT.
- MQTT ist aktiv, vermutlich kam der Befehl über MQTT. Um das zu überprüfen kannst Du MQTT vorrübergehend deaktivieren.
- Vielleicht hatte die DTU rebootet und bekam über den MQTT-Broker einen retained Befehl zum ausschalten.

Hier noch eine Einladung ins Supportforum auf Discord
https://discord.gg/WzhxEY62mB

mxgross, 2024-03-14:
Schon mal Danke die zügige Rückmeldungen:)

Also das mit den Zeiten war in dem Screenshot vielleicht etwas missverständlich. Es kommt mehrmals pro Stunde vor und das Log und der Screenshot lagen etwas zeitlich versetzt da ich bei der Erfassung des Issues unterbrechen musste.

Also ich habe in der Tat einen ioBroker, der die MQTT Daten empfängt und in Influx speichert für eine Anzeige ein Grafana.
Dort habe ich aber keinerlei (zumindest nicht bewusst) Befehle konfiguriert die via MQTT gesendet werden.
Ich muss mal schauen wie ich das genauer feststellen kann und das ioBroker Log mal im Auge behalten wenn das nächste mal ein Shutdown passiert.

tictrick, 2024-03-14:
Welchen Adapter verwendest Du im ioBroker? mqtt-Broker server/client oder nur client?

mxgross, 2024-03-14:
Ich verwende für die Kommunikation mit Ahoy-DTU den Adapter "MQTT Broker/Client" Version 5.2.0
[image]

Zusätzlich habe ich noch einen Sonoff Adapter für die Kommunikation mit einem Tasmota der am Stromzähler hängt, um meinen Stromverbrauch und den verschenkten Strom zu tracken. Sonoff deshalb, da mir der Tasmota sonst nur JSON Objecte schickt die ich in Grafana hätte auseinanderdröseln müssen. Weiß mittlerweile aber auch das geht und könnte alles auf MQTT Broker umstellen, wenn das relevant ist.
Das sind alle installierten Adapter:
[image]

tictrick, 2024-03-14:
Dann musst mal Googeln wie man retained Messages aus dem Broker löscht. Eventuell einfach mal die Einträge von Ahoy in den Objekten unter mqtt löschen.

lumapu, 2024-03-14:
schau mal in ioBroker, dass unter `ctrl` alles auf `(null)` steht. Das ist ein Problem gibt es mWn. nur mit ioBroker bisher.

[image]

mxgross, 2024-03-15:
Alles unter CTRL ist null
[image]

Ich habe mal im MQTT Broker "Eigene States beim Verbinden publizieren" und "States bei subscribe publizieren" **deaktiviert**.
[image]
[image]

Bisher läuft noch alles, ich werde es mal weiter beobachten.

Vielen Dank, dass ihr euch hier die Zeit für mich genommen habt und mich auf die richtige Spur gelenkt habt! :)

closed 2024-03-15 by mxgross