lumapu/ahoy #1499: No connect to inverter after update to 0.893

Hanz2420, 2024-03-14:
### Platform

ESP8266

### Assembly

the DTU was already assembled

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.93

### Github Hash

0c39f1d 

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Keines , nach dem Fehler "Inverter  #0: ... ist gerade nicht verfügbar" habe ich versucht den Power Level anzupassen(Min, Low, High), aber keine Änderung. Beim Standard IRQ=D4(GPIO2) kommt immer Fehler "Interupt Pin funktioniert = unbekannt / gelb". Wenn ich auf IRQ= D3(GPIO0) ändere, wird der Fehler dann wenigstens nicht mehr angezeigt: Interupt Pin funktioniert = JA - grün. Aber trotzdem kommt kein Connect zum Inverter Hoymiles HM-600 mehr zustande

### Error description

Nach Update: Wechselrichter #0: .... ist gerade nicht verfügbar
Alle Anzeigen bleiben auf null, im Web und auf der Anzeige des DTU
Auch nachdem ich den Inverter gelöscht und wieder neu eingetragen habe, habe auch ein Reset auf Factory Def. durchgeführt und alles neu eingerichtet - aber kein Änderung.. es gibt keinen Kontakt zum Inverter... Vor dem Update lief es seit Mai 23 (mit der damaligen FW stabil).... hätte ich mal nicht ein Update eingespielt..

Kann jemand helfen ? habe ja bereits andere Probleme hier gelesen, allerdings bringt aktuell kein Downgrade der FW (230817_ahoy_0.7.36_ba218ed_esp8266 probiert) - Fehler bleibt bestehen:
[image]

[image]

[image]
[image]

Hanz2420, 2024-03-14:
Danke dir, allerdings...
IRQ musste ich auf 3 setzten, sonst gibt es zusätzlich noch den Fehler (wie beschrieben:  

Interrupt Pin funktioniert	unbekannt - gelb ...

[image]

wenn ich es so setze:
[image]

Oder hast du ein anderes Setting gemeint ?

Hanz2420, 2024-03-14:
Ja hatte Version vom Mai 23 - kann dir nicht sagen welche. Kam mit der Box aus der Bucht...
Habe es seit gestern versucht, auch immer laufen gelassen.. blieb den Tag über auch immer so..
Also Umpinnen  in den Settings auf D4 wie unter 4.  So beschrieben ? - ist ja der Standard.. der auch so nach dem Update eingestellt war -aber halt nix mehr lief..
[image]

Ich lass es laufen und gucke morgen für nochmals..
Ich berichte dann..
thx und schönen Abend...

lumapu, 2024-03-14:
ich denke hier ist alles gesagt und kann das geschreibene nur bestätigen. Vielen Dank an alle Beteiligten. @Hanz2420 kannst du bitte morgen nochmal Feedback geben - hoffentlich mit Erfolg.

lumapu, 2024-03-14:
im Zweifel sollte die erste Anlaufstelle unsere Dokumentation sein:
https://docs.ahoydtu.de/de/latest/faq.html#mein-wechselrichter-wird-nicht-erkannt

Hanz2420, 2024-03-14:
So dann ?

[image]

kiu77, 2024-03-14:
Nein, genau nicht. Andersrum. CE = D3 und IRQ = D4.
Siehe: [
[image]
](url)

Hanz2420, 2024-03-15:
Es hat leider nicht funktioniert. Ich habe die DTU komplett gekauft - also nix mit Löten..
[image]

[image]

zusätzlich bleibt die Anzeige jetzt auch noch schwarz (LCD am DTU Kästchen),nachdem ich es eimal vom USB-Kabel (Strom) getrennt habe. Hatte ich ach gestern schon - nur finde ich in dieser FW jetzt nicht mehr das Setting für das Display...

Please Help !

Hanz2420, 2024-03-15:
Habe auf [GIT SHA: 7b030a3 :: 0.8.91] zurück "geladen", da gibt es dann eine Display Einstellung die im Standard dann auf "keins" steht. Hab das erste Display ausgewählt, dann zeigt es wieder etwas en. Allerdings ist das Hauptproblem:
[image]

weiterhin vorhanden..

Was mache ich denn falsch ? Hab nur vorgestern die alte FW (Stand Mai 23) upgedatet und dann war's geschehen...

thx

Dirk-Ahoy, 2024-03-15:
Entweder an den Verkäufer wenden oder die Kiste aufmachen und schauen wie die Belegung ist.

Hanz2420, 2024-03-15:
So habe das Problem wohl gelöst. Mit den Pins und Belegung funktioniert es nur so:
[image]

[image]

Aber das wichtigste, die DTU benötigt jetzt ständig (jedenfalls zum ersten Connect) eine Internetverbindung, nur die Systemzeit via Browser senden genügt nicht mehr. Was habt ihr da verändert ? Warum "schwätzt" die DTU jetzt immer mit dem Web ? Hatte es extra via Pihole und Router Setting nur im LAN belassen ... Bitte mal darstellen warum jetzt ein Internet Zugang nötig ist ?

Thx
Hanz

lumapu, 2024-03-16:
der Fehler mit dem Internetzugang ist bekannt und wird als nächstes angegangen. Schaue auch in den anderen issues, Stichwort ist hier `NTP`

Hanz2420, 2024-03-18:
danke für die Info -  werde ja dann sehen wenn es neue Updates gibt. Danke euch auch für die Tipps - sehr guter Service !

Hanz

closed 2024-03-18 by Hanz2420