lumapu/ahoy #1500: Wo ist bei 0.8.93 dieses Leistungsdiagramm geblieben? (Tagesverlauf)

firedp, 2024-03-15:
Wo ist bei 0.8.93 dieses Leistungsdiagramm geblieben? (Tagesverlauf)

knickohr, 2024-03-15:
Changelogs lesen, wenn man schon immer an der neuesten DEV klebt 😉

https://github.com/lumapu/ahoy/blob/development03/src/CHANGES.md#development-changes

Unter 0.8.92 findet man die Antwort.

cibis8, 2024-03-16:
ok das mit der History, hab ich gecheckt (ist für mich auch nicht so relevant).  Doch wo ist denn die Display Config abgeblieben? Ab Version 92 finde ich diese in keinem der DEVs mehr. Auch kommt beim Updaten neuerdings der Hinweis: Warning your environment may not match the update file! Update geht zwar dann trotzdem weiter (beim Drücken von Continue, aber ich finde bei keinem der Update Files die Display Einstellungen :(

firedp, 2024-03-16:
In 94 ist doch wieder alles da?
Habe jetzt mutig auf 8266 all de 0.8.94
Da kam zwar die Meldung das mein Chip nicht geeignet wäre, aber all heißt all :-)
Alles läuft super und viel schönere Diagramme drin. TOP

cibis8, 2024-03-16:
@firedp Ich habe nochmals auf die 0.8.94 aktualisiert.. bei dne -all Versionen (englische sowie die deutsche) enthalten bei mir weder die eine noch die andere eine Display Config!  Bin nun wieder zurück zur 0.8.89. Auch was dieser Hinweis soll, das meine Umgebung nicht mit dem Update zusammen passen solle habe ich erst ab der Version .92 erhalten.. Also irgendwas stimmt hier nicht, hatte noch nie Probleme mit Updates! .. naja ausser das mal irgendeins weniger stabil lief (kein Zugriff via IP Adresse) und dann durch ein Neustart behoben wurde. Solche Eigenheiten waren bisher nie.

lumapu, 2024-03-16:
@cibis8 dein Problem heißt "Broswer-Cache". Da du versch. Versionen der gleichen Firmwareversion ausprobierst aktualisiert sich noch nicht der Browser-Cache. 
Entweder du löschst ihn manuell oder du lädst z.B. die Settings Seite mit `STRG+F5` (das umgeht den Cache).

cibis8, 2024-03-16:
@lumapu Danke für den Hinweis... hmm hätte ich auch drauf kommen können :) nun geht's bzw. ist die Display Config wieder da. Morgen kann ich dann auch noch berichten, ob das Display was anzeigt - das war mit Versionen ab der .90 leider nicht immer der Fall. Seltsamerweise hatte ich dieses Problem mit der Browser Cache und dem Refresh bisher nie gehabt bei Vorgänger Versionen.. und ich teste relativ viele DEVs..

lumapu, 2024-03-16:
wenn du genau schaust, dann hängt oben an der URL auch die Version mit dran. Wahrscheinlich hast du bisher immer verschiedene Verisonen getestet - und jetzt das erste Mal verschiedene Build-Varianten einer Version.

cibis8, 2024-03-16:
das kann möglich sein.. und was hat es mit diesem "Warning" beim Updaten zu tun?

lumapu, 2024-03-16:
leider haben in Vergangenheit versch. Leute versucht z.B. eine ESP32 Firmware auf einem ESP8266 zu installieren. Diese Meldung soll einfach dazu da sein, dass man sich bewusst ist was man gerade macht.

Hanz2420, 2024-03-18:
Moinsen, die Grafik (Tagesverlauf - also hab nur eine gefunden) werden doch nur in der HTML-Ansicht im Browser angezeigt oder kommen die doch irgendwie auf das kleine Display auf der DTU ? 
habt ihr da mal Beispiele ?
Ich finde nur dieses:
[image]

thx 
Hanz

lille2000, 2024-03-18:
Der Tagesverkauf ist auch an Display.
Dazu muss man das „Verhältnis“ auf zb 50% stellen.

hier ist ein älteres Bild:
[image]
[image]

Zordandi, 2024-03-19:
> Der Tagesverlauf ist auch an Display. Dazu muss man das „Verhältnis“ auf zb 50% stellen.

Was auch etwas versteckt ist: Und man muss den Standort unter Sonnenaufgang -untergang einstellen. Sonst kann das Diagramm Start/Ende nicht berechnen.

tobi119, 2024-03-20:
> Was auch etwas versteckt ist:

Wo kann ich es finden? Was muss ich sonst einstellen, um den Graf auf dem Display zu erhalten?
Ist der Tagesverlauf auch auf dem ePaper Display möglich?

closed 2024-08-16 by lumapu