lumapu/ahoy #1501: Question about transmission power HM1500, Reception is bad  / Frage zur Sendeleistung  HM1500 , Empfang ist schlecht

ralfberger, 2024-03-15:
## Hardware
  * [x ] ESP8266
 
### nRF24L01+ Module
* [ x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode

### Antenna:
* [x ] external antenna (SMA)

### Version / Git SHA: 
Version: 08.83

Kann mir jemand sagen ob die Sendeleistung der Hoymiles HM-1500 Wechselrichter sehr schwach ist ?
Meine Wechselrichter sind am Carport das ca. 8 Meter entfernt steht montiert. Ich habe frei Sicht. Wenn ich den ESP mit NRF24L01+ (mit externer Antenne) im Haus direkt hinterm Fenster aufstelle ist der Empfang sehr schlecht. Mal funktioniert aber sehr oft lese ich nur Wechselrichter ist gerade nicht verfügbar. 
Oder kann ich am Setting etwas ändern?
Unter Wechselrichter / Funkmodul / Sendeleistung habe ich  LOW, MIN und HIGH getestet.  Ich meine mit HIGH ist der Empfang am besten. 
Grüße

** or in English **  
Can anyone tell me if the transmission power of the Hoymiles HM-1500 inverters is very low?
My inverters are mounted on the carport which is about 8 meters away. I have a clear view. When I place the ESP with NRF24L01+ (with external antenna) in the house directly behind the window, the reception is very poor. Sometimes it works but very often I only read Inverter is not available at the moment. 
Or can I change something in the setting?
I have tested LOW, MIN and HIGH under Inverter / Funkmodule / Sendeleistun. 
 I think the reception is best with HIGH.

Dirk-Ahoy, 2024-03-15:
Du kannst nur die Sendeleistung beeinflusse von der DTU - welches Modul hast du?  Ev mal in ein Ebyte Modul tauschen wenn du so ein günstiges hast.  Ich lese einen HM1500 im Garten aus ca 80m Luftlinie  8m ist kein problem.

ralfberger, 2024-03-16:
O.K.  wenn 80 Meter geht sollte es bei mir auch funktionieren.
Ich habe mir alternativ schon ein Funkmodul mit integrierter Antenne bestellt 
Dann werde ich mal den Lötstation anheizen und es testen.
Danke!

ralfberger, 2024-03-16:
Es war das Funkmodul. Ich habe jetzt ein anderes angelötet. Der Empfang ist aktuell stabil. Ich hoffe es bleibt so.

closed 2024-03-16 by ralfberger