lumapu/ahoy #1502: MQTT warning can't decode

Zordandi, 2024-03-16:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

square dot

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a

### Build & Flash Method

was already installed

### Setup

- MqTT added

### Error description

Die Payload im Homeassistant wirft bei mir ein Warning und ich kann AhoyDTU daher auch nicht einbinden (das Warning kommt für alle Werte):
2024-03-16 12:02:25.765 WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'{"name":"CH1_U_DC\\u0000@\xff?|\\f\xff?\xa8I\xff?\xd4I\xff?\xd4I\xff?$a#@","stat_t":"AhoyDTU/LahmeGarage/ch1/U_DC","unit_of_meas":"V"... (ab hier scheint der Rest zu stimmen)

rmayergfx, 2024-03-16:
@Zordandi 
Nimm bitte mal die aktuelle dev 0.8.94 und probiere es damit. https://github.com/lumapu/ahoy/actions

Zordandi, 2024-03-16:
> @Zordandi Nimm bitte mal die aktuelle dev 0.8.94 und probiere es damit. https://github.com/lumapu/ahoy/actions

Oh man, ich hätte vorher im Dev nachschauen sollen. Ich hab die 0.8.84 ausprobiert und kann bestätigen, dass der Fehler nicht mehr auftritt.
Vielen Dank!!!

closed 2024-03-16 by Zordandi