lumapu/ahoy #1503: [Bug] Limit lässt sich mit REST API nur für einen Wechselrichter setzen

RALFISBORN, 2024-03-16:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

- zweiter HM 600 wurde hinzugefügt
- Ahoy wurde auf aktuelle Version 0.8.83 geupdated
- Die Limitierung erfolgt über Node-Red mittels http-request

### Error description

Ich hatte zuvor nur einen HM 600 mit Ahoy verbunden und es war eine Version installiert, welche noch keine Authentifizierung über die REST API benötigt hat.

Jetzt habe ich einen zweiten HM 600 mit Ahoy verbunden. Die Werte werden richtig angezeigt und das Limit kann auch über die Weboberfläche von Ahoy an beiden HMs separat eingestellt werden.

Wenn ich jedoch das Limit via REST API verändern möchte, kommt beim zweiten HM (also der neu hinzugefügte) immer als Rückmeldung "ERR_LIMIT_NOT_ACCEPT". => Zu diesem Zeitpunkt war der bestehende HM offline.

Zunächst dachte ich, dass es evtl. am neuen HM liegt, welcher eine neuere HW Version hat. Ansonsten sind Firmware Version und Bootloader Version exakt gleich.
Deshalb habe ich dann den Test nochmal mit dem zuerst installierten HM probiert, als dieser Online war. Dabei trat folgender Fehler auf:

Wenn ich den Befehl:
{
    "token": "{{payload.token}}",
    "id": "Seriennummer HM Bestand",
    "cmd": "limit_nonpersistent_relative",
    "val": 95
}

oder 

{
    "token": "{{payload.token}}",
    "id": "Seriennummer HM neu",
    "cmd": "limit_nonpersistent_relative",
    "val": 95
}

ausführe, jedes mal das Limit am bestehenden HM durchgeführt wird.

Sobald der bestehende jedoch offline ist, bekomme ich konstant die Fehlermeldung ERR_LIMIT_NOT_ACCEPT. Egal, welche Seriennummer in der id drin steht.

Habe ich hier etwas falsch gemacht und handelt es sich hierbei um einen Bug?

Vielen Dank

lumapu, 2024-03-16:
als ID verwendest du aber nicht die 12-stellige Seriennumer? Für dieses Kommando wird die ID von `0` kommend hochgezähtl, in deinem Fall also `0` und `1`

RALFISBORN, 2024-03-17:
> als ID verwendest du aber nicht die 12-stellige Seriennumer? Für dieses Kommando wird die ID von `0` kommend hochgezähtl, in deinem Fall also `0` und `1`

Ok, das war der Fehler von mir :(. Vielen Dank für den Hinweis :). Mir ging aus der Setup Page + Manual nicht ganz hervor, dass die beiden eine eindeutige ID haben und diese nicht der Seriennummer entspricht. Vielleicht könnte man ja auf der Settingsseite noch eine Spalte hinzufügen das besser hervorgeht, welcher Wechselrichter welche ID hat.

Trotzdem Danke für den Tipp :). Jetzt klappts!

RALFISBORN, 2024-03-17:
> @RALFISBORN danke für deine post snippets, macht alles sofort klar. also bitte 0 oder 1 für die id verwenden.
> 
> ps: no prob, im opendtu tracker hatten wir auch vor kurzem einen esperanto api mix eines users. passiert evtl. wenn man nur sporadisch mit verschiedenen dtus arbeitet, oder nebenbei über verschiedene dtus liest.
> 
> hihihi, analog zu spanglish: sollen wir's openahoylisch oder ahoyopenlisch nennen 😉
> 
> greetings metachuh

Das beruhigt mich ja, das ich nicht der einzige war/bin, dem das passiert ist :D

Ich würde es ahoyopenlisch nennen :D

closed 2024-03-17 by RALFISBORN