lumapu/ahoy #1504: [Bug]Jump in reported solar yield (total kWh) with new firmware

boboG, 2024-03-16:
### Platform

ESP32

### Assembly

the DTU was already assembled

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.94

### Github Hash

6af07f5

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Hoymiles HM-600 with two Trina Solar 395 Wp - TSM-395 panels.

### Error description

Changed from 0.8.83 (stable) to 0.8.94 (dev) today because of the MQTT bug in firmware 0.8.83 which prevented HA from discovering the DTU. There was then a sudden jump in total yield from 713 kWh to 740 kWh (and respectively on the two channels). There is no such jump in the daily yields. It looks like the total yields have been recalculated resulting in a higher value.

lumapu, 2024-03-16:
There was a effiency factor for older versions. The default effiency was 0.955%. For you values it would calculate as:
740kWh * 0.955 = 706.7kWh

Check also issue #1243

boboG, 2024-03-16:
Ah, I see. So the factor was removed. Thanks.

closed 2024-08-16 by lumapu