lumapu/ahoy #1505: ESP8266 Mqtt not in minimal version?

Elarion245, 2024-03-17:
## Hardware
  * [x l] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x] external antenna (SMA)

### Power Stabilization:
* [x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 08.95
Github Hash: _______

### Build & Flash Method:
* [x] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

I switched to the minimal version of ESP8266, as the normal one only lasts about 4 days prior to a reboot.
as I am not using a display I wanted to switch to the minimal version (0.7.36 managed 15 days without reboot and more).
My issue with the minimal version is, that there is no mqtt connection possible. Both ahoy and HomeAssistant report mqtt connected = false.
Using the non minimal version it connects immediately. The behavior is the same in 94 and 95.

Based on the change logs I would have expected mqtt to be working in the minimal version?

Elarion245, 2024-03-17:
Thanks a lot! This one is working fine. So takeaway is to use the version without any suffix in my case.

Peertje-62, 2024-03-17:
8266.bin
only MqTT
this version is the prefered 8266 version, most stability

8266-minimal.bin
nothing except basic functions
use this version for debugging purposes

8266-all.bin
MqTT
Display
History
everything included without looking to stability

8266-prometheus.bin
MqTT
Display
History
Prometheus
everything + Prometheus

https://github.com/lumapu/ahoy/issues/1495#issuecomment-1987343702

regards

closed 2024-03-18 by Elarion245