lumapu/ahoy #1507: WebSerial, buffer overflow!

Otto-Normal, 2024-03-18:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.93

### Github Hash

0c39f1d

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Inverterabfrage mit 15 s Intervall
MQTT Daten werden auch mit 0.8.94 oder 0.8.95 gesendet 

### Error description

Nach Update auf  0.8.94, oder auch auf 0.8.95 zeigt die Konsole permanent nur noch „webSerial, buffer overflow!“. Sonst nichts.
Nach Downgrade auf 0.8.93 werden alle Daten wieder ausgegeben.

lumapu, 2024-03-18:
wie stehen die Einstellungen der seriellen Ausgabe?
Das wäre mein Vorschlag:

[image]

Von .94 zu .95 wurde an der Ecke nichts geändert.

lille2000, 2024-03-18:
Bei meinem ESP8266 (all-de, 2 Inverter, Display, keine Nutzung von MQTT) bekomme ich mit 0.8.94 und 0.8.95 auch kaum mehr Daten bzw. die Webseite wird instabil. Langsam ist der 8266 wohl am Ende

0.8.93 ist stabil und gut.

Otto-Normal, 2024-03-19:
Habe Livedaten nun deaktiviert. Daten im WebSerial werden nun angezeigt. Kein buffer overflow mehr.
Ehrlich gesagt ist mir nicht klar was diese Livedaten genau sind. Dachte es sind die aktuellen Werte der PV-Module.
Übrigens, kleiner Tipp, geht es sprachlich nicht generell besser um Werte? Im Auto sind es ja auch die Werte vom Drehzahlzahlmesser…😉 

Danke!

Otto-Normal, 2024-03-19:
Ups! Das war wohl nur temporär. 
Wenn  der Haken gesetzt, kommt nur webSerial buffer overflow! und einige Radio Infos. Keine Livedaten!

lumapu, 2024-03-21:
ja Werte wären uU. hier tatsächlich besser. Ist echt schwierig in einer Welt wo Deutsch und Englisch immer weiter verschwimmen.