lumapu/ahoy #1509: Frage: Bei zwei Wechselrichter, zeigt das Display die aktuelle Leistung von jeden Wechselrichter an?

ralfberger, 2024-03-18:
## Hardware
  * [x ] ESP8266

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x ] circuit board
* [ ] external antenna (SMA)

### Version / Git SHA: 
Version: 0.8._83
Github Hash: _______

Bevor ich mir ein Display kaufe und anschließe, kann mir jemand sagen, welche Werte das Display anzeigt. Ich habe mir diverse Youtube  Videos angesehen aber leider finde ich dort auch keine Antwort.  
Auf Bildern und im Video  sehe ich, dass die aktuelle Leistung und auch die Gesamt Leistung angezeigt wird. 
Ich habe in der DTU zwei Wechselrichter eingerichtet. Da wäre es schön wenn ich die  aktuelle Leistung von jeden Wechselrichter ablesen könnte.

lille2000, 2024-03-18:
Es wird jeweils die Summe im Display angezeigt (Leistung , die Energiemenge des Tages und die Gesamtenergie)

Peertje-62, 2024-03-18:
Hi, 
https://discord.com/channels/984173303147155506/1065266691463974972/1179850228464107541
regards
[image]

Peertje-62, 2024-03-18:
Hi,
I would also suggest you use an esp 32 instead of an 8266  as the 8266 is reaching its limits with the current software,
even better and more stable;
https://3printd-solution.eu/Ahoy-openDTU-Fusion-fuer-HM-HMS-HMT-mit-Display
bye

closed 2024-03-19 by ralfberger

ralfberger, 2024-03-19:
Ja, die Darstellung des Graphen sieht toll aus. :+1: