lumapu/ahoy #1510: [Feature Request] Send WiFi-Channel thru MQTT

DB1BMN-collection, 2024-03-18:
## Hardware
  * [x] ESP8266

### Version / Git SHA: 
Version: 0.8.92
Github Hash: 96eb787

Since I got this fancy error message this morning i suggest sending WiFi channel via MQTT for Debug purposes.
[image]

BTW, hat to restart my Speedport router because noting worked then: No AHOY, no TASMOTA, no Home Assistant, no DNS, ... Don't know the reason

Gubi2023, 2024-03-18:
see  https://github.com/lumapu/ahoy/issues/1381#issuecomment-1915552946

stefan123t, 2024-10-30:
@lumapu prinzipiell eine tolle Sache die WiFi Channels per MQTT mitzuloggen. Ich wäre noch für die letzte WiFi Disconnect Reason.

In der OpenDTU haben wir für die Analyse der WiFi Disconnect's (evtl. im Rahmen von 802.11 k / v/r roaming und mesh steering) auch die Disconnect Reason mit in der Serial Console geloggt. Aber MQTT geht ja dann grade nicht, wenn das Netz weg ist :shrug: 

Keine Ahnung ob das mit dem WiFi Channel geht, wenn keine Verbindung da ist, weil Channel 12 / 14 genutzt wird, dann kann er vielleicht auch den WiFi Channel nicht mehr per MQTT verschicken.