lumapu/ahoy #1511: Doku: Zeilen teilweise abgeschnitten[Bug]

Gubi2023, 2024-03-19:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.93

### Github Hash

????

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Doku

### Debug Serial Log output

```shell
Screenshot
```

### Error description

in der Doku sind teilweise die Zeilen abgeschnitten. Siehe Anlage
[image]

mroenne2022, 2024-03-19:
Durch "wischen" nach links kann man den Text rüberschieben und die einzelnen Zeilen bis zum Ende lesen  -  zumindest am Smartphone ist es so möglich.

Gubi2023, 2024-03-19:
ja, aber nur am Smartphone.... Am PC geht das nicht

closed 2024-03-19 by Gubi2023

mroenne2022, 2024-03-19:
Der Scrollbalken erscheint wenn man rüberfährt mit der Maus, ist eigentlich ok. Wenn er sichtbar wäre dann wäre es perfekt. Ich kann nicht beurteilen wieviel Arbeit man reinstecken muss um ihn dauerhaft sichtbar zu machen, ich denke wir können damit leben :-)