lumapu/ahoy #1513: display 2.42 pouce

jacques3629, 2024-03-20:
## Hardware
  
  * [ ok ] ESP32
  
Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Good morning

Ahoydtu Works wonderfully  HM+HMS

Sorry if this question has already been asked
not being very young, I would simply like to display the power in Walt on the 2.42 inch screen au maxi (exit the rest of the information).
Is this possible

THANKS

kiu77, 2024-03-20:
Not with this software...

Peertje-62, 2024-03-21:
Hallo,
You could try Ahoy 0.8.36 or 0.8.34,
has other display information
Back up the settings before you change the firmware
good luck 
regards

jacques3629, 2024-03-21:
> Hallo & I am 62, You could try Ahoy 0.8.36 or 0.8.34, https://github.com/lumapu/ahoy/releases has other display information : less Back up or write down the settings before you change the firmware good luck regards

thanks for the information
I haven't found anything that allows me to delete information
I just want to keep the WATT
to be able to view further

Sincerely

[image]

[image]

[image]

lumapu, 2024-03-21:
What do you mean by "pouce"? 
The display contents are fixed, so the simple answer is: no
As @kiu77 mentioned your only change to do so will be compile your own firmware which simply modifies what is displayed.

jacques3629, 2024-03-22:
Good morning

@MetaChuh
thank you for my little wooden box, it is not expensive at all, made from bamboo on AliExpress, lots of choice
bedienungs und sehhilfen option is a good suggestion for the evolution of the software
maybe in the future ahyo-dtu will support larger oled displays

@lumapu: create my own firmware, I don't know how to do it, I don't have your talent :)

jacques3629, 2024-03-22:
Sorry bad handling, I don't want to close

jacques3629, 2024-03-23:
@ MétaChuh, lumapu no worries, continue to do the excellent work
I recently discovered home automation
I had a lot of fun doing ahyo-dtu, I'm not at all in the programming world, that doesn't stop me from trying to follow your progress
and yet I got there thanks to you, also to the forum

Sincerely

closed 2024-08-16 by lumapu