lumapu/ahoy #1514: CMT2300A: Muss da IRQ-Pin funktionieren?

kiu77, 2024-03-20:
## Hardware
ESP32

### RF Module
SMT2300A

### Antenna:
external antenna (SMA)

### Power Stabilization:
100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
3.3V Voltage stabilizing via extra voltage regulator

### Version / Git SHA: 
Version: 0.8.95

Frage:
Ich habe mir testweise eine DTU mit ESP32 und dem Funkmodul CMT2300A gebaut, da ich einen Hoymiles HMS-800-2T ersten will (der Inverter ist noch nicht da). Nach dem Flashen der Firmware habe ich den NRF24 in den Settings ab- und den CMT2300A eingeschaltet. Unter "System" sieht man aber, dass der Interruopt-Pin nicht funktioniert. Muss er das?

[image]

knickohr, 2024-03-20:
Ja, er muß.

kiu77, 2024-03-20:
Und welches ist der Interrupt-Pin?
Ich habe zwei Platinen mit 30- und 38-poligem ESP32 gemacht und muss dann ja an beiden diesen Pin falsch angeschlossen haben. Mich irritiert dass trotzdem hinter SMT2300A "verbunden" in grün steht.

tictrick, 2024-03-21:
Poste mal Deinen Schaltplan, dann kann man sagen welcher Pin es ist.

Wenn Du schon dabei bist, dann noch ein Foto von den Pin-Einstellungen. Dann kann man das gleich kontrollieren.

kiu77, 2024-03-21:
Hier der Schaltplan und die Einstellungen (sind auf default):
[image]

[image]

lumapu, 2024-03-21:
SCLK und MISO sehen falsch aus.

blueline13, 2024-03-21:
Hier meine Verkabelung. Wenn du die keine Platine hast und das gelötet ist, dann versuch mal die beiden Pins 12 & 14 zu drehen (Die Kabel als auch die Settings). Sieht bei mir so aus und Vielleicht meint das auch @lumapu. Da Bezeichnungen(MISO & SCLK) die @lumapu meint siehts du unter Settings im ESP.  Ob auch der CSB geändert werden muss glaube ich erst mal nicht. ABer die Konfig läuft hier bei 2 DTUs ohne Probleme.

[image]

kiu77, 2024-03-21:
@blueline13 
wenn ich sowohl die Settings für die ESP32-Pins 12 und 14 als auch die physikalischen Anschlüsse vertausche, dann bin ich ja wieder bei den Defaults. Ist wie eine doppelte Negation.
Und meine Schaltung bzw. die Verbindungen unterscheidet sich ja in nichts von den bei "getting started" vorgeschlagenen.
Vertausche ich nur die Settings, ist das CMT2300A-Modul nicht mehr verbunden, was ja nicht gut ist.
Und bei Dir steht dann bei "Interrupt-Pin verbunden" ein "ja"?

blueline13, 2024-03-21:
@kiu77 
Ja bei mir ist alles grün.  Habe aber eben schnell den Lötkolben angeworfen und mal deine Konfig probiert und die geht bei mir auch. Wenn du eine Platine hast, dann schau mal ob sich da irgendwo ein Lötbrücke eingeschlichen hat.

kiu77, 2024-03-21:
@blueline13 
Interesting. Und das dann gleich auf zwei unterschiedlichen Platinen!
Aber ich checke das natürlich nochmal genau...

blueline13, 2024-03-21:
@kiu77 Habe mal das Kabel vom Interrupt Pin abgelötet und bekomme dann genau dein Fehlerbild. Auch bekomme ich das Fehlerbild wenn der GPIO Pin vom ESP auf Ground liegt. Vielleicht hilft dir das.

kiu77, 2024-03-21:
@blueline13 
Danke.
Welcher Pin des CMT2300A ist denn bei Dir der Interrupt?
Oder hast Du zusätzlich auch den NRF24 bestückt?

blueline13, 2024-03-21:
@kiu77 
Der Interrupt Pin ist wie bei dir auch D34 am ESP.  Ein NRF habe ich auch mit dran hängen.

blueline13, 2024-03-21:
@kiu77 
Ich habe nur Kabel gelötet. Mir kommt da gerade eine Idee. Wenn elektor sich an der OpenDTU orientiert dann ist ist vermutlich der GPIO3 Pin vom Funkmodul nicht mit dem ESP Verbunden ist. Im schlimmsten Fall habe die den GPIO3 PIN vom Funkmodul auf Ground gelegt. Du musst mal schauen was die mit dem GPIO3  Pin gemacht haben. Hast du noch eine unbestückte Platine und kannst sehen ob der PIN überhaupt angeschlossen ist. Sonst klingel mal durch wohin der GPIO3 geht. Da gab es hier schon mal den Fall das der GPIO3 auf einer Platine nicht angeschlossen war. Bei Elektor konnte ich nichts zu der Version finden, nur eine Platine mit einem NRF Modul.

blueline13, 2024-03-21:
@kiu77
Jetzt bin ich mir nicht sicher ob ich das aus deiner Worten richtig lesen. Geht der GPIO3 vom Funkmodul des CMT2300 auf den GPIO16 vom ESP, wenn ja dann bitte in den Settings vom ESP bei GPIO3 den GPIO16 auswählen. Wenn nicht, dann bitte den Schlatplan hier einmal einstellen.

kiu77, 2024-03-21:
@blueline13 
GPIO3/CMT2300A geht an GPIO34/ESP32 - genau wie in meinem Schaltplan und dem von www.ahoydtu.de sowie den screenshots der Settings. Müsste also alles richtig sein.

blueline13, 2024-03-21:
@kiu77 
Ok jetzt gehen mir auch die Ideen aus. Falls du noch ein Funkmodul und einen anderen ESP übrig hast würde ich mal mit Kabeln löten, nicht das du durch Zufall eine schlechte Charge erwischt hast. Vielleicht haben die Entwickler hier noch eine Idee warum der Interrupt Pin im Webinterface einfach nicht grün werden will.

lumapu, 2024-03-21:
irgenwie kommt mir gerade der Schaltplan komsich vor. 
Es müsste doch `GPIO13 (MOSI)` mit `SDIO` vom CMT verbunden werden, genauso wie `GPIO14 (SCLK)` mit dem `SCLK` vom CMT verbunden werden muss. Die beiden CMT-Pins `CSB` und `FCSB` können beliebige Pins am ESP sein, die sind also ok. Der Pin `GPIO3` (CMT) an `GPIO34` ist auch iO.

kiu77, 2024-03-22:
@lumapu 
Hmm, weder in meinem Schaltplan noch im "Offiziellen" von [
[image]
](url) oder den Default-Einstellungen ist GPIO13 des ESP32 beschaltet. Stattdessen ist in beiden Schaltplänen und den Default-Einstellungen SDIO (CMT) mit GPIO14(ESP) verbunden. Immerhin zeigt die Webseite meiner DTU ja an, dass das CMT-Modul verbunden ist.
Sowohl in beiden Schaltungen als auch in den Default-Einstellungen der FW 0.8.95 ist GPIO12 (ESP) mit SCLK(CMT) verbunden.
Daher wird das CMT-Modul ja als verbunden angezeigt.
Zur Vereinfachung meiner Schaltung habe ich mal nachfolgend den Display-Anschluss und das Modul NRF24 entfernt. So sieht man klar, dass meine Schaltung und die von der Ahoydtu-Webseite identisch sind:

[image]

lumapu, 2024-03-23:
sehr schön! Wie weit ist dein Wechselrichter ca. weg?

kiu77, 2024-03-23:
@lumapu 
Entfernung etwa 5m durch eine 32cm Tonziegelwand, also nichts besonderes. Der Pegel ist aber nichtsdestotrotz echt hoch!

lumapu, 2024-03-23:
die HMS / HMT machen seit Anfang an Spaß, da deren Kommunikation schon immer sehr stabil ist. Die HM haben wir aber jetzt auch sehr gut im Griff, vergleichbar gut.
Ich bin ähnlich weit weg (Dachaufbau dazwischen) und habe einmal -46dB und einmal -52dB.

kiu77, 2024-03-23:
Übrigens hat eine zweite Platine mit 30-Pin ESP32 und CMT2300A richtig Streß gemacht. Sie wollte keine Verbindung herstellen. Nach ein paar Stunden war sie dann doch ein paar mal in Kontakt mit dem Inverter, hier aber mit einem Empfangspegel von -79dB. Da wurden die Bits vermutlich mit dem Wind verbreitet ;-)
Gerade habe ich die Platine untersucht und es ist ein Kurzschluss (0,7Ω) zwischen Masse und Antennenausgang des CMT2300A. Mal schauen, ob ich das Modul ablösen muss.

kiu77, 2024-03-23:
@lumapu 
bei mir schwanken die Pegel (wohl je nach Bewegung von mir in der Nähe) zwischen -33 und -37dB. Echt gut.
Ich vermute, damit ist eine Reichweite von zig Metern ohne Probleme möglich.

lumapu, 2024-03-23:
> Gerade habe ich die Platine untersucht und es ist ein Kurzschluss (0,7Ω) zwischen Masse und Antennenausgang des CMT2300A. Mal schauen, ob ich das Modul ablöten muss.

das wird ein Spaß 😉, möchte ich nicht machen.

> Ich vermute, damit ist eine Reichweite von zig Metern ohne Probleme möglich.

ja @DirkAhoy hat eine Funkverbindung von 80m

kiu77, 2024-03-23:
Sodele, auch das ist gefixt.
1. Alle CMT2300A haben wohl am Antennenausgang parallel eine Spule gegen Masse, daher die 0,7Ω. Es war also kein Fehler.
2. War der Innenleiter bei der Antennenbuchse etwas zu weit rausgedreht, daher praktisch keine Antennenwirkung (siehe Bild). So simpel kann es sein.
3. Verhielt sich die digitale Elektronik wieder mal sehr analog: Ers nach einem Update (= neues flashen) auf 0.8.67 hatte auch diese Platine sofort Verbindung mit dem Inverter. Pegel -41 bis -46dB. Diese Pegelmessungen sind wohl eher eine grobe Schätzung, aber trotzdem ganz hilfreich.

[image]

kiu77, 2024-03-23:
@lumapu 
Noch ein winziger Nachtrag, nachdem ich ja jetzt schon gute Laune habe:
Wo hängt es bei mir, wenn rechts im Display keine Feldstärke fürs WLAN angezeigt wird?
Ich habe die Platine jetzt mit CMT2300A und NRF24L01+ bestückt und einen HSM-800 sowie einen HM-800 dran.
Empfang ist gut und Daten werden geholt.

[image]

kiu77, 2024-03-23:
Habs mir selbst beantwortet. Das Board ist im AP-Modus....
Sorry fürs Fragen.

kiu77, 2024-03-23:
Alles tut nun wie gewünscht.
Vielen Dank an alle für die Unterstützung.
Ich kann den Thread nun schließen.

[image]

closed 2024-03-23 by kiu77

RealNBB, 2025-03-10:
Hallo,

ich muss mich hier mal bitte einklinken:

Ich möchte einen HMS-1600 in meine bstehedne Ahoy-Installation einbinden. Bisher werkelt hier ein HM-800 zusammen mit einem HM-1500 (klar, mittels einem NRF24L01+).

Jetzt habe ich mir das CMT2300A aus dem Anhang geholt.

Pin 9 bis 12 auf dem CMT2300A ist klar, nur habe ich nun schon mehrmals gelesen, dass der „gpio3“ vom CMT2300A zwingend mit dem ESP32 verbunden sein muss – zumindest, wenn man Ahoy als DTU nutzen möchte.

Meine Frage ist nun, wo finde ich denn nun den „gpio3“ auf dem CMT2300A von oben? Oder gibt’s da gar keinen? Und stimmt das überhaupt, dass bei Ahoy der „gpio3“ benutzt werden muss?

Danke und Grüße!

[image]
[image]

stefan123t, 2025-03-10:
##### ESP32 variant with 30 pins
[image]

##### ESP32 variant with 38 pins
[image]

RealNBB, 2025-03-11:
Danke für deine Antwort.

Allerdings ist es so, dass sich der CMT2300A ja bereits auf einer Platine befindet (siehe Bild). Da gibt es eben die Stiftleiste, bei der ich nicht weiß, wo sich GPIO3 verstecken könnte.

Auf der anderen Seite – wenn man sich die Pinout-Spec anschaut – bleibt eigentlich nur noch Pin 13 (PB4 -> Module NSS Pin) übrig. Ich werde daher einfach mal testen, ob es der ist.

Falls er es nicht ist, löte ich einfach ein Jumperkabel direkt an den GPIO3 am CMT2300A an.

JogiAUT, 2025-05-09:
> Danke für deine Antwort.
>
> Allerdings ist es so, dass sich der CMT2300A ja bereits auf einer Platine befindet (siehe Bild). Da gibt es eben die Stiftleiste, bei der ich nicht weiß, wo sich GPIO3 [...]

Guten Tag, 
wie hast du denn das Problem gelöst? Ich stehe irgendwie vor selbigem. 

Nur...und jetzt wirds komisch....bei mir funktioniert es, ich weiß aber nicht warum. 🤪 

Wenn der Gpio34 am ESP32 unberührt bleibt, funktioniert es nicht. 

Deaktiviere ich ihn in den Pin-Settings geht es nicht.  

Fasse ich Pin 34 an (Haptisch mit meinen Fingern), kommt plötzlich eine Verbindung zum WR zustande und es klappt. ( Die angezeigte Verbindungsqualität geht inerhalb von einer Sekunde auf 4 Balken)

Stecke ich ein Kabel dran, geht es auch. (Ohne dass am anderen Ende was gesteckt wird)

Versuche ich es am Board vom Cmt2300A anzustecken geht es ebenfalls nicht. (Richtigen Pin kommte ich nicht herausmessen)

Wenn ich den Kontakt GPIO3 am Entwicklerboard zum CMT2300A herausmessen möchte (Durchgang) um richtigen Pin auf eben dieser Platine festzustellen erhalte ich kein Ergebnis.  

Also um das ganze zusammenzufassen: Sobald am Pin 34 ein Kabel lose rumbaumelt geht es. 😅

Ich würde aber gerne verstehen warum. Solche losen Enden in meinen Projekten (im wahrsten Sinne des Wortes) bereiten mir schlaflose Nächte.  😂