lumapu/ahoy #1515: Tagesertrag zurücksetzen (Ahoy DTU mit Batteriespeicher)

okapi21, 2024-03-20:
Version 0.8.83
Ich betreibe meine Anlage mit einem Batteriespeicher-System.
Welche Einstellung muss ich vornehmen, um den Tagesertrag um Mitternacht oder zu einen anderen Zeitpunkt zurückzusetzen?
Grüße Ralf

Peertje-62, 2024-03-20:
[image]
Hope this wil do
regards

okapi21, 2024-03-21:
Nur mit den oben gezeigten Einstellungen funktioniert es nicht. 
Also wie Metachuh schreibt, muss ich den Wechselrichter um 23:59 Uhr zurücksetzen.
Leider bin ich **totaler Laie** was API Befehle und Programmieren angeht. Gibt es irgendwo eine Schritt für Schritt Anleitung?
Grüße Ralf

okapi21, 2024-03-22:
Ich betreibe den Zenture Solarflow, und es geht darum den reinen Tagesvertrag, zurückzusetzen.
Von den oben aufgeführten Geräten läuft keins Nachts.
Grüße Ralf

okapi21, 2024-03-22:
Ein Shelly Pro 3em (2 Generation) ist voll in Betrieb (funzt auch super). Ein Skript über Diesen wäre natürlich eine Alternative.
Wie müsste das Shelly Skript aussehen?

okapi21, 2024-03-22:
Super, und keine Hektik.👍

okapi21, 2024-03-22:
Ich habe nur einen Wechselrichter Hoymiles HM -1500.
Die feste IP der Ahoy DTU lautet 192.168.188.47
Grüße Ralf

okapi21, 2024-03-23:
Vielen Dank für Ihre Mühe und die umfangreiche Anleitung schon mal im Voraus.  Werde zeitnah Feedback geben.👍
Grüße Ralf

lumapu, 2024-03-23:
ich kenne mich mit Shelly Scripten nicht aus, sieht sehr nach Javascript aus

okapi21, 2024-03-23:
Hallo, Metachuh
habe alles bis Punkt 7 abgearbeitet
```
7. mit klick auf das Schedules menü, in den zeitschaltuhr bereich wechseln.
hier sollte jetzt restart-battery-inverter-via-ahoydtu(meta).mjsals aufgabe erscheinen und bereits aktiviert sein (schieberegler rechts türkis).
```
[image]
leider erscheint in Schedules bei mir nichts.
was könnte ich noch ändern?
Grüße Ralf

okapi21, 2024-03-24:
Hallo Metachuh
```
könntest du vom shelly editor bereich einen screenshot oder foto von ganz oben ab zeile 1 machen ?
(dein passwort unkenntlich machen, aber die anführungszeichen sichtbar lassen zur kontrolle)
```
Zeile 1-23
[image]
Zeile 23-46
[image]
Zeile 47-70
[image]
Zeile 71-94
[image]
Zeile 95-118
[image]
Zeile 118-121
[image]
```
bei der script liste muss nur auf den play button geklickt werden, um das schedule anzulegen.
der autostart nach reboot schieberegler im scriptbereich kann deaktiviert (grau) bleiben
```
[image]
```
wenn das script einmalig korrekt durchgelaufen ist, sieht man im menü 
Advanced > KVS die zeile Script-Schedule-[nummer] mit einem wert, der 1 sein sollte

```
[image]

Bis hierhin funktioniert alles so wie du es beschreibst nur das Schedules-Menü bleibt leer.
Grüße aus der Oberlausitz
Ralf

`

RealNBB, 2024-08-01:
Bei mir läuft das Skript leider auch nicht auf einem Shelly PRO 3EM bzw. kein Eintrag wie bereits beschrieben unter Schedules.

Kann es aber sein, dass ein Leerzeichen im Skript in Zeile 8 fehlt --

 > var dtu_inverter_id =		0;<

stefan123t, 2024-10-30:
@RealNBB nein das Script funktioniert so wie es geschrieben ist nur für genau einen Inverter.
Ggf. mehrmals mit unterschiedlichen Namen `restart-battery-inverter-via-ahoydtu-0.mjs` anlegen und jeweils eine andere `dtu_inverter_id` angeben

@MetaChuh ich habe Dein Script mal bei mir auf einem Shelly Plus1PM installiert, es wird aber unter Schedules als `Call may not work as expected` angegeben, obwohl ich die Sonderzeichen `(` und `)` vorsorglich aus dem Script / Dateinamen entfernt habe.
Der Schedule selbst wird ordentlich übernommen und wie im MCRON Eintrag beschreiben `At second 0. At minute 59. At hour 23. Every days. Every months. Every weekdays.` eingeplant.

[image]

Im Advanced Schedule bzw. Edit Schedule steht dann unter 

**Actions:**
`{"method":"script.eval","params":{"id":1,"code":"shelly_meta_task()"}}`

[image]

Der KVS Key-Value Storage hat die von Dir erwähnte Variable `Script-Schedule-1` mit dem Wert `1`

**Key-Value Storage**

Key | Value
-- | --
`Script-Schedule-1` | `1`
`appliance_types` | `4`

[image]

stefan123t, 2024-10-30:
@MetaChuh ich habe es auch auf meinem Shelly Pro3EM mal kurz eingespielt und laufen lassen, leider noch weniger Erfolg.

Dort habe ich ähnlich wie @okapi21 keinen Schedule Eintrag erhalten. Lediglich im KVS wird die Variable `Script-Schedule-1` angelegt.

Hier die Ausgabe auf der Diagnostic Console

```
shos_rpc_inst.c:242 script.start via WS_in 192.168.178.42:36790 21:40:45
shelly_user_script.:211 JS RAM stat: initial: 70596 after: 70568, used: 28 21:40:45
shelly_script.cpp:90 Running 'script_1.js'... 21:40:45
shelly_ejs_rpc.cpp:41 Shelly.call KVS.Get {"key":"Script-Schedule-1"} 21:40:45
shelly_user_script.:223 JS RAM stat: after user code: 70596 after: 69436, used: 1160 21:40:45
shelly_notification:162 Status change of script:1: {"id":1,"running":true} 21:40:45
shos_rpc_inst.c:242 KVS.Get via loopback 21:40:45
shelly_ejs_rpc.cpp:41 Shelly.call Schedule.List {} 21:40:45
shos_rpc_inst.c:242 Schedule.List via loopback 21:40:45 
```

[image]

Vielleicht hängt es mit der API Version [`Gen 2+ Device API`](https://shelly-api-docs.shelly.cloud/gen2/) des Shelly Plus1PM und des Shelly Pro3EM zusammen ?

* Shelly Plus1PM: https://shelly-api-docs.shelly.cloud/gen2/Devices/Gen2/ShellyPlus1PM
* Shelly Pro3EM: https://shelly-api-docs.shelly.cloud/gen2/Devices/Gen2/ShellyPro3EM/
* Shelly Script Language Features: https://shelly-api-docs.shelly.cloud/gen2/Scripts/ShellyScriptLanguageFeatures