lumapu/ahoy #1516: HMS-1600 mit Buchstabe in der Seriennummer

maikal80, 2024-03-20:
### Platform

ESP32

### Assembly

I did the assebly by myself

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.36

### Github Hash

###

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

AhoyDTU mit der aktuellsten Firmware.

### Debug Serial Log output

```shell
None
```

### Error description

Ich habe einen Hoymiles HMS-1600. In der Seriennummer ist an der 5.Stelle ein Buchstabe. Mir wird dann auf der AhoyDTU Weboberfläche der Eintrag rot markiert. Eine Kommunikation mit dem Wechselrichter kommt nicht zustande.
Was kann ich machen oder ist es ein Grundsätzliches Problem? 
[image]
[image]

lumapu, 2024-03-21:
das sollte kein Problem darstellen - kommt nur zum ersten mal vor. Ich werde das fixen. Vielen Dank für den Hinweis

beegee3, 2024-03-22:
A steht für das Baujahr 2024. 1 = 2015, 2 = 2016, ..., F = 2028. Spätestens 2029 funktioniert diese Zählung also nicht mehr, mal sehen, was HM dann macht. Nach dem Baujahr sollte eigentlich die Produktionswoche (01 bis 52) angegeben sein. Hier steht aber 00; merkwürdig 🤔.
@lumapu: ist evtl. nicht ganz so einfach, zumindest kann es bei JSON Ein- und Ausgaben schnell inkompatibel zu den vorherigen Versionen werden. Hab' die Seriennummer spaßeshalber mal auf String umgestellt und mir prompt beim Einlesen einer alten Datei den ESP abgeschossen. Daher mach ich besser keinen Code-Vorschlag 😄

lumapu, 2024-03-22:
schade, wo ich deinen Namen gelesen habe dachte ich gleich "ui schon gelöst". Ich probiere mich auch mal, danke für den Hinweis mit dem JSON sowohl als Setting als auch Export

lumapu, 2024-03-22:
@maikal80 bitte testen 😉

maikal80, 2024-03-23:
Mit der neuen Firmware 0.8.97 funktioniert es ohne Probleme. Die Seriennummer mit Buchstabe wurde akzeptiert. Vielen Dank.

lumapu, 2024-03-23:
Danke fürs Feedback, die Kommunikation geht auch?

beegee3, 2024-03-23:
😆 einfach in JSON die Serial HEX Ausgabe als "note". 👍👍👍

theunknownstarman, 2024-03-23:
Super Sache mit der Änderung. GENAU das Problem habe ich bei dem diese Woche gelieferten WR auch :)

lumapu, 2024-03-23:
> 😆 einfach in JSON die Serial HEX Ausgabe als "note". 👍👍👍 

ja, das war smart oder? 😇
Blöd mit beim Import einer solchen JSON wird dieses unnötige Feld dann mitgeschleppt. Man könnte das noch beim Import eliminieren ...

beegee3, 2024-03-24:
naja, man könnte natürlich auf dieses Feld verzichten. Jeder, der sich die JSON Serial ansieht, kann bei Bedarf selbst die Zahl in HEX umwandeln (z.B. mit dem Windows Rechner in der Programmierer Ansicht). Oder du gibst in der Web Anzeige neben der HEX Serial den uint64_t Wert aus. Die `helper.cpp` `Serial2u64` Funktion wird ja jetzt nur noch für die Umrechnung der 'alten' JSON Serial gebraucht. Vielleicht baust du noch ein, das bei 'alter' Serial automatisch die 'neue' gespeichert wird. Dann kann (spätestens nach einer neuen Release Version) die `Serial2u64` entfernt werden.
(off topic zur Webanzeige: kannst du bei Gelegenheit die deutsche Übersetzung der token "BTN_INV_ADD" und "TAB_INPUTS" auf Umlaute umstellen? Und die Texte von "BTN_NEXT", "TEST_CONNECTION" und "ERR_INDEX" enthalten "ü" statt "&uuml;", ist das OK?)

lumapu, 2024-03-24:
weiß nicht, ob das mit der Übersetzung so geht, da diese Buttons in C definiert werden und nicht in der HTML Seite. Hatte da anfänglich Probleme und keine weitere Muße tiefer reinzuschauen.
Die Anregungen bzgl. Export finde ich gut, ich denke den HEX Export braucht man, wenn man es manuell wieder eingeben will und die JSON als Vorlage nimmt - leider in letzter Zeit häufiger ein use-case ...

beegee3, 2024-03-25:
bei "TAB_INPUTS" ist es einfach "Eing&auml;nge". Bei "BTN_INV_ADD" funktioniert &uuml; nicht, da auf Großbuchstaben gewandelt wird. Aber Unicode geht immer: "Wechselrichter hinzuf\u00FCgen".

lumapu, 2024-03-26:
danke 😉

MiniOh, 2024-06-16:
Hallo @lumapu 
Ich habe auch einen neuen HMS-1600-4T mit 2 Buchstaben in der SN.
Dazu habe ich von der letzten stable auf die 0.8.126 geupdatet.
Danach lässt sich die SN hinzufügen, aber bisher keine Komunikation mit dem Inverter.
Bisher mit 863, 865 und 868MHz getestet.

Inverter "ist gerade nicht verfügbar"

Ideen?

MiniOh, 2024-06-17:
@lumapu 

Ich nehme meine Frage zurück.
Gestern ne Stunde getestet, ohne Erfolg.
Heute morgen funktionierte die Kommunikation mit dem HMS-1600 wie erwartet.

Danke dennoch.

lumapu, 2024-07-11:
wollte schreiben: warte mal auf den nächsten Morgen, aber du hast es ja selbst gemerkt 😅

stefan123t, 2024-07-14:
Jahreszahl jetzt in Hex :smile: 

@lumapu schau Dir bitte mal analog dazu die beiden HERF Inverter issues / topics bei OpenDTU an:

* https://github.com/tbnobody/OpenDTU/issues/519
  hier geht es zwar speziell um HERF Inverter, die verwenden das für OEMs gedachte `sn92long()` Serial Number Format mit einem anderen Charset:
  * https://github.com/tbnobody/OpenDTU/issues/1751

Aber auch dafür muß die Eingaberoutine aufgebohrt / angepaßt werden, am besten geht das vermutlich auch bei AhoyDTU im Browser / JavaScript.

dbMerkel, 2024-07-22:
Hallo, wollte auch gerade den HMS 1600 4T mit AhoyDTU verbinden. Leider kann ich die Serialnummer nicht eingeben.
1164A00CA84A.
Habe die Version ==>  You are using the current stable release: 0.8.83

Wann kann ich mit einem Update rechnen, oder mache ich etwas falsch?

stefan123t, 2024-07-31:
@dbMerkel Du bist auf der Stable, der Entwicklungsbranch development03 ist derzeit bei v0.8.129

Alternativ kannst Du ein Backup Deiner Settings machen und die Serial ID in der json Datei anpassen. Evtl wird die ID beim Hochladen der Config nicht geprüft. Sozusagen ein Workaround.

closed 2024-08-16 by lumapu

eugen2304, 2025-04-20:
Moin Moin an alle, Frohe Ostern :)
bin neu hier, habe ehrlich gesagt nicht viel Ahnung davon.
hatte früher HMS800 gehabt, habe mir jetzt andere Wechselrichter gegönnt HMS 1600 -4T, auf Ahoy update gemacht auf 0.8.126 Version, wie oben einer geschrieben hat, hat geklappt, könnte S/N mit Buchstaben eingeben, alles geklappt bis auf das:
1. Interrupt Pin funktioniert -unbekannt-
2. Mqtt nicht aktiviert.
 
habt ihr eine Idee, was ich machen kann?

Gruß Eugen

TrandyAndy, 2025-07-04:
selbiges Problem mit der version: 0.8.155 gehabt dann auf 250309_ahoy_0.8.155_5feb293_esp32-wroom32-de.bin gewechselt und es ging.
und alles schön auf Deutsch.
Nur der mDHCP Name wird jetzt nicht mehr übernommen.