lumapu/ahoy #1519: Vorschlag: "update" und "Import"- Buttons erst freigeben, wenn Datei eingetragen ist

Gubi2023, 2024-03-21:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0-8.95

### Github Hash

????

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Bedienung

### Error description

wäre es möglich,  "update" und "Import"- Buttons erst freizugeben, wenn auch eine entsprechende Datei angewählt wurde?
für mich ist es immer wieder verwirrend, wen die DTU neu gestartet wird und ich weiss gar nicht, was gerade geladen wird....

Gubi2023, 2024-04-18:
danke, so finde ich es besser...

Gubi2023, 2024-04-18:
hallo, was mir gerade noch aufgefallen ist: könnte man den Eintrag noch auf den richtigen Typus überprüfen (Json bei Import bzw. .bin bei update) oder ist der Aufwand zu gross?

closed 2024-08-16 by lumapu

Gubi2023, 2024-08-16:
Hallo, finden meine letzten Wünsche keine Berücksichtigung???
https://github.com/lumapu/ahoy/issues/1519#issuecomment-2063387440

lumapu, 2024-08-16:
mach am besten einen neuen issue auf, der hier ist abgearbeitet (das ursprüngliche Thema)

PR #1586 (2024-04-17): Disable upload and import buttons when no file is selected
“Upload” and “Import” are disabled until a file is selected. Closes #1519