lumapu/ahoy #1520: ESP8266: ESP.getResetInfo() statt ESP.getResetReason() verwenden

Zordandi, 2024-03-22:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

square dot

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.84

### Github Hash

f500999

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

- MqTT running

### Error description

Könnte man bei der Ausgabe der ESP8266 System Information nicht ESP.getResetInfo() statt ESP.getResetReason() verwenden (RestApi.h)?

Vorteil: man bekommt in den Fehlerfällen "Hardware Watchdog", "Exception" und "Software Watchdog" etwas mehr Info bzgl. des Fehlers. Insbesondere die Exception Causes (EXCCAUSE).
Nachteile: statt eines const Buffers werden am Heap 200 Byte für die Erzeugung des Strings gebraucht und die Ausgabe ist entsprechend breiter.

Auszug aus Esp.cpp:
        sprintf_P(buff, PSTR("Fatal exception:%d flag:%d (%s) epc1:0x%08x epc2:0x%08x epc3:0x%08x excvaddr:0x%08x depc:0x%08x"),
            resetInfo.exccause, resetInfo.reason, getResetReason().c_str(), resetInfo.epc1, resetInfo.epc2, resetInfo.epc3, resetInfo.excvaddr, resetInfo.depc);

stefan123t, 2024-10-30:
@Zordandi  hier ist der relevante Code aus der [`RestApi.h`](https://github.com/lumapu/ahoy/blob/2d5d62eb0e2b35c743db2b75079fb039c0f9c191/src/web/RestApi.h#L825-L870)

https://github.com/lumapu/ahoy/blob/2d5d62eb0e2b35c743db2b75079fb039c0f9c191/src/web/RestApi.h#L825-L835
https://github.com/lumapu/ahoy/blob/2d5d62eb0e2b35c743db2b75079fb039c0f9c191/src/web/RestApi.h#L863-L870

Hast Du es einmal versucht die Zeile 869 nur für den ESP8266 anzupassen ?

Zordandi, 2024-11-01:
> Hast Du es einmal versucht die Zeile 869 nur für den ESP8266 anzupassen ?

Nein, ich habe den Source Code noch nie selber versucht zu kompilieren.
Inzwischen bin ich auch auf einen ESP32 umgestiegen.

stefan123t, 2024-11-02:
@lumapu kannst Du das mit ESP.getResetInfo() anstelle von ESP.getResetReason() auf einem ESP8266 testen ?

@Zordandi oder sollen wir den Feature Request von Dir schließen ?

Zordandi, 2024-11-03:
> @lumapu kannst Du das mit ESP.getResetInfo() anstelle von ESP.getResetReason() auf einem ESP8266 testen ?

Ich hab den hier noch rumliegen und könnte das versuchen zu testen.
Allerdings wäre es gut ein fertiges Binary zu bekommen, da ich noch nie selber kompiliert habe.