lumapu/ahoy #1524: Info: Kommunikation HMS/CMT2300A sehr viel besser als HM/NRF24L01+

kiu77, 2024-03-24:
## Hardware
ESP32 38 Pin mit ext. Antenne

### nRF24L01+ Module
nRF24L01+ mit externer Antenne
CMT2300A mit externer Antenne

### Power Stabilization:
Extra 3,3V-regler mit 3 Vielschicht-Cs 22µF

### Version / Git SHA: 
Version: 0.8.97

### Build & Flash Method:
AhoyDTU Webinstaller

Nur zur Info:
Ich habe gerade meine DTU mit zweifacher Funkmodul-Bestückung (NRF24L01+ und CMT2300A) fertiggestellt und zwei Inverter (HM-800 und HMS-800) damit angebunden. Es sind der externe Antennen verbaut.
1. Funktioniert es prima.
2. Ist die WLAN-Anbindung besser: 4 vs drei Balken bei externer vs gedruckter Antenne.
3. Ergibt sich mit den Hoymiles-Invertern der neuen HMS-Reihe eine deutlich stabilere Funkverbindung.
    Meine DTU zeigt -27dB bei CMT2300A/HMS an, was eine enorme Feldstärke ist. Entfernung = 5 m durch eine 32cm Mauer.
    Beim NRF24L01+/HM ergibt sich eine Feldstärke von -64dB bei 10m durch eine 32cm Mauer (Tonziegelhohlsteine).
    Das rechnerische Verhältnis beträgt 37dB = Faktor 70. Kalkuliert man die Entfernungsdifferenz mit ein, ergeben sich
    noch 25dB = Faktor 17 - immer noch super.
    
Ergo würde ich definitiv keinen HM-Inverter mehr kaufen. Das freie 868MHz-Band wird zudem heutzutage weniger genutzt als WLAN bzw. der Bereich um 2,4GHz, auf dem der NRF24 funkt. Von daher dürften auch 100m Distanz zwischen Inverter und DTU beim CMT2300A kein Problem sein. Super Sache das. 
Besonders interessant finde ich, dass eine gemischte Bestückung sehr gut klappt und man als nicht einen schon vorhandenen HM-Inverter gegen einen HMS austauschen muss, wenn man einen neuen HMS dazu kauft.

So sieht das Teil aus. Schwarz = 2,4GHz und weiß = 868MHz.

[image]

Dirk-Ahoy, 2024-03-24:
Die Anzeige beim NRF steht immer auf 64dbm da die Treiber keine Auswertung anbieten - einzigst wird unterschieden ob  <64 oder>64 ist.   Daher kann man den Wert nicht auf guten oder schlechten Empfang zur Auswertung heranziehen bzw mit HMS vergleichen.  Da muss man eher in der Statisik gucken.  Ich betreibe im Garten einen HM1500 der knapp 90m entfernt ist ohne Probleme.  Ahoy steht in der Küche zum auslesen. 

 Ich hab bei mir auch beide laufen am Fusion Board.   Leider gibts die HMS800 ja meist nur noch in "W" Ausführung die man dann nicht mit Ahoy nutzen kann - muss man genau aufpassen beim Kauf.

Schönen Sonntag

kiu77, 2024-03-24:
@Dirk-Ahoy 
Danke für die Aufklärung. Okay, dann ist ein größerer Teil meiner Überlegungen sinnlos :(
90m mit HM ist echt gut.
Ja, das stimmt, dass man mit dem W bei den HMS. Ich hab den richtigen ;-)

DB1BMN-collection, 2024-03-24:
Sieht geil aus das Teil!
Scheint so, als ob die Tage von HM + WeMos D1 mini + nRF gezählt sind.
Da wird man wohl was neues designen müssen.

kiu77, 2024-03-24:
Danke!
Für mich sieht das fast aus wie ein 80er Jahre Robby :D

closed 2024-03-26 by kiu77