lumapu/ahoy #1528: 240325_ahoy_0.8.98_f47cb10_esp8266-all-de: History fehlt

Gubi2023, 2024-03-26:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.98

### Github Hash

f47cb10

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

update auf esp866-all-de

### Error description

240325_ahoy_0.8.98_f47cb10_esp8266-all-de: History fehlt

lumapu, 2024-03-26:
duplicate #1530

closed 2024-03-26 by lumapu

Gubi2023, 2024-03-26:
tut mir leid, in der 8.100 fehlt immer noch die History in der all de Version! in der all-Version ist sie vorhanden!