lumapu/ahoy #1529: Problems with NTP [Bug]

mellow5, 2024-03-26:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.97/0.8.98

### Github Hash

9971ed7

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

After update from 0.8.96 to either 0.8.97 or 0.8.98

### Error description

After reboot the time doesn't synchronize unless I manually synchronize.

lumapu, 2024-03-26:
veni vidi vici
-> solved 😉

hrolofs, 2024-03-26:
Update auf die 0.8.99, 240326_ahoy_0.8.99_d5ed027_opendtufusion.bin.

Danach Fehlermeldung: time not set. No communication to inverter possible

Auch manueller Sync klappt nicht.

Im Download der 0.8.99 von heute (26.03) liegen Files mit dem Zeitstempel vom 24.03. drin.

Downgrade auf 0.8.97, ntp funktioniert wieder.

lumapu, 2024-03-26:
sorry für die Nachtaktiven 😂 - der commit kommt erst noch, die gefixte Version ist erst `0.8.100`

hrolofs, 2024-03-26:
ok, alles klar. Tschuldigung für meine Ungeduld. ;-)

lumapu, 2024-03-26:
find's irgendwie lustig und auch erstaulich dass so viele noch aktiv sind - es ist Dienstag 23:55 ...

hrolofs, 2024-03-26:
oh, schon Dienstag. ;-)

lumapu, 2024-03-26:
so jetzt sinds nur noch 5 minuten bis zum Test ... gute Nacht 💤

mellow5, 2024-03-27:
Funktioniert bei mir mit 0.8.100. 👍

closed 2024-03-27 by mellow5