lumapu/ahoy #1530: V0.8.98: keine Werte bei Total Power Today /Sunset-Sunrise fehlt in System

Gubi2023, 2024-03-26:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.98

### Github Hash

f47cb10

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

update auf esp8266-all

### Error description

V0.8.98: keine Werte bei Total Power Today /Sunset-Sunrise fehlt in System

lumapu, 2024-03-26:
das liegt am defekten NTP Sync, gefixt in `0.8.100` 😉

Gubi2023, 2024-03-27:
V0.8.100: Sunset-Sunrise fehlt immer noch in System  
und in den Settings ist kein Display mehr enthalten

lille2000, 2024-03-27:
Bei mir fehlt „History“ immer noch in der Menüleiste mit 0.8.100. (esp8266 de all)

Gubi2023, 2024-03-28:
fixed mit 0.8.101

closed 2024-03-28 by Gubi2023