lumapu/ahoy #1531: 7 Platinen zu verschenken / für Bestückung mit NRF24 + CMT2300A

kiu77, 2024-03-26:
## Hardware
ESP32

### RF Module
nRF24L01+
CMT2300A, Typ: E49-900M20S

### Antenna:
* [ ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
3 x 22uF multilayer capacitors
plus low power 3.3V voltage regulator

Ich hätte noch 4 Platinen für ESP32-38-Pin und 3 Platinen für ESP32-30-Pin zu verschenken.
Beide ESP-Versionen sind mit aufgedruckte WLAN-Antenne, insbesondere beim 38-Pin
ist das wichtig, da die Version mit SMA-Steckverbindung für externe Antenne breiter ist
und nicht passt.
Beide Platinen sind Prototypen, aber sind funktionieren gut. Sie sind für
die Bestückung mit nRF24L01+ (für die HM-Serie von Hoymiles) und CMT2300A
(für die HMS-Serie ohne WLAN) gedacht. Es gehen auch beide Funkmodule gleichzeitig,
sodass man HM- und HMS-Inverter gemischt abfragen kann. Klappt bei mir gut.

Statt den auf dem ESP32-Boards integrierten 3,3-V-Regler zu verwenden, ist
hier ein kleiner dreipoliger SMD-Regler für 3,3V vorgesehen (1=GND, 2=Out, 3=In),
wodurch das Funkmodul optimal entkoppelt wird. Man muss nicht unbedingt den
angegebenen Typ AMS1117-3.3 nehmen, es gibt auch Vergleichbare. Statt eines Elkos
sind 3 kleine SMD-Vielschicht-Kondensatoren mit 22µF/10V vorgesehen (SMD 1210 oder 1206),
was ebenfalls positiv für die Stabilität ist.
Bestückt werden muss die Platine mit einem 30- bzw. 38-poligen ESP32-BOB (mit aufgedruckter
Antenne) und einem oder beiden Funkmodulen. Beim nRF24L01+ gibt es Versionen mit
aufgedruckter Antenne oder das von eByte mit Schraubanschluss für eine 2,4GHz-Antenne.
Die Schaltung ist simpel und nachfolgend abgebildet. Das Löten eines CMT2300A ist nicht trivial,
aber ich kriege es gut und auch Anhieb fehlerfrei hin, trotz großen Händen.

Wer zuerst kommt, mahlt zuerst. Anfragen per Text-Datei in einem Ordner auf auf meinem NAS ablegen.
Format = .txt Da bitte eine Datei mit Nickname und kompletter Anschrift sowie den Wunsch nach der
Platine mit 38- oder 30-poligem ESP32 ablegen.

Link: [https://quickconnect.to/tomesc](https://github.com/lumapu/ahoy/issues/url)
User: PCB
PW: AhoyDTU

Es wäre nett, wenn man dafür eine kleine Spende an lumapu vornimmt:
[https://www.paypal.com/paypalme/lupusch](https://github.com/lumapu/ahoy/issues/url)
Die Briefmarke und den Briefumschlag spendiere ich.

Hier die Version für 30-polige ESP32 (davon habe ich 4 Stück):
[image]
[image]

Und hier die Version für 38-polige ESP32 (davon habe ich noch 3 Stück):
[image]
[image]

kiu77, 2024-03-26:
So sieht eine Platine fertig bestückt mit 30-poligem ESP32 und CMT2300A aus:
[image]

tester277, 2024-03-26:
Hallo kiu77 leider sind keine Schreibrechte für den PCB vorhanden.
Sonst wäre ich gern dabei.
Danke und Gruß

kiu77, 2024-03-26:
@tester277 
Sorry, habs vergessen zu aktivieren. Jetzt geht es...

kiu77, 2024-03-26:
Noch 4 x 30er und 2 x 38er da

firedp, 2024-03-26:
> Noch 4 x 30er und 2 x 38er da

ich würde mich über eine 30er sehr freuen. Bin aber scheinbar mit der Forumsbedienung überfordert. Wo darf man seine Adresse für den Versand hinterlegen?

kiu77, 2024-03-26:
Oben ist ein Link auf meinen NAS, dort im Ordner Deine Datei ablegen. Bei mir musste ich den URL kopieren und in einem neuen Tab im Browser öffnen, damit der Link aufgerufen wird.

kiu77, 2024-03-27:
@tester277 
Deine Platine ist unterwegs. Vor der Bestückung. Rufe mich mal kurz an...

hackman6, 2024-03-30:
Hallo kiu77,
hast Du rein zufällig noch eine 30er am Start?

kiu77, 2024-03-30:
Klar, lege Deinen Nick und Anschrift in den Ordner ab...

MaDill, 2024-04-07:
Hallo kiu77, wenn es noch eine 38er Platine gibt, würde ich mich darüber freuen. Danke schon mal.

kiu77, 2024-04-07:
@MaDill 
Okay. Welche Version hättest Du gerne?
38-pol. geht klar. Da gibt es noch die Variante mit dem ESP32 mit aufgedruckter Antenne (standard in weiß) oder die mit dem ESP mit externer Antenne (SMA-Buchse auf den BoB).

MaDill, 2024-04-07:
> ESP32 mit aufgedruckter Antenne

Den habe ich. Danke

kiu77, 2024-04-07:
@MaDill 
Ist eingetütet und geht morgen raus.
Die Schaltung kannst Du Dir von meinem NAS runterladen.
Bei Fragen: melde Dich. Kontaktdaten sind im Briefumschlag.

kiu77, 2024-04-11:
Noch 1 x 38er- und 4 x 30er-Platinen sind da.

RF-Signal, 2024-04-14:
Hallo,
würde mich über eine "ESP32-38-Pin" PCB freuen.
Danke.

kiu77, 2024-04-14:
@RF-Signal 
Eine 38er ist noch da. Dann lege wie oben beschrieben eine Textdatei mit Nickname und Anschrift in den Ordner auf meinem NAS ab.

gitandibra, 2024-05-04:
Hallo,
würde mich  sehr über eine "ESP32-30-Pin" PCB freuen.
(sofern noch vorhanden)
Habe wie beschrieben die Daten gesendet.
Danke.

kiu77, 2024-05-04:
@gitandibra 
Leider sind da keine Daten. Lege bitte wie beschrieben auf dem NAS ein Text-File ab.
Es sind noch 30er Platinen da.

kiu77, 2024-05-05:
@gitandibra 
Sodele, Daten sind da und Platine ist eingetütet.

giGa977, 2024-05-08:
Hallo,
Ist noch eine für ein 30pin übrig?

grüssen

kiu77, 2024-05-12:
@giGa977 
ja, es sind noch drei da.

giGa977, 2024-05-12:
> @giGa977 ja, es sind noch drei da.

Datei hatte ich schon erstellt, dann würde ich mich doch glatt über ein Exemplar sehr freuen.

kiu77, 2024-05-13:
@giGa977 
Ich finde die Textdatei auf dem NAS nicht.
Legst Du sie nochmals da ab?

giGa977, 2024-05-13:
Liegt noch im Ordner, /home/PlatineVerschenken oder ist der ordner falsch ?

kiu77, 2024-05-13:
Oh sorry, ich sehe es jetzt. Auf die Idee bin ich nicht gekommen ;-)
habs im Eingangs-Post vermerkt. Ich tüte Deine Platine ein.

giGa977, 2024-05-13:
Ich habe eben auch eine spende abgesetzt aber irgendwie scheint was schief gelaufen ist was angekommen ?

kiu77, 2024-05-13:
Auf die Spenden habe ich keinen Einfluss.
Der Link: [https://www.paypal.com/paypalme/lupusch](https://github.com/lumapu/ahoy/issues/url)
funktioniert bei mir am Mac jedenfalls auch nicht,. Aber ein man ihn kopiert und damit eine neue Seite öffnet,
erscheint die Paypal-Seite von lumapu und man kann was senden. Bei mir klappt das.

giGa977, 2024-05-13:
Wie auch immer mittlerweile sehe ich es als gesendet.  ;) Vielen dank und einen Schönen abend noch.

Pattis74, 2024-05-18:
Hallo,
würde mich auch über eine PCB freuen.
Da ich beide Varianten von Esp32 habe, (38 Pin und 30 Pin), was noch vorhanden ist.
Danke!

kiu77, 2024-05-19:
@giGa977 
Irrerweise hat die Post den brief zurückgeschickt. Angeblich zu schwer, obwohl er nur 15g wog...
Habe die Platine neu eingetütet und nochmals verschickt.

[image]

kiu77, 2024-05-19:
@Pattis74 

Eine 30er Platine ist unterwegs.

giGa977, 2024-05-21:
Alles gut die Deutsche Post ist halt ab und anmal bekloppt :) , anstelle es zuzustellen nene lieber zum absender zurück. kostet ja nichts :) grüße

giGa977, 2024-05-22:
@kiu77 Platine ist angekommen, werde sie noch heute bestücken vielen Dank dafür. Riesiges Kompliment für diese tolle Aktion.

grüsse giGa

Pattis74, 2024-05-22:
@kiu77  Vielen Dank! Ist bei mir auch angekommen.

Grüsse Pattis

stefan123t, 2024-10-30:
@kiu77 das ist ein schönes Platinen-Layout, besonders die Pin-Header für ein CMT2300A Modul und ein NRF24L01+ Modul sind m.E. sehr praktisch gelöst.

Ich habe mal die Links korrigiert und hoffe Du kannst noch lange die Leute mit PCBs versorgen :+1:

hackman6, 2024-10-31:
Hi, ist die eine, die noch übrig ist, 30-Pin?