lumapu/ahoy #1533: V8.100:Fehlermeldung beim öffnen der Webside[Bug]

Gubi2023, 2024-03-27:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.100

### Github Hash

????

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

update auf V0.8.100

### Error description

heute nacht von 0.8.93 auf V0.8.100 upgedatet.
Heute morgen DTU nicht mehr sichtbar/erreichbar.
Navh Spannungswiederkehr DTU sichtbar, aber beim Versuch, eine Webseite auszuwählen, kam Fehlermeldung. Siehe Screenshot
[image]

Gubi2023, 2024-03-27:
@MetaChuh 
danke für die schnelle Rückmeldung. ich teste immer alle neue Versionen an, bei allen gleiche Fehlermeldung. 
Im 2. oder 3. Versuch gehts dann, aber Verbindung bricht seit 8.97 ständig ab bzw. DTU startet sich ständig neu. Bin deshalb zurück auf 8.93

kiu77, 2024-03-27:
@MetaChuh 
Ja, bei mir ist es das Gleiche mit ESP32: Der Menüpunkt Verlaufs bzw. history ist weg. Mit Deinem Direktlink geht es dann.

[image]

Gubi2023, 2024-03-27:
du Glücklicher, bei mir läuft die V8.100 gar nicht, inzwischen zickt sogar die 8.93! (DTU nicht mehr erreichbar, obwohl sie in derFritzbox angezeigt wird!)

kiu77, 2024-03-27:
Oh, in solche quasianalogen Fällen half bei mir auch nur komplett neu flashen mit erase des flash Speichers und dann updaten. 
Evtl. hilft auch Strom weg und nach dem Hochbooten kurz via WLAN unter der http://192.168.4.1 zugreifen. Wenn das geht, dann kann man zumindest die Einstellungen sichern, falls noch nicht geschehen.

Gubi2023, 2024-03-28:
@MetaChuh: tut mir leid, ich verstehe nur Bahnhof.
Ich habe einen ESP8266, NRF24 ohne externe Antenne und ohne Elko. Ich frage den WR in 20 sec.Intervallen ab.
MqTT verwende ich nicht, da ich keinen IO-Broker oder ähnliches habe, d.h ich greife nur über die Webseite auf die DTU zu...
Ich kann weder Progamme lesen, geschweige denn schreiben oder interpretieren.
Mir fällt nur auf, dass wenn ich nicht auf die DTU zugreife, alles anstandslos funktioniert und sobald ich die Webseite öffne, die DTU sich ständig neu startet. (Seit V0.8.94) mit V 0.8.93 läüft sie tagsüber, erst gegen Abend startet sie sich dann auch wieder in unregelmässigen Abständen neu, ohne dass ich etwas gemacht hätte.

kiu77, 2024-03-28:
> ESP8266, NRF24 ohne externe Antenne und ohne Elko

Kein Wunder. Den Elko sollten Du mal nachrüsten, zumindest. Ein NRF24 ohne externe Antenne? Um den Typ zu bestimmen könntest Du mal ein Foto davon einstellen...

Gubi2023, 2024-03-28:
ich hab das Teil fertig gekauft mit Gehäuse. Bei einem Wechsel aus ESP32 bräuchte ich alles neu, incl. Gehäuse.
Open-DTU geht ja angeblich mit ESP8266 nicht..
Elko nachrüsten ist nicht so einfach, da ich mir das nicht mehr zutraue (zu zittrig/ zu mini)

Gubi2023, 2024-03-28:
[image]

kiu77, 2024-03-28:
Bez. Elko. Du könntest ihn nachträglich so drauflegen (oder löten lassen):

[image]

Gubi2023, 2024-03-28:
@kiu77 
danke, versuch ich als Soforthilfe. 
@MetaChuh 
Wenn nichts nützt: würde AllianceApps OpenDTU auch gehen, da ich mein Balkonkraftwerk vermutl. nicht erweitern werde ?

kiu77, 2024-03-29:
@Gubi2023 
das von MetaChuh empfohlene Modell hat weiter den Vorteil eine externen Antenne für das Funkmodul, was bessere Funkverbindung mit dem inverter mit sich bringt.

Gubi2023, 2024-06-06:
Habe heute mal nach längerer Zeit wieder meine 8266 reaktiviert (V0.8.122) 
Fehlermeldung vom ursprünglichen Issue  kommt immer noch...
[image]

closed 2024-08-16 by lumapu

Gubi2023, 2024-08-16:
jetzt auf  #1725