lumapu/ahoy #1534: [Bug] 0.8.100 history is missing in ESP32_de and ESP8266_de

tavoc, 2024-03-27:
### Platform

ESP32

### Assembly

I did the assebly by myself

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.100

### Github Hash

GIT SHA: 4b75e72 :: 0.8.10

### Build & Flash Method

ESP Tools (flash)

### Setup

the newest update removed the history button in the webinterface. I was coming from 0.8.97 (by the way I very much like the new look of the history).

The ESP32 and ESP8266 (booth german version) looks the same.
By adding /history on ESP32 I can see it, on ESP8266 it is an empty page
[image]
[image]

### Error description

the newest update removed the history button in the webinterface. I was coming from 0.8.97 (by the way I very much like the new look of the history).

The ESP32 and ESP8266 (booth german version) looks the same.
By adding /history on ESP32 I can see it, on ESP8266 it is an empty page

Gubi2023, 2024-03-27:
see also  #1530

Di-Di-Du, 2024-03-27:
on my ESP8266 , I can see the history correct, wehen adding "/history" beehind the IP in the browser adress line.

Gubi2023, 2024-03-27:
see #1533

kiu77, 2024-03-27:
Beim Fixen fiel mir noch was Sprachliches auf:
Beim Verlauf wäre oben besser: "Leistungsverlauf letzte 20 min"
Und dann unten "Leistungsverlauf heute".
Denn es dreht sich ja nicht um eine Gesamtleistung, also nichts Kumuliertes, sondern um einen Verlauf.

tavoc, 2024-03-27:
das sind nicht immer 20 Min. Beim ESP32 ist es bei mir eine Stunde. Ich glaube es gibt bereits ein Issue, wo das Wording geändert wurde

Gubi2023, 2024-03-27:
Aktualisierung ist abhängig vom eingestellten Intervall beim WR

kiu77, 2024-03-27:
@Gubi2023 
Ahja. Das macht Sinn. Eine Speicher für soundsovielte Messungen, daher. Bei mir 5s für 2 Inverter. Vermutlich also 240 oder 120 Messungen für den kurzfristigen Verlauf.
Demnach wäre also folgende Beschriftung sinnvoll:
Beim Verlauf wäre oben besser: "Leistungsverlauf letzte XX min", wobei XX ja errechnet und angezeigt werden kann.
Und dann unten "Leistungsverlauf heute".

kiu77, 2024-03-27:
Hab nochwas bei History gefunden: Ober wird die korrekte Ortszeit angegeben und unten ist man schon eine Stunde weiter...

[image]

tavoc, 2024-03-27:
das unten ist die errechnete Zeit des Sonnenuntergangs. Oben rollierend

kiu77, 2024-03-27:
@tavoc 
Schon klar, aber die Zeit unten ist falsch. 1h zuviel.

Gubi2023, 2024-03-27:
ja, aber die Anzeigen sollten trotzdem an denselben Stellen anfangen und enden

kiu77, 2024-03-27:
@Gubi2023 
Anfangen sollten sie oben und unten zu unterschiedlichen Zeiten, da der obere Graph ja ein Zoom des Endes des unteren ist. Aber enden bzw. die letzten Werte sollten sich auf die gleiche Zeit beziehen.
Es sieht so aus, als sei unten die aktuell noch aktive Winterzeit ignoriert und vorzeitig auf Sommerzeit geschaltet worden, aber eben nur unten.

Gubi2023, 2024-03-27:
@kiu77 sorry, hast recht, hab mich falsch ausgedrückt

Gubi2023, 2024-03-27:
V0.8.101: sieht erst mal gut aus, History und Display config werden angezeigt

tavoc, 2024-03-28:
Die V.0.8.101 funktioniert auf dem ESP32. Menüpunkt Verlauf ist wieder da.

Auf einem ESP8266 fehlt es jedoch weiterhin:
[image]

wenn man history manuell eingibt, dann ist die Seite leer:

[image]

lumapu, 2024-03-28:
@tavoc welche ESP8266 Variante hast du installiert? Du benötigst die `esp8266-all-de`

tavoc, 2024-03-28:
klassisches Layer8 Problem, Dankeschön. Geht mit der all Version (hatte vorher immer nur die _de genommen)

kiu77, 2024-03-28:
Bei mir hat das Update auf 0.8.101 die Display-Konfiguration gelöscht.
Haben sich die Defaultwerte für das Display (Bildschirmschoner) verändert?

kiu77, 2024-03-28:
@MetaChuh 
Danke für die Erläuterung!

tavoc, 2024-03-29:
@lumapu mir fällt auf wie bereits @kiu77  festgestellt, dass die Zeiten doch nicht stimmen:
[image]

Es ist grade 15:50. Aber unten sind wir schon nach 16 Uhr.
Der obere Wert sieht auch eher wie 15:40 aus

Die NTP Zeit von Ahoy ist aber korrekt: 15:50

kiu77, 2024-03-29:
@tavoc 
Du hast nicht die aktuelle Firmware 0.8.101?
Bei .98 war ein Versatz von 1h der Fall. 
Ich habs gerade bei mir überprüft:
Mit der neuesten Beta ist der 1h-Versatz unten weg.

@lumapu 
Aber: Die untere Skala ist noch nicht wirklich genau, zeitlich. Jetzt geht sie geschätzt um 15-25 Minuten vor.
Das sieht man erst gegen Abend, wenn man genau drauf schaut.

[image]

tavoc, 2024-03-29:
Das Foto ist von der 101 Version, siehe unten rechts

kiu77, 2024-03-29:
Ach ja.
Dann ist der Fehler bei Dir größer. Vermutlich hängt er mit der Abfragerate zusammen, bei mir 5s.

Gubi2023, 2024-03-29:
ihr habt Probleme! das ist doch Pillepalle, er geht doch um die Tendenz!

kiu77, 2024-03-29:
@Gubi2023 
Na und? Dass es nicht das Wichtigste ist, dürfte klar sein, auch ohne Deinen Hinweis.

lumapu, 2024-03-29:
ich weiß, dass bei den Skalen irgendwas schief geht, beim oberen Grafen seit Anfang an.
Bei Gelegenheit nehme ich mich drum Thema an.

closed 2024-04-01 by tavoc