lumapu/ahoy #1536: Verbesserungsvorschlag max power Live Seite

mellow5, 2024-03-27:
Ich bin nicht 100% sicher, da ich mir nicht die Zeit genommen haben es nachzuvollziehen, aber ich vermute auf der Live Seite wird unter Total/max power einfach die Summe der max power der Umrichter ausgegeben, die ja aber nicht gleichzeitig aufgetreten sein müssen. Besser wäre hier die Tagesmaximum Leistung des Tageshistorygraphen anzugehen.

mellow5, 2024-03-28:
Bei den Graphen passt es meiner Meinung nach. Hier ein Screenshot meines aktuellen Tagesverlaufs.

[image]

Allerdings sollte im auf der "Live" Seite im Total Bereich auch dieses Maximum ausgegeben werden.

[image]

Beispiel der Umrichter 1 leistet 1000 W irgendwann am Vormittag und Umrichter 2 1000 W am Nachmittag. Dann werden auf der "Live" Seite 2000 W bei Maximum unter "Total" angegeben, wobei die beiden Maxima nicht gleichzeitig aufgetreten sind und das Totalmaximum der Erzeugung niedriger als 2000 W liegt und dem Maxima aus dem Tagesverlaufsgraphen entspricht.

0.8.100 und 0.8.101 identisch und vorherige Versionen auch.

kiu77, 2024-03-28:
Ah!
Du meinst also, die 1031,2W sind nicht gleichzeitig aufgetreten und unter Max Power sollte daher 954W aufgetragen sein, statt einfach die Summe der mehrerer Maximalwerte von mehreren Invertern?
Das scheint mir logisch, da nie ein Gesamtmaximum von 1031,2W aufgetreten ist.

Außerdem sollten die Einheiten in der zweiten Zeile unten auch nicht kleingeschrieben sein, denn Großschreibung ist die Definition dieser Einheiten.

lumapu, 2024-03-30:
Dieses Maximum ist nicht so einfach zu generieren, da man dafür alle Werte 'mitbekommen' muss. Das ist aber leider auf der `/live` Seite nicht so. Aktuell wird, wie ihr richtig geschrieben habt, einfach die Summe aller Maxima gebildert. Ich überlege, wie man das Gesamtmaximum elegant bekommen könnte - im aus meiner Sicht schlimmsten Fall müsste man es direkt im ESP erzeugen

kiu77, 2024-03-30:
Etwa so: 
Per Abfragerunde in Primivsprache, falls sowas nicht zuviel Platz/Zeit kostet:

```
daymax = 0
maxwert = 0
For i = 1 to #Inverter
   maxwert = malwert + AbfragekWh(i)
end i
if daymax < maxwert
   daymax = malwert
   updatedisplay(daymax)
end if

```

kiu77, 2024-04-01:
In der 0.8.101 stimmen die maximalen Leistungswerte auf der Live Seite für die einzelnen Inverter und auch für die Gesamtleistung nicht. Da wird schon morgens das Maximum der Inverter angezeigt bei "Max AC Leistung" bzw. "Maximale Leistung". Das stammt wohl von vor ein paar Tagen, als es  mal wirklich sonnig war. Heute war bis jetzt die Maximale "Gesamtleistung heute" im Verlauf bisher höchstens 1/4 dieses Werts.

lumapu, 2024-04-01:
> In der 0.8.101 stimmen die maximalen Leistungswerte auf der Live Seite für die einzelnen Inverter und auch für die Gesamtleistung nicht. Da wird schon morgens das Maximum der Inverter angezeigt bei [...]

dafür gibt es einen "Reset" in den Einstellungen

kiu77, 2024-04-01:
Ich hab den Fuß von der Leitung genommen. Danke fürs Schubsen @lumapu :-)
Das also:
[image]

mellow5, 2024-05-17:
Ist aus meiner Sicht jetzt richtig umgesetzt.

closed 2024-05-17 by mellow5