lumapu/ahoy #1540: DTU verliert verbindung  nur zu einem Wechselichter HM 1500

Oxbow6, 2024-03-29:
Hallo 
Ich habe ab der Firmware 0.8.96 - 2024-03-21 das Problem dass plötzlich immer der gleich Wechselrichter keine Daten mehr zum Ahoy senden kann.
Ich habe 7 Hoymiles Wechselreichter am laufen und immer die Nummer 6 funktioniert dann nicht mehr, 
Ich muss dann die Firmware 0.8.95 hochladen und den Ahoy stromlos machen dann läuft der Wechselrichter Nummer 6 auch wieder
Mit der Firmware 0.8.95 - 2024-03-17 funktioniert allles bestens. 

Gruss 

Martin

## Hardware
  * [ ] ESP8266
  * [x ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

knickohr, 2024-03-29:
Wie sind die Power-Einstellungen der WRs ?  Sind alle gleich weit entfernt oder gibt es lokale Unterschiede ?

Oxbow6, 2024-03-29:
Hallo
Lokale unterschiede gibt es nicht , alle 7 Wechselrichter sind gleich weit von der DTU entfernt.
Was meinst Du mit den Powereinstellungen ? 
Ich muss wenn ich dann wieder die Firmware 0.8.95 am laufen habe den Wechselrichter plus die Ahoy stromlos machen.

[image]

knickohr, 2024-03-29:
Es reicht schon wenn ein Inverter von eine Mauer oder Gebäude abgeschattet wird (Funk) oder anders ausgerichtet ist.

Jeder inverter kann individuell eingestellt werden.

[image]

Probier doch einfach mal den Probleminverter etwas aufzudrehen.

Oxbow6, 2024-03-29:
Die stehen bei mir alle auf LOW und Wechselrichter 5 und 6 sind am gleichen Ort montiert.
Mit der Firmware 0.8.95 funktioniert ja alles wieder und sobald ich höher gehe funktioniert es nicht mehr.
Er findet den Wechserichter aber er bekommt keine Daten mehr von dem.

knickohr, 2024-03-29:
Das hört sich aber verdächtig nach fehlendem Elko an wenn mehr Power nicht geht !

Dirk-Ahoy, 2024-03-29:
Jo und am Funkprotokoll wurde ja in den letzten Devs nix geändert.

Oxbow6, 2024-03-30:
Ok , ich werde heute mal meine DTU öffnen und schauen ob ein Kondensator montiert ist oder nicht.
Habe die mal fixfertig so gekauft,

Oxbow6, 2024-03-31:
Hallo 
So ich habe nun wieder die Version 0.8.101 am laufen und das Problem besteht auch mit Kondensator immer noch.
Habt Ihr eine Idee ? bin wieder zurück auf die Version 8.0.95
Warum sich der Wechselrichter 5 immer aufhängt bei der Kommunikation ?

LOG
17:13:31.223 I: (#0) RX  27ms | 27 CH61 | 95 01
17:13:31.225 I: (#0) RX  80ms | 27 CH61 | 95 02
17:13:31.226 I: (#0) RX 127ms | 27 CH40 | 95 03
17:13:31.227 I: (#0) RX 175ms | 27 CH40 | 95 84
17:13:31.229 -----
17:13:31.452 I: (#1) RX  27ms | 27 CH75 | 95 01
17:13:31.453 I: (#1) RX  75ms | 27 CH61 | 95 02
17:13:31.454 I: (#1) RX 125ms | 27 CH61 | 95 03
17:13:31.455 I: (#1) RX 175ms | 27 CH40 | 95 84
17:13:31.457 -----
17:13:32.092 I: (#2) RX  36ms | 27 CH75 | 95 01
17:13:32.094 I: (#2) RX  84ms | 27 CH61 | 95 02
17:13:32.094 I: (#2) RX 132ms | 27 CH61 | 95 03
17:13:32.095 I: (#2) RX 186ms | 27 CH61 | 95 84
17:13:32.098 -----
17:13:32.722 I: (#3) RX  31ms | 27 CH61 | 95 01
17:13:32.723 I: (#3) RX  79ms | 27 CH61 | 95 02
17:13:32.724 I: (#3) RX 133ms | 27 CH61 | 95 03
17:13:32.725 I: (#3) RX 176ms | 27 CH61 | 95 84
17:13:32.727 -----
17:13:32.968 I: (#4) RX  66ms | 27 CH40 | 95 01
17:13:32.969 I: (#4) RX 138ms | 27 CH23 | 95 03
17:13:32.970 I: (#4) RX 194ms | 27 CH23 | 95 84
17:13:33.331 I: (#4) RX  62ms | 27 CH40 | 95 02
17:13:33.334 -----
17:13:33.586 -----
17:13:33.824 I: (#5) RX 137ms | 27 CH23 | 95 03
17:13:33.825 I: (#5) RX 190ms | 27 CH23 | 95 84
17:13:34.068 I: (#5) RX 127ms | 27 CH23 | 95 03
17:13:34.070 I: (#5) RX 195ms | 27 CH40 | 95 84
17:13:34.472 I: (#5) RX 158ms | 27 CH23 | 95 03
17:13:34.720 I: (#5) RX 146ms | 27 CH23 | 95 03
17:13:34.721 I: (#5) RX 199ms | 27 CH23 | 95 84
17:13:34.951 I: (#5) RX  99ms | 27 CH40 | 95 02
17:13:34.952 I: (#5) RX 181ms | 27 CH23 | 95 84
17:13:34.953 I: (#5) timeout, no attempts left
17:13:34.954 -----
17:13:35.130 I: (#6) RX  35ms | 27 CH40 | 95 01
17:13:35.131 I: (#6) RX  82ms | 27 CH40 | 95 02
17:13:35.132 I: (#6) RX 129ms | 23 CH23 | 95 83
17:13:35.134 -----
17:13:35.135 I: com loop duration: 4134ms
17:13:35.136 -----
[image]

knickohr, 2024-03-31:
Schalt mal etwas mehr Logging an und laß das mal ein paar Minuten mitlaufen.

[image]

Oxbow6, 2024-03-31:
Hallo
Hier mal das Log von der Version 0.8.95 wo ich das Problem nicht habe.
Wede morgen dann mal die 101 Version aufspielen.
[Ahoylog.txt](https://github.com/lumapu/ahoy/files/14816909/Ahoylog.txt)

Dirk-Ahoy, 2024-04-01:
poste mal ein längeres Log der neueren Version ich vermute grenzwertigen Empfang.

Oxbow6, 2024-04-01:
Hallo
Hier mal ein 10 min Log der 0.8.101 Version.
Welche Statistik meinst Du denn ?

[Ahoylog0.8.101.txt](https://github.com/lumapu/ahoy/files/14823343/Ahoylog0.8.101.txt)

Oxbow6, 2024-04-01:
[image]

lumapu, 2024-04-01:
genau diese Statistik.
Das sieht doch perfekt aus, im Log sieht man die Statistik auch: 

`14:20:17.491 I: (#6) Radio infos: -2 -2 -2 -2 2 | t: 47, s: 45, f: 1, n: 1 | p: 0`

-> Transmit: 47, Success 45, Fail: 1

Dirk-Ahoy, 2024-04-01:
jo sieht gut aus - sollte gehen

knickohr, 2024-04-01:
Mal abgesehen das es um #5 geht, sieht das aber gut aus, ich kann da kein Problem feststellen. Einzig was auffällt, #5 hat Power Low, alle anderen min.

Oxbow6, 2024-04-01:
Hallo 
Ich hatte gestern alle 7 Wechselrichter mit Min und der Version 0.8.101 am laufen als sich der Wechelrichter #5 verabschiedet hatte. 
Habe nun diesen zum testen auf LOW gesetzt und warte ab ob er sich wieder aufhängt.
Wie gesagt bis zur Version 0.8.95 hatte ich auf allen 7 Wechselrichtern nie ein Problem

Dirk-Ahoy, 2024-04-01:
Im Log sieht eher der #4 nicht so gut aus.  Obwohl der ACKs sendet - wie gesagt am Funkprotokoll ist bei den letzen Versionen nichts geändert worden daher versteh ich dein Problem gerade nicht

Oxbow6, 2024-04-02:
Ich habe nun gestern zum testen die neue Version 0.8.102. aufgespielt und bis jetzt ist der Fehler nicht mehr aufgetreten beim WR5. Komisch.

[image]

Oxbow6, 2024-05-18:
Hallo 
Leider ist diese Thema immer noch aktuell bei mir.
Habe die Version 0.8.115 am laufen, bei dem funktionieren alle Wechselrichter. Wenn ich nun eine neuere Version aufspiele, habe ich wieder das gleiche Problem bei dem Wechelrichter Nr. 1
Ich denke irgendwas stimmt hier nicht.

lumapu, 2024-05-30:
hast du mal an deiner Antenne gedreht? Also die Position mal verändern, den Winkel und beobachten.

closed 2024-08-16 by lumapu