lumapu/ahoy #1542: NTP Server nicht erreichbar [Bug]

zappoXX, 2024-03-31:
### Platform

ESP32

### Assembly

I did the assebly by myself

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [X] I will attach/upload an Image of my wiring

### Version

0.8.101

### Github Hash

CC BY-NC-SA 4.0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

-

### Error description

Moin,
seit dem Update auf die Version 0.8.101 ist der NTP Server nicht mehr erreichbar. Auch der Zugriff auf z.B. die Einstellungen funktioniert dann nur noch sporadisch und endet meist mit einer Browsermeldung das die jeweilige Seite nicht geöffnet werden kann. Nach unzähligen Versuchen schaffe ich es dann ein Downgrade auf die Version 0.8.97 zu machen, welche übrigens hervorragend funktioniert und keinerlei Probleme mit dem eingetragenen NTP Server macht...

[image]
[image]
[image]

rmayergfx, 2024-03-31:
Welcher ntp Server ist denn eingetragen? Der eigene Router? Der ist die erste Wahl, da auch ohne Internet die Zeit synchronisiert wird.

zappoXX, 2024-03-31:
NTP Server ist der eigene Router… hat auch bisher einwandfrei funktioniert… erst mit 0.8.101 erhebliche Probleme… zurück auf 0.8.97 und alles läuft…

hrolofs, 2024-03-31:
Hab das gleichen Problem (opendtufusion board). 0.8.97 läuft einwandfrei, 0.8.100 & 0.8.101 haben das NTP Problem direkt nach dem flashen oder boot. Downgrade auf die 0.8.97 und alle läuft wieder fein.

rmayergfx, 2024-03-31:
Hier mit der Version 0.8.101 auf WROOM ESP32 mit der 240327_ahoy_0.8.101_2a4c836_esp32-wroom32-de.bin keinerlei Probleme. Sichert doch mal eure Settings und macht einen Werksreset vor dem Update auf 0.8.101 bzw. einen Clean Install.
NTP Settings:
IP der Box 192.168.x.x
Port 123
Intervall 720

Kann auch jederzeit auf "NTP SYNCHRONISIEREN" drücken und bekomme keinerlei Fehlermeldungen.

dtuuser, 2024-03-31:
Bei mir auch alles okay.  ESP32mini 101 in englisch.  NTP über lokale  Fritzbox

rmayergfx, 2024-03-31:
@zappoXX 
Wurden denn die empfohlenen Elkos verbaut, sehe oben in den Angaben leider nichts. Elko einlöten und Netzteil mal tauschen. Oft genug gibt es damit Probleme.
@hrolofs
Kenne das Board nicht, daher kann ich da leider nicht weiterhelfen.

hrolofs, 2024-03-31:
@rmayergfx 
https://github.com/markusdd/OpenDTUFusionDocs

Werde mir gleich was Zeit nehmen die DTU neu mit Standardsettings upzugraden.

reserve85, 2024-03-31:
Gleiches Problem bei mir, kontinuierliche reboots mit der .101. aus Zeitmangel hab ich nicht weiter danach geschaut, habe daher vorerst die 0.8.97 noch laufen.

hrolofs, 2024-03-31:
Gerade die DTU frisch aufgestzt mit  der 0.8.101

ahoy_v0.8.36\ESP32-S3\bootloader.bin
ahoy_v0.8.36\ESP32-S3\partitions.bin
ahoy_v0.8.36\ESP32-S3\ota.bin
240327_ahoy_0.8.101_2a4c836_opendtufusion.bin

Ich benutze bootloader, partition, ota aus der 0.8.36 
da in den aktuellen dev im ESP32-S3 Ordner keine vorhanden sind.

Der Accesspoint (192.168.4.1) ist für ein paar Sekunden zugänglich dann ist die Verbindung wieder weg.

DTU neu aufgesetzt mit der 0.8.97 und Settings manuell über den AP  eingetragen und alles läuft wieder

ahoy_v0.8.36\ESP32-S3\bootloader.bin
ahoy_v0.8.36\ESP32-S3\partitions.bin
ahoy_v0.8.36\ESP32-S3\ota.bin
240322_ahoy_0.8.97_9971ed7_opendtufusion.bin

Ollipop030, 2024-03-31:
Vielleicht nur ein Bug in der DE Version? Habe die .101 EN auf einem ESP32, keine Probleme bisher, NTP synchronisiert sofort.

hrolofs, 2024-03-31:
Das ist die ESP32-S3 Version für das opendtufusion board (english ohne ethernet.

sebastianrothe, 2024-04-01:
Habe das gleiche Problem. Zeit kann nicht vom NTP-Server geholt werden.
- 0.8.101 (EN, ALL)
- de.pool.ntp.org

rmayergfx, 2024-04-01:
Ich glaube das Problem kommt daher, das ihr hier eine URL eingebt statt der IP.
Wie ihr sicherlich in der Presse schon mitbekommen habt, hat sich jemand die Domain fritz.box registriert.
Siehe auch https://www.heise.de/news/Fritz-box-Domain-aus-dem-Verkehr-gezogen-9647776.html
Wenn nun euere FRITZ!Box oder andere Systeme versuchen fritz.box aufzulösen, kann es je nach DNS Einstellungen in der FRITZ!Box oder PiHole etc. dazu kommen, das immer noch die ungültige IP der externen Domain aufgelöst wird. Stellt das ganze mal auf IP um, dann sollte der Fehler weg sein!
@sebastianrothe 
Nimm als ntp Server direkt die IP der FRITZ!Box damit bist du immer auf der sicheren Seite, selbst wenn es gerade eine Störung beim ISP gibt. Solange der Router Strom hat, kann er auch im Hausnetz die Uhrzeit verteilen.

sebastianrothe, 2024-04-01:
> Ich glaube das Problem kommt daher, das ihr hier eine URL eingebt statt der IP. Wie ihr sicherlich in der Presse schon mitbekommen habt, hat sich jemand die Domain fritz.box registriert. Siehe auch [...]

Okay, probier ich mal. Ich hatte jedoch nicht nur `fritz.box` ausprobiert, sondern auch `de.pool.ntp.org` und die IP vom Router.

Mit v0.8.97 gibt es keine Probleme mit `de.pool.ntp.org`.

zappoXX, 2024-04-01:
@rmayergfx 
Nutze grundsätzlich nur die IP-Adresse meiner FRITZ!Box... Problem bleibt bestehen... aber auch jeder andere NTP Server funktioniert nicht... mehrere probiert!!!

sebastianrothe, 2024-04-01:
Ich bekomme mit allen Versionen `> 0.8.97` auch sehr viele Verbindungsabbrüche im WLAN. Ein Ping auf die DTU geht nur alle 10x durch. Mit der ´0.8.97´ geht es noch.

lumapu, 2024-04-01:
ja da habe ich massiv an den Netzwerkimplementierungen geschraubt - in der nächsten Version ist dieses Problem wieder Geschichte

sebastianrothe, 2024-04-01:
Mit `102` geht es, aber nur nach manuellem NTP Sync:

- WLAN immer noch instabil, wenn keine Zeit gesetzt ist
- Manueller Sync geht, dann geht auch WLAN stabil
- Wizard geht auf, `Stop`-Button zeigt auf die IP vom AP, obwohl schon eine IP vom Router da ist
- History zeigt keine Daten, aber Fehler in der Konsole:
```
Unexpected value NaN parsing y attribute. [api.js:71:20](http://192.168.100.46/api.js?v=0.8.102)
Unexpected value NaN parsing y1 attribute. [api.js:71:20](http://192.168.100.46/api.js?v=0.8.102)
Unexpected value NaN parsing y2 attribute. [api.js:71:20](http://192.168.100.46/api.js?v=0.8.102)
Unexpected value NaN parsing x1 attribute. [api.js:71:20](http://192.168.100.46/api.js?v=0.8.102)
Unexpected value NaN parsing x2 attribute. [api.js:71:20](http://192.168.100.46/api.js?v=0.8.102)
Uncaught ReferenceError: refresh is not defined
    parsePowerHistory http://192.168.100.46/history?v=0.8.102:184
    setTimeout handler*parsePowerHistory http://192.168.100.46/history?v=0.8.102:183
    p http://192.168.100.46/api.js?v=0.8.102:199
    getAjax http://192.168.100.46/api.js?v=0.8.102:190
    <anonymous> http://192.168.100.46/history?v=0.8.102:202
```
- Knopf `Sync NTP` wirft Fehler, klappt aber am Ende:
```
Uncaught SyntaxError: JSON.parse: unexpected end of data at line 1 column 1 of the JSON data
    p http://192.168.100.46/api.js?v=0.8.102:199
    getAjax http://192.168.100.46/api.js?v=0.8.102:190
    syncTime http://192.168.100.46/setup?v=0.8.102:466
    onclick http://192.168.100.46/setup?v=0.8.102:1
```

Weitere Fehler in der Konsole:
```
Uncaught SyntaxError: JSON.parse: unexpected end of data at line 1 column 1 of the JSON data
    p http://192.168.100.46/api.js?v=0.8.102:199
    getAjax http://192.168.100.46/api.js?v=0.8.102:190
    v http://192.168.100.46/:92
    setInterval handler* http://192.168.100.46/:92
```

lumapu, 2024-04-01:
danke für den Report, den muss ich Stück für Stück verdauen.
Der Stop Button kann nicht auf die entgültige IP zeigen, da man ja noch mit dem AP verbunden ist und nicht gleichzeitig mit dem konfigurierten WLAN verbunden ist.

Loetnase, 2024-04-02:
Siehe Kommentar in #1556 ganz unten,
geht bei mir mit 8.103 auch nicht

lumapu, 2024-04-02:
habe den PR gesehen, teste ich bei Gelegenheit und übernehme es, wenn auch ich es für gut befinde

Loetnase, 2024-04-02:
Bei mir funktioniert die NTP Zeit unter 0.8.103 wieder.
Ich hatte seit langer Zeit für die Fritzbox 
Port 0 
eingetragen, das hat bis 0.8.97 funktioniert. 
Jetzt  muss Port 123 eingetragen werden. 
So wie oben @rmayergfx schrieb.
NTP Settings:
IP der Box 192.168.x.x
Port 123
Intervall 720
Gibt es einen Grund weshalb 0 auf 123 geändert wurde?

lumapu, 2024-04-02:
der Port wurde nicht geändert, aber die Methode die Zeit abzufragen.
Port 123 ist richtig, 0 ging wohl wegen einem Fehler, der mir nie bekannt war

SebiWolze, 2024-04-02:
Selbes Problem mit der 0.8.97 und mit der letzten stable.
Sobald ich heimisches WLAN eintrage und sage "save" fährt er noch bis zu diesem Punkt hoch, danach nicht mehr erreichbar.
Ubiquiti Controller und Fritzbox sehen die DTU noch, aber kein Ping geht mehr durch.

[image]

Loetnase, 2024-04-03:
Dies jetzt nur zur Info und Verständnis an @lumapu!

Bei falsch eingestellten NPT Port 0 anstatt 123 in Version 0.8.103
[image]
Bekomme ich diese 3 Meldung, mit keine Kommunikation.....
[image]
aber die Livedaten kommen interesanter weise in Grau
[image]
Die Kommunikation zum Umrichter ist wohl doch vorhanden :-)

zappoXX, 2024-04-03:
@Loetnase 
Ja hab ich... aber auch jeder andere NTP Server hat das gleiche Problem... bin jetzt wieder bei 0.8.97... da funktioniert alles einwandfrei...

zappoXX, 2024-04-07:
Heute mal die neueste Version installiert... auch in Version 0.8.106 NTP Problem... zurück nach 0.8.97...

[image]

Loetnase, 2024-04-07:
Du bekommst wohl keine Zeit vom NPT Server.
Probier mal SYNC VOM BROUSER ob du damit eine Zeit und dann Komunikation bekommst.
Ansonste zeig mal deine Einstellungen vom NTP Server

zappoXX, 2024-04-07:
@Loetnase 
Screenshot aus Version 0.8.97... 
Version 0.8.98 - 0.8.106 funktioniert nicht mehr...

[image]

Loetnase, 2024-04-07:
Genau das  würde mich von der 8.106 interessieren wo es nicht tut. eventuell die 123 neu eingeben vielleicht hat sich ein Leerzeichen versteckt. 
Und mal Synch vom Browser probieren ob da die Zeit genommen wird.
Bei mir läuft die 106

zappoXX, 2024-04-07:
@Loetnase 
Nach Installation der 0.8.106 ist alles leer, also neu eingeben und Reboot... dann NTP Daten kurz da und dann wieder weg...
Komischerweise keinerlei Probleme mit 0.8.97... da läuft alles wie es soll, erst nach dieser Version Probleme mit dem NTP-Server!!
[image]

Loetnase, 2024-04-07:
Komisch, kannst du von der 106 eine json exportieren und schauen ob da was sinnvolles oder Mist drinsteht?
Ansonsten mit der 106 einen Werksreset machen dann Spannung weg und nach neustart eine Sicherung der 8.97 drüber bügeln

zappoXX, 2024-04-07:
@Loetnase 
Klicke ich in Version 0.8.106 bei leeren Feldern im Bereich NTP Server auf NTP Synchronisieren dann funktioniert plötzlich alles... aber nur bis zu einem Reboot... dann wieder nur Probleme... NTP Zeitserver nicht erreichbar. Werde heute Abend mal deinen Tipp mit dem Werksreset probieren... Aber schon mal Danke...!!!

hrolofs, 2024-04-07:
Schau auch mal ob du "System Config/Reboot Ahoy at midnight" aktiviert hast und schalte das gegebenefalls aus. Bei mir hat das all die NTP/Reboot Probleme  bei den 0.8.1XX Versionen wohl beeinflusst. Hab das bei meinen DTUs vor dem updaten von 0.8.97 auf die 0.8.106 ausgeschaltet und das update lief sauber durch.

Loetnase, 2024-04-07:
könnte ein Grund sein das es bei mir mit 106 läuft
[image]

zappoXX, 2024-04-07:
Das war's... Haken hatte ich drin... jetzt funktionierts...!! Muss man erst mal drauf kommen!!!

sebastianrothe, 2024-04-07:
Denkt die DTU dann, es ist Mitternacht und startet deswegen neu.

hrolofs, 2024-04-07:
Denke der "Mitternachtscheck" / Zeitvergleich liefert immer true oder sowas. Werden wir in den nächsten Patchnotes lesen. ;-)

hrolofs, 2024-04-07:
Denke bei den meisten die aktuell mit der 0.8.1XX haben könnten das die Ursache sein.

Loetnase, 2024-04-07:
Und deshalb gibt es Hakenlose wie mich die keine Probleme haben.
Wie ist Mitternacht definiert als 00:00 Uhr? 
Somit nach dem Boot, noch ohne Zeit, steht 00:00 Uhr drin und dann wird wieder neu gebootet usw.
Nur so als Fehleridee.

sebastianrothe, 2024-04-07:
Es klingt so. Wenn heute Abend etwas Zeit ist, schaue ich mal in den Code.

sebastianrothe, 2024-04-07:
Die Stelle in https://github.com/lumapu/ahoy/blob/435aa1748af556c914e1d8f1e6043dee8566f1a1/src/app.cpp#L224 wurde in letzter Zeit nicht geändert. 

Mir ist noch aufgefallen, dass durch die Zeitumstellung am 31.3. meine Zeitzone falsch war (richtig ist jetzt mit Sommerzeit `+2`). Vielleicht hatte das einen Seiteneffekt. Wenn ich es zurückstelle, scheint es mit der `106` aber auch zu gehen. 🤷

closed 2024-04-15 by zappoXX