lumapu/ahoy #1544: 0.8.83  PowerLimit setzen im Inrterface

welano2, 2024-03-31:
Hallo, 

ich habe gerade meine DTU von Version 0.6.9 auf 0.8.83 (ESP8266) geupdated. Das Ganze läuft jetzt super stabil aber mir fehlen jetzt die Möglichkeit das PowerLimit für meinen Wechselrichter zu setzen. Auch per API kommt ich nicht mit an /api/ctrl 

Muss ich noch irgendetwas nach dem Update aktivieren? Gibt es eigentlich eine API Doku irgendwo?

Danke und LG

dtuuser, 2024-03-31:
Unter der Live Ansicht auf APC klicken

Gubi2023, 2024-03-31:
Muss ich noch irgendetwas nach dem Update aktivieren? Gibt es eigentlich eine API Doku irgendwo?
https://ahoydtu.de/screenshots/

lumapu, 2024-04-01:
leider nicht konsistent, aber am vollständigsten hier: 
https://github.com/lumapu/ahoy/blob/main/manual/User_Manual.md#control-via-rest-api

hawei13, 2024-04-23:
Vielen Dank! Ich finde die neue Option das Power Limit zu setzen jetzt zwar deutlich gelungen, aber es ist dennoch nicht wirklich intuitiv zu finden. Ich habe min 30 min danach gesucht und nicht gefunden und bin jertzt erst durch das Forum auf die Option aufmerksam geworden.
Mein Vorschlag wäre, entweder den Text als Link zu kennzeichnen (z.B. durch Unterstreichung) dort ein Icon zu ergänzen, das symbolisiert, dass man das Limit hier bearbeiten kann.

Gubi2023, 2024-04-23:
> Vielen Dank! Ich finde die neue Option das Power Limit zu setzen jetzt zwar deutlich gelungen, aber es ist dennoch nicht wirklich intuitiv zu finden. Ich habe min 30 min danach gesucht und nicht [...]
Es gibt eine Doku:
https://ahoydtu.de/screenshots/ 
Da stehts drin

hawei13, 2024-04-23:
Die Seite ist sehr hilfreich, danke! Sie ist wirklich gut gemacht.
Ich habe sie aber auch nicht gesehen, da ich unter "Screenshots" keine Anleitung erwartet habe.
Vielleicht könnte man den Link ja in "Instructions for WebUI" oder so umbenennen?
In der Regel findet man unter Screenshots ja eher ein paar Bildchen für Leute, die das System noch nicht kennen. 
Ich nutze AhoyDTU aber schon über ein Jahr und fand den Link daher irrelevant.
Also gleich als weiterer Verbesserungsvorschlag. 
(Ich bin selbst Entwickler und durchaus IT affin, ich denke, andere könnten ähnliche Probleme haben...) :-)

Gubi2023, 2024-04-23:
> Die Seite ist sehr hilfreich, danke! Sie ist wirklich gut gemacht. Ich habe sie aber auch nicht gesehen, da ich unter "Screenshots" keine Anleitung erwartet habe. Vielleicht könnte man den Link ja [...]
@lumapu

stefan123t, 2024-10-30:
@ghost @MetaChuh ist hier noch etwas zu tun ?

Der Abschnitt [Control via REST API](https://github.com/lumapu/ahoy/blob/main/manual/User_Manual.md#control-via-rest-api) im UserManual.md sollte doch gut zu finden sein oder brauchen wir noch einen FAQ Eintrag ?