lumapu/ahoy #1546: Grid - Profil lesen HMS2000 Ahoy  0.8.101 DEV Meldung : Unbekanntes Profil : Bitte erstelle einen neuen Issue ....

rolandl1969, 2024-04-01:
Meldung beim Gridprofil auslesen :

Unbekanntes Profil
Bitte erstelle einen neuen Issue auf https://github.com/lumapu/ahoy und kopiere die Rohdaten hinein. 

00 01 03 E8 00 00 03 E8 00 00 00 00 00 00

lumapu, 2024-04-01:
das ist aber ein sehr kurzes Grid Profile

rolandl1969, 2024-04-01:
[image]

Das ist die komplette Meldung. Der gleichzeitig betriebene HM1500 kann problemlos ausgelesen werden.

lumapu, 2024-04-01:
ich finde leider inm iner Sammlung kein Gridprofile mit dieser ID (0x0001). Evtl. bedeutet das, das kein Gridprofile gespeichert ist.

rolandl1969, 2024-04-02:
Mit 0.8.102 klappt das Auslesen wieder :-)
[image]

closed 2024-04-02 by rolandl1969

lumapu, 2024-04-02:
eigenartig, dass es den Aussetzer gab, danke für's schließen 👍