lumapu/ahoy #1547: Antennen für DTU aus China

Zurrmaxe, 2024-04-01:
Hallo zusammen

Wie ich feststellen musste, werden bei den DTU-Bausätzen oft billige Antennen aus China mitgeliefert.  
Diese sind leider **nicht für alle Anwendungen** geeignet. Man kann sie jedoch ganz einfach öffnen und sich das Innenleben einmal anschauen.

Dazu etwas Hintergrundinformation für den Laien:

Die Antennenlänge richtet sich nach der jeweiligen Frequenz.  Je höher die Frequenz, desto kürzer die Antenne. 
Die Antennenlänge beträgt immer ein Vielfaches oder Teiler ( 1./2./4.) der jeweiligen Wellenlänge.
Beispiel:  Bei 150 Mhz beträgt die Wellenlänge 2 Meter (300/150 = 2). Ein Lamda/Viertel strahler somit 50 cm.
Bei 2,4 GHZ sind es etwa 3,1 cm und bei 865 Mhz(CMT2300A) noch 8,6 cm.(ungefähr). 

Betrachtet man nun die mitglieferten China-Antennen so stellt man fest, dass diese nur für 2400 MHZ (WLAN pp) ausgelegt sind. **Beim CMT2300A funktionieren sie nicht !** (Oder nur ein paar Meter !)

Für das CMT2300A-Moul  diese Antennen daher zu verlängern.(siehe Anleitung).

Abgleich geht wie folgt :
Wenn man AHOY benutzt kann man in der Webkonsole die Feldstärke in dB ablesen. 
**Achtung ! Es ist ein Minus-Wert ! Je weniger, desto besser !**
Die Antenne etwas länger lassen und nach Bedarf in kleinen Stücken (5mm) abschneiden bis der beste Wert gefunden wurde. 
Falls man sich verschnitten hat, einfach einen neuen Draht anlöten. 
Eine alte Kunststoffmine aus einem Kugelschreiber wird zur Verstärkung über den dünnen Draht geschoben  und hat zufällig auch etwa die richtige Länge.

mfg.
Alfred
[image]

lumapu, 2024-04-01:
sehr cool 👍

kiu77, 2024-04-01:
Man kann es sich auch einfach machen und die richtigen Antennen in China bestellen. Ich habe z.B. die genommen und die funktionieren gut:
[https://de.aliexpress.com/item/33007602630.html?](url)

kiu77, 2024-04-01:
Eine weitere Möglichkeit besteht in der **induktiven Verlängerung** der Antenne. Hierbei wird direkt an Antennenfuß eine keine "Spule" in Form einiger Windung Draht angelötet und dann erst das dann kürzere Stück Draht.
In bin kurz überfragt, wieviele Windungen mit welchem Durchmesser geeignet wären, damit der komplette Draht in das obere Antennegehäuse passt. Man kann das experimentell herausfinden oder auch berechnen (wenn ich wüsste, wo man die Vorschrift finden). Bei professionellen Antennen wird das oft so gemacht. So z.B. kommen sind die Stummelantennen bei Autos aufgebaut.

Wenn man hier den originalen Draht von 30mm lassen würde, dann würde ich nach Bauchgefühl etwa Kupferackdraht mit 0,5 bis 0,6mm ø auf eine Kugelschreibermine wickeln (und die hinterher wieder herausziehen) und mal mit 15 Windungen anfangen. Wer experimentieren will, kann es mit unterschiedlichen Windungszahlen probieren und sich die RSSI-Werte aufschreiben.

geronet1, 2024-04-01:
> Achtung ! Es ist ein Minus-Wert ! Je weniger, desto besser !

Moment mal, ein höherer Wert ist doch besser?
Beispiel: -40 dBm sind doch besser als -80 dBm ?

Zurrmaxe, 2024-04-01:
Diese Info war für den Laien gedacht., welche sich die Konsole anschaut und beim Abgleichen nur die Messwerte anschaut.
Der Profi kennt natürlich die Hintergründe und weiß , das weniger mehr ist......

Zurrmaxe, 2024-04-01:
> Eine weitere Möglichkeit besteht in der **induktiven Verlängerung** der Antenne......  <

Ja sicher, man kann eine Antenne durch Spulen und Kapazitäten verlängern oder verkürzen. 
Praktisch benötigt man dazu die genauen Angaben zu Fußpunktwiderstand der Antenne, den Anschlusswerten an dem Modul (50 Ohm?) pp.. sowie ein entsprehendes Stehwellenmesserät bzw Spektrumanalyser.

Das überschreitet jedoch die Möglichkeiten der meisten User. Solche komplizierten Antennen sind nur etwas für Profis.
Für Laien geht am besten Lambda/Viertel und dann die Länge abschneiden. 

Hier mal ein Beispiel eine verlängerten Antenne:
[image]

DB1BMN-collection, 2024-04-06:
Cool, hat schon mal jemand diese Antenen mit dem nanoVNA durchgesweept?

kiu77, 2024-04-06:
@Zurrmaxe 
Im Prinzip hast Du vollständig Recht. In der Praxis kann man aber auch einfach ausprobieren und schauen, was es für Resultate gibt.

Zurrmaxe, 2024-04-06:
Dazu müsste man die genauen Daten vom Modul und den Anschlussleitungen wissen.Sind es 50 Ohm ? Oder irgendwas anders? Daher  "cut and error"

kiu77, 2024-04-06:
@Zurrmaxe 
Nein, beim Ausprobieren muss man das nicht genau wissen. Man geht einfach vom Gegebenen aus und kann es mit verschiedenen Spulen testen und die RSSI-Werte als Kriterium nehmen. Irgendwo gibt es ein Optimum. Und viel mehr holt man auch durch Berechnung nicht raus.
Als Anfang muss man mal schauen, welche Drahtlänge in die obere Hülle passt. Geschätzt sind es 5-6cm. Einen cm für die Spule abziehen und den Draht so lang wie möglich lassen.
Ich hab noch eine fertige 869-MHz-Antenne. Ich schau mal nach, ob sie sich öffnen lässt und mache ggf. ein Foto davon.

Zurrmaxe, 2024-04-06:
In den Gehäusen passt l/4 nicht hinein. Daher hatte ich auch den Draht mit oben herauskommen  lassen.Durch Anbringen einer Spule unten oder Dachkapazität kann man die Länge natürlich beeinflussen. Üblicherweise werden Verkürzungsspulen in unmittelbarer Nähe des Fusspunktes eingesetzt. Bei diesen Stummel wurde über dem Draht noch eine Hülse geschoben und mit der Abschirmung verlötet.Diese war als Gegengewicht zum 3 cm Draht gedacht. Bei einer Verlängerung des Strahlers müsste daher auch dieses "Gegengewicht" angepasst werden. Eigentlich.Dazu bedarf es entweder umfangreiche Messungen  oder die übliche Methode des Abschneidens.Wer Zeit hat, kann es gerne versuchen.

kiu77, 2024-04-06:
Ich hab nochmal geschaut: Meine 868-MHz-Antennen lassen sich leider nicht ohne Zerstörung öffnen. Daher leider kein Foto vom Aufbau. Es wäre interessant gewesen, da sie genau so lang sind wie meine 2,4-GHz-Exemplare. Also müssen sie induktiv verlängert sein.

Zurrmaxe, 2024-04-07:
Da diese Antennen oftmals nicht auf Metallgehäuse montiert werden, fehlt denen das erforderlichen "Gegengewicht" um als l/4   zu funktionieren.Daher die Konstruktion, mit der Messinghülle, an die der Aussenleiter gelötet wurde.Somit wäre es eher ein Dipol, welcher kein Gegengewicht benötigt.Mfg.

kiu77, 2024-04-07:
Auf die Gefahr hin, HF-Fachsimpelei zu produzieren (und ich bin kein HF-Spezialist, aber Elektroniker mit Jahrzehnten Erfahrung):
Das sind Prinzipien, Lehrsätze etc. 
In der Praxis reicht keine kleine Messinghülle als "Gegengewicht" aus, denn sie müsste ja auch etwa λ/4 oder mehr an Länge haben.
Doch auf der anderen Seite des eigentlichen Strahlers ist ja auch ohne Metallgehäuse nicht nichts. Da ist die Abschirmung des Kabels zur CMT2300A-Platine, und diese Abschirmung ist (zumindestens bei mir) lang genug, um als "Gegengewicht" zu dienen und der Antenne eine Charakteristik von irgendwas zwischen unipolar und bipolar zu geben. Jedenfalls genug, um ausreichend gut zu funktionieren.