lumapu/ahoy #1553: CMT Inverter nicht erreichbar [Bug]

geronet1, 2024-04-01:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.93 ETH

### Github Hash

f315fb

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

ETH Version mit W5500 und CMT

### Debug Serial Log output

```shell
0.8.93 - f315fb1
---------------
18:54:52.002 I: (#0) Radio infos: t: 4, s: 4, f: 0, n: 0 | p: 0
18:54:52.003 I: (#0) TX 863.00Mhz | 15 00 80
18:54:52.135 I: (#0) RX  79ms | 27 -41dBm | 95 81
18:54:52.136 I: (#0) Payload (14)
18:54:52.139 -----
18:54:52.140 I: (#0) Radio infos: t: 5, s: 5, f: 0, n: 0 | p: 0
18:54:52.141 I: (#0) TX 863.00Mhz | 15 0b 80
18:54:52.466 I: (#0) RX  82ms | 27 -41dBm | 95 01
18:54:52.467 I: (#0) RX 132ms | 27 -41dBm | 95 02
18:54:52.468 I: (#0) RX 182ms | 27 -41dBm | 95 03
18:54:52.469 I: (#0) RX 232ms | 27 -41dBm | 95 04
18:54:52.470 I: (#0) RX 273ms | 15 -41dBm | 95 85
18:54:52.482 I: (#0) Payload (66)
18:54:52.484 -----
18:54:52.485 I: (#1) Radio infos: t: 4, s: 0, f: 0, n: 4 | p: 0
18:54:52.486 I: (#1) TX 863.00Mhz | 15 05 80
18:54:52.626 I: (#1) request timeout: 135ms
18:54:52.626 I: (#1) TX 868.00Mhz | 56 15 02
18:54:52.630 -----
18:54:52.631 I: (#1) Radio infos: t: 5, s: 0, f: 0, n: 5 | p: 0
18:54:52.633 I: (#1) TX 863.00Mhz | 15 0b 80
18:54:52.634 W: CMT TX failed, code: 2
18:54:53.069 I: (#1) request timeout: 436ms
18:54:53.070 I: (#1) TX 863.00Mhz | 56 15 02
18:54:53.074 -----
18:54:53.075 I: (#10) Radio infos: t: 4, s: 0, f: 0, n: 4 | p: 0
18:54:53.077 I: (#10) TX 863.00Mhz | 15 05 80
18:54:53.078 W: CMT TX failed, code: 2
18:54:53.213 I: (#10) request timeout: 135ms
18:54:53.213 I: (#10) TX 868.00Mhz | 56 15 02
18:54:53.217 -----
18:54:53.218 I: (#10) Radio infos: t: 5, s: 0, f: 0, n: 5 | p: 0
18:54:53.220 I: (#10) TX 863.00Mhz | 15 0b 80
18:54:53.221 W: CMT TX failed, code: 2
18:54:53.506 I: (#10) request timeout: 285ms
18:54:53.506 I: (#10) TX 863.00Mhz | 56 15 02
18:54:53.510 -----
18:54:53.511 I: (#11) Radio infos: t: 4, s: 2, f: 0, n: 2 | p: 0
18:54:53.513 I: (#11) TX 863.00Mhz | 15 05 80
18:54:53.514 W: CMT TX failed, code: 2
18:54:53.649 W: (#11) frame 1 missing: request retransmit (3 attempts left)
18:54:53.650 I: (#11) TX 863.00Mhz | 15 d5 81
18:54:53.788 I: (#11) RX  71ms | 27 -42dBm | 95 01
18:54:53.789 W: (#11) frame 2 missing: request retransmit (2 attempts left)
18:54:53.790 I: (#11) TX 863.00Mhz | 15 d6 82
18:54:53.929 I: (#11) RX  70ms | 27 -42dBm | 95 02
18:54:53.931 W: (#11) frame 3 missing: request retransmit (1 attempts left)
18:54:53.932 I: (#11) TX 863.00Mhz | 15 d7 83
18:54:54.070 I: (#11) RX  68ms | 27 -42dBm | 95 03
18:54:54.072 W: (#11) frame 4 missing: request retransmit (0 attempts left)
18:54:54.073 I: (#11) TX 863.00Mhz | 15 d0 84
18:54:54.211 I: (#11) RX  68ms | 27 -42dBm | 95 04
18:54:54.212 I: (#11) timeout, no attempts left
18:54:54.213 -----
18:54:54.214 I: (#11) Radio infos: t: 5, s: 2, f: 0, n: 3 | p: 0
18:54:54.215 I: (#11) TX 863.00Mhz | 15 0b 80
18:54:54.537 I: (#11) RX  80ms | 27 -42dBm | 95 01
18:54:54.538 I: (#11) RX 130ms | 27 -42dBm | 95 02
18:54:54.539 I: (#11) RX 180ms | 27 -42dBm | 95 03
18:54:54.540 I: (#11) RX 230ms | 27 -42dBm | 95 04
18:54:54.541 I: (#11) RX 270ms | 15 -42dBm | 95 85
18:54:54.553 I: (#11) Payload (66)
18:54:54.556 -----
18:54:54.557 I: com loop duration: 2556ms
18:54:54.557 -----
18:55:02.001 I: (#0) Radio infos: t: 6, s: 6, f: 0, n: 0 | p: 0
18:55:02.003 I: (#0) TX 863.00Mhz | 15 11 80
18:55:02.182 I: (#0) RX  81ms | 27 -41dBm | 95 01
18:55:02.183 I: (#0) RX 127ms | 23 -41dBm | 95 82
18:55:02.184 I: (#0) Payload (26)
18:55:02.187 -----
18:55:02.188 I: (#0) Radio infos: t: 7, s: 7, f: 0, n: 0 | p: 0
18:55:02.189 I: (#0) TX 863.00Mhz | 15 0b 80
18:55:02.511 I: (#0) RX  80ms | 27 -41dBm | 95 01
18:55:02.512 I: (#0) RX 130ms | 27 -41dBm | 95 02
18:55:02.513 I: (#0) RX 180ms | 27 -41dBm | 95 03
18:55:02.514 I: (#0) RX 230ms | 27 -41dBm | 95 04
18:55:02.515 I: (#0) RX 270ms | 15 -41dBm | 95 85
18:55:02.527 I: (#0) Payload (66)
18:55:02.529 -----
18:55:02.530 I: (#1) Radio infos: t: 6, s: 0, f: 0, n: 6 | p: 0
18:55:02.532 I: (#1) TX 863.00Mhz | 15 05 80
18:55:02.671 I: (#1) request timeout: 135ms
18:55:02.671 I: (#1) TX 868.00Mhz | 56 15 02
18:55:02.676 -----
18:55:02.676 I: (#1) Radio infos: t: 7, s: 0, f: 0, n: 7 | p: 0
18:55:02.678 I: (#1) TX 863.00Mhz | 15 0b 80
18:55:02.679 W: CMT TX failed, code: 2
18:55:03.114 I: (#1) request timeout: 436ms
18:55:03.115 I: (#1) TX 863.00Mhz | 56 15 02
18:55:03.120 -----
18:55:03.120 I: (#10) Radio infos: t: 6, s: 0, f: 0, n: 6 | p: 0
18:55:03.122 I: (#10) TX 863.00Mhz | 15 05 80
18:55:03.123 W: CMT TX failed, code: 2
18:55:03.258 I: (#10) request timeout: 135ms
18:55:03.258 I: (#10) TX 868.00Mhz | 56 15 02
18:55:03.262 -----
18:55:03.263 I: (#10) Radio infos: t: 7, s: 0, f: 0, n: 7 | p: 0
18:55:03.265 I: (#10) TX 863.00Mhz | 15 0b 80
18:55:03.266 W: CMT TX failed, code: 2
18:55:03.551 I: (#10) request timeout: 285ms
18:55:03.551 I: (#10) TX 863.00Mhz | 56 15 02
18:55:03.555 -----
18:55:03.556 I: (#11) Radio infos: t: 6, s: 3, f: 0, n: 3 | p: 0
18:55:03.558 I: (#11) TX 863.00Mhz | 15 05 80
18:55:03.559 W: CMT TX failed, code: 2
18:55:03.694 W: (#11) frame 1 missing: request retransmit (3 attempts left)
18:55:03.695 I: (#11) TX 863.00Mhz | 15 d5 81
18:55:03.833 I: (#11) RX  72ms | 27 -42dBm | 95 01
18:55:03.834 W: (#11) frame 2 missing: request retransmit (2 attempts left)
18:55:03.835 I: (#11) TX 863.00Mhz | 15 d6 82
18:55:03.974 I: (#11) RX  70ms | 27 -42dBm | 95 02
18:55:03.976 W: (#11) frame 3 missing: request retransmit (1 attempts left)
18:55:03.976 I: (#11) TX 863.00Mhz | 15 d7 83
```

### Error description

14 Inverter eingetragen, sobald ich einen enable der nicht verfügbar ist (nicht angeschlossen) wird der nächste "enabelte" in der Liste auch nicht mehr erreichbar obwohl vorhanden
Deaktiviere ich den nicht verfügbaren geht der nächste wieder..
alles HMS bzw. HMT mit dem CMT Modul
bei einem anderem NRF System ist das nicht der Fall
#0 ist erreichbar, #1 nicht angeschlossen, #10 wird nicht erreicht obwohl vorhanden und erreichbar, #11 ist auch erreichbar

lumapu, 2024-04-01:
das ist der gleiche Fehler wie der, den du in deinem PR behebst, nehme ich an. Auch hier wird der JSON Buffer zu klein sein, alles zu speichern.
Vorzugsweise sollte man die Größe des Buffer anhand der Anzahl der konfigurierten Weichselrichter wählen und nicht mehr statisch.

geronet1, 2024-04-01:
Leider nicht, der Puffer ist bei mir groß genug ansonsten könnte ich die Inverter gar nicht konfigurieren. Irgendwo bei der Empfangsroutine ist etwas daß er den request timeout zum nächsten verschleppt. Wird warscheinlich nicht auffallen solange alle erreichbar sind, kann ich erst die Woche testen.

lumapu, 2024-04-01:
ok, ich überlege mal kann mir aber nur sowas vorstellen. Hast du mal geprüft wie der Export der settings aussieht? Evtl. gibt uns das Hinweise.

geronet1, 2024-04-01:
Sieht bestens aus, kann ich dir ja mal schicken bei Bedarf. Demnächst krieg ich wieder ein CMT Modul dann kann ich weiter testen und such seriell debuggen.

lumapu, 2024-04-05:
ich habe eine Idee und hoffentlich wird die 0.8.106 diesen Fehler beheben

lumapu, 2024-04-08:
konntest du testen, wäre interessant, ob mein fix wirkt?

geronet1, 2024-04-10:
Getestet mit der 0.8.108:
Alle drei blinken, nur der erste geht:
`I: (#0) Radio infos: t: 11, s: 11, f: 0, n: 0 | p: 0
I: (#0) TX 863.00Mhz | 15 0B 80
I: (#0) RX  83ms | 27 -14dBm | 95 01
I: (#0) RX 133ms | 27 -14dBm | 95 02
I: (#0) RX 183ms | 27 -14dBm | 95 03
I: (#0) RX 233ms | 27 -14dBm | 95 04
I: (#0) RX 274ms | 15 -14dBm | 95 85
I: (#0) Payload (66)
-----
I: (#1) Radio infos: t: 12, s: 0, f: 0, n: 12 | p: 0
I: (#1) TX 863.00Mhz | 15 05 80
I: (#1) request timeout: 135ms
I: (#1) TX 868.00Mhz | 56 15 02
-----
I: (#1) Radio infos: t: 13, s: 0, f: 0, n: 13 | p: 0
I: (#1) TX 863.00Mhz | 15 0B 80
I: (#1) request timeout: 285ms
I: (#1) TX 868.00Mhz | 56 15 02
-----
I: (#2) Radio infos: t: 12, s: 0, f: 0, n: 12 | p: 0
I: (#2) TX 863.00Mhz | 15 05 80
I: (#2) request timeout: 135ms
I: (#2) TX 868.00Mhz | 56 15 02
-----
I: (#2) Radio infos: t: 13, s: 0, f: 0, n: 13 | p: 0
I: (#2) TX 863.00Mhz | 15 0B 80
I: (#2) request timeout: 285ms
I: (#2) TX 868.00Mhz | 56 15 02
-----
I: com loop duration: 1156ms`

lumapu, 2024-04-11:
ich muss den fix zurücknehmen, da auch bei mir jetzt Inverter aussteigen, ein Downgrade auf eine andere Version hat sofort wieder beide online gebracht.

geronet1, 2024-04-12:
Gibt es irgendwo eine Pausenzeit zwischen den Invertern? Evtl. liegt es ja daran, diese vllt. mal testweise erhöhen.

Wenn du zwei HMS hast (von mir haste ja einen gekauft) kannst du den ersten DC-Seitig abstecken? Dann wird auch der zweite nicht mehr antworten.

geronet1, 2024-04-18:
Hab gerade noch ein wenig herumprobiert: Wenn alle drei Inverter erreichbar sind und man die Frequenz vom ersten ändert ist dieser und der zweite nicht mehr erreichbar. Würde ja gerne den Fehler suchen aber bei der Funkkommunikation steig ich nicht durch..

lumapu, 2024-05-30:
dem Problem sollten wir uns nochmal annehmen. Habe weiterhin noch keine Idee was genau passiert.

stefan123t, 2024-10-31:
@lumapu wir haben dieses bzw. ein ähnliches Problem auch seit einiger Zeit auf der OpenDTU.
Offenbar hängt es auch ein wenig von der Firmware Version des Inverters ab, aber auch die Hoymiles DTU Pro S verwendet evtl. noch andere Kommandos.
Hier mal ein Beispiel Trace aus einer DTU Pro S mit jeweils fünf Channel Change Commands und einem Real Time Run Data Debug.
Ich habe es auch noch nicht ganz durchstiegen, weil ja die Commands zwischen ST32 MCU und dem CMT2300A mitgeschnitten werden, da gibt es ja noch das Hoymiles SPI Wire-Protocol :wink.

Hier die ChannelChangeCommands der Original Hoymiles DTU Pro S:
https://github.com/tbnobody/OpenDTU/issues/2137#issuecomment-2427788809