lumapu/ahoy #1558: Max. Werte durcheinander mit 2xHMS1600

slg60, 2024-04-02:
## Hardware
  * [ ] ESP8266
  * [x] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.8.101
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Es sieht so aus, als ob bei den MAX Werten die Daten der beiden Wechselrichter durcheinander kommen.
Exakt gleiche Max Daten bei unterschiedlichen Modulen sind verdächtig, zumal die PV-Stall nur 410W Module hat, somit nicht auf diese Werte kommen sollte. Die MQTT Daten sehen realistischer aus.

[image]

lumapu, 2024-04-02:
das ist sehr eigenartig und kann ich nicht bestätigen, bei mir sind die max Werte alle unterschiedlich, habe 3 Wechselrichter und die 0.8.103 installiert

Loetnase, 2024-04-02:
Die 429,5 Watt Max Power kann auch das Maximum der Inverterkanäle sein. 
Bei kühlen Temperaturen guter Sonne und perfektem Einfallswinkel kann ein 410 Wp Modul auch mehr wie 410 Watt kurzzeitig bringen. Vielleicht sind wirklich alle 8 Kanäle kurzzeitig am Limit gehängt. 
Must du mal beobachten

slg60, 2024-04-02:
Ich habe die 410er Module schon 1 Jahr lang an einem HM 1500 mit einer 8266 Dtu betrieben und dort gingen die Module nie über 403Watt. Es ist halt auch merkwürdig das es überall dieselben Werte sind. Das ist schon auffällig weil die neuen Module auch 435er bifazial sind und dann deutlich höhere Leistung liefern müssten. Für mich sieht das so aus als ob da die Max Werte irgendwie für beide WRs herangezogen werden.

lumapu, 2024-04-02:
du kannst bei dir mal die API prüfen:
http://[IP]/api/inverter/id/0

da sind die max Werte auch dabei, evtl. ist es ein Javascript Problem (was ich nicht vermute)
Werden bei dir die max Werte nachts zurückgesetzt?

slg60, 2024-04-02:
Ja die Werte werden jede Nacht zurück gesetzt

slg60, 2024-04-02:
In der API sind die Max Werte Identisch zur GUI bei beiden Invertern.

lumapu, 2024-04-02:
danke für's bestätigen, jetzt bleibt herauszufinden, wie die da rein kommen

lumapu, 2024-04-02:
kannst du tagsüber oder besser in den Morgenstunden prüfen, ob sich der Max Wert immer so verhält, dass er überall gleich ist?

slg60, 2024-04-02:
Ich werde mir das anschauen. Werde berichten.

lumapu, 2024-04-02:
ich vermute du bist an die technische Leistungsgrenze deiner 1600er gekommen, egal mit welchen Modulen, da es wohl beide schaffen. Gerade im Frühling mit kalten Temperaturen und klarem Himmel herrschen je nach Einstrahlwinkel sehr gute Bedingungen, die zu sowas führen können.

slg60, 2024-04-02:
Ist möglich, 4x429,5=1718 davon 95% sind 1632W AC. Das würde passen. Würde aber im Umkehrschluss bedeuten, dass der 1600er die Modulleistung drosselt wenn er an seiner ac Grenze angekommen ist. Da habe ich keine Erfahrungen mit was der WR bei Max Last tut.

Die 435er Module haben im Vergleich zu den 410ern eine flachere Dachneigung, das bestärkt die These auch etwas.

lumapu, 2024-04-02:
er muss ja an der Quelle was machen, um AC seitig sein Limit nicht zu überschreiten, bin gespannt ob du es morgen (inzwischen schon heute) aufklären kannst.

slg60, 2024-04-03:
Heute werde ich wohl nicht annäherungsweise an die Max Werte heran kommen. Ich warte auf Sonne und melde mich dann zurück.

lumapu, 2024-04-03:
und wie verhalten sich bei diesem Wetter die max Werte? Sind die unterschiedlich? Wenn ja stimmt sehr wahrscheinlich meine These von gestern.

Loetnase, 2024-04-03:
Schau mal wie sich die 8 Werte heute verhalten, sind alle etwas unterschiedlich oder gibt es doppelte.
Bei dem nächsten Sonnentag kannst du 2 Versuche machen 
1. Decke ein Solarmodul etws ab, damit es nicht die volle Leistung bringt und schau was die 4 Werte dieses Inverters machen,
bleibt der abgedeckte unter den anderen und kommt nicht auf das Maximum.
2. Begrenze den 2. Inverter mal auf 70% oder so was, ich denke du wirst dann 4 ähnliche Werte bekommen aber keine identischen,
das kannst du dann vieleicht mit 90, 95 und 98% nochmals wiederholen um zu sehen ob du wieder auf die identischen 429,5 Watt kommst, welche dann eben der Anschlag vom Inverter ist.

slg60, 2024-04-03:
[image]
So, hier jetzt die Werte von heute. Alles so wie es sein soll. Das mit der Leistungsgrenze wird es wohl sein, mich wundert nur stark, das es die 410er Module schaffen Knapp 430 Watt zu erzeugen. Aber da ich vorher den 1500er hatte wurden die vermutlich vorher schon deutlich früher gedrosselt.

lumapu, 2024-04-03:
sehr schön, dann sehe ich diesen Issue als gelöst an 😊
Ist doch ein schönes Gefühl, wenn die 410W Module mehr leisten als sie versprechen

closed 2024-08-16 by lumapu