lumapu/ahoy #1559: Permanenter Neustart der Ahoy DTU [Bug]

Tiezy, 2024-04-02:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

08103

### Github Hash

7ee1f99 

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

-

### Debug Serial Log output

```shell
-
```

### Error description

Coming from 08.97 and update to 08.101 leads to permanent reboot of the DTU. For short time the DTU can be entered via the webinterface and there is mentioned that the NTP timeserver is not available. I tried also 08.103 but same issue remains.  It looks like also that the WLAN connection / password was lost as the DTU open the Ahoy access point but entering the wlan name and password doesn't solve the issue.
In case of downgrade back to 08.97 everything works fine as expected. Provision of log seems to be impossible as the DTU permanent reboots.
What I'am doing wrong??
Thanks for your feedback. Maybe I'am the only one?

Tiezy, 2024-04-02:
I missed to add that a manual NTP sync via webinterface failed with the Error: ERR_PROTECTED same as sync via browser.

lumapu, 2024-04-02:
'protected' feels like a password protected DTU. Is that right?
We fixed recently with 102 the NTP issue. Can you check 102 or 103 again?
Do you have configured the default NTP server?

You can also answer in German if preferred.

SebiWolze, 2024-04-02:
Guten Abend,

habe selbiges Problem, allerdings mit der letzten Stable Version.
Geflasht, 192.168.4.1 -> alles super.
Eigene WIFI Daten rein, reboot.
Er zeigt an, dass er sich mit meinem WLAN verbindet, macht ein NTP Update, allerdings 2 Stunden vor unserer Zeit -> nicht mehr erreichbar.
Exakt 2 Pings gehen nach jedem Reboot durch, danach nicht mehr erreichbar bis ich neu flashe, danach wieder über 192.168.4.1 zu erreichen.

[image]

Tiezy, 2024-04-03:
Danke schon mal für die Rückantwort. Die Fragen kann ich wie folgt beantworten.

- 'protected' feels like a password protected DTU. Is that right? Jawohl das ist korrekt.
- We fixed recently with 102 the NTP issue. Can you check 102 or 103 again? 103 zeigt das gleiche Verhalten.
- Do you have configured the default NTP server? Ja habe ich. Hat sich dazu etwas geändert also muss man beim update von 08.97 auf 08.101 / 08.103 die NTP setting z.B. neu eingeben?

mfg

edw00d4711, 2024-04-03:
Ich hab das gleiche Problem. Bin schon auf eine USB-C ESP gewechselt, weil ich dachte, die Platine hat einen Schaden. Aber da das gleiche Problem. Ständige Reboots. Fehler mit dem NTP zu verbinden.

edw00d4711, 2024-04-03:
.103 funktioniert. NTP wird direkt synchronisiert. Keine WLAN Aussetzer und keine Reboots mehr. Ich war schon am verzweifeln.

lumapu, 2024-04-03:
ah ok, ich weiß bescheid, wird wieder rückgängig gemacht, die 104 ist im anrollen 😊

Tiezy, 2024-04-04:
Hallo Lukas,

super mit der 104'er sieht es besser aus, ABER: direkt nach dem update startet die DTU wieder mit permanenten Reboots.
Die Webside der DTU kam aber trotz der dauernden reboots hoch und ich konnte lesen das die NTP Zeitsync nicht funktioniert. Es funktionierte nun aber der Zeitsync über den Browser (wahrscheinlich gerade im richtigen Moment den Buttom gedrückt) und die DTU läuft nun durch. Ein manueller NTP sync über das Webinterface Menue / settings / NTP Server funktionierte ebenfalls.

Also läuft da beim Startup noch was nicht ganz rund. Also zumindestens bei mir :-).

Danke für dein Engagement.

lumapu, 2024-04-05:
leider kann ich den NTP Fehler bisher noch nicht nachstellen. Würdet ihr bitte bestätigen, dass der Fehler in 0.8.106 weiterhin auftritt?
Ich habe mal die stable Release 0.8.83 installiert (mit Erase) und eingerichtet. Danach bin ich dann per OTA auf die 0.8.106 gegangen.

Tiezy, 2024-04-06:
Also nach dem update von 08.103 auf die 0.8.106 habe ich folgendes beobachtet. Das Problem mit permanenten reboots besteht immer noch. Offensichtlich ist der NTP Server beim starten nicht erreichbar. Ein Zeitsync über den Browser klappt manchmal. Ich habe jetzt mal den DTU reset für Mitternacht und das lesen des Grid Profiles deaktiviert und reboot after successfull save angeklickt. Danach hatte die DTU das Netzwerkpasswort wohl wieder vergessen und der neue Setup Wizard kam hoch. Nach korrektem Eintrag startet die DTU aber auch hier nicht von selbst neu. Der Wizard muss abgebrochen werden und das Netzwerk manuell in der Weboberfläche eingetragen werden. Downgrade zur 0.8.97 und alles läuft.

hrolofs, 2024-04-07:
Schau auch mal ob du "System Config/Reboot Ahoy at midnight" aktiviert hast und schalte das gegebenefalls aus. Bei mir hat das all die NTP/Reboot Probleme bei den 0.8.1XX Versionen wohl beeinflusst. Hab das bei meinen DTUs vor dem updaten von 0.8.97 auf die 0.8.106 ausgeschaltet und das update lief sauber durch.

sebastianrothe, 2024-04-07:
Also bei mir geht die 106 mit beiden Einstellung (Reboot at Midnigh an/aus).

hrolofs, 2024-04-07:
Nach Aktivierung Save&Reboot gemacht? Bei mir gehen die beiden DTUs danach in den Dauer reboot.

sebastianrothe, 2024-04-07:
Ja, habe mehrfach neugestartet (von Hand). Kommt alles wie gewollt hoch.

Kannst du mal dein Timezone überprüfen? Die war bei mir noch von vor der Zeitumstellung auf `+1`, muss aber jetzt auf `+2`.

hrolofs, 2024-04-08:
Habs gerade nachgeschaut, steht auf +1. Bin nicht sicher ob das jetzt +2 sein muß. Urzeit im Display und Web-page ist ja richtig. Habs mal auf +2 geändert und reboot. Uhrzeiten in der Anzeige sind immer noch richtig. Keine Ahnung worauf die DTU diesn Parameter anwendet. 

Danach die Midnight Option aktiviert. Und nach einem Reboot geht die DTU wieder in den Dauer Boot Modus.

sebastianrothe, 2024-04-08:
Laut https://time.is/Leipzig#time_zone ist `+2` richtig. 

Dann scheinen sich die ESPs da anders zu verhalten. Sorry :)

Ollipop030, 2024-04-08:
Auf meinem ESP32 (WROOM 30 Pin) macht die DTU auch alle paar Minuten einen Neustart. Reboot at midnight an oder aus spielt keine Rolle, längste Uptime war heute 30 Minuten.

lumapu, 2024-04-08:
fiexd mit 107?

edw00d4711, 2024-04-09:
Leider nicht für ESP32. Da scheint der Build nicht durchgelaufen zu sein. 
https://fw.ahoydtu.de/fw/dev%2Fahoy_v0.8.107/
Auch unter den "Actions" nicht. Aber ich vermute es verlinkt auf das gleiche Verzeichnis

hrolofs, 2024-04-09:
@sebastianrothe 
> Laut https://time.is/Leipzig#time_zone ist `+2` richtig.

Ist Definitionssache und hängt von ab wie das implementiert ist.

Vorsicht: Klugscheißer Modus on

Wir leben in der MEZ, die einen Offset von +1 Stunde zur UTC hat.

Zusätzlich hat die MEZ noch eine DST (Daylight Saving Time), zusätlicher temporärer Offset von +1 Stunde

Das ergibt aktuell einen Offset zur UTC von + 2 Stunden.

Manchmal konstruiert man daraus eine MESZ mit +2 Stunden zur UTC. Dazu müsste es dann 
aber auch einen negativen DST Offset für die Winterzeit geben.

Windows gibt z.B. für uns immer "(UTC+01:00) Berlin" an. Im Sommer halt mit zusätzlichem 
DST Offset von + 1 Stunde.

Klugscheißer Modus off

Man kann das auch ganz anders sehen bzw. interpretieren. 

Hängt halt davon ab wie's implementiert bzw. programmiert wurde. ;-)

Ollipop030, 2024-04-10:
Version .108 läuft bisher ohne Reboot.

edw00d4711, 2024-04-10:
Bei mir läuft es auch mit .108. Dafür das epaper nicht mehr. Aber das ist ein anderer Thread

Tiezy, 2024-04-11:
Hallo Lukas,
als issue opener kann ich dir mitteilen das es mit der 0.8.108 nun tadellos klappt, Keine reboots nach dem Update und das update selbst verlief ebenfalls reibungslos.
Prima, Vielen Dank

closed 2024-04-11 by Tiezy