lumapu/ahoy #1560: Findet HMS-500-1T nicht

schnuller123, 2024-04-02:
Hallo

Ich hab schon länger 4 Wechselrichter mit AhoyDTU in Verwendung, alles super
Nun neue Hoymiles HMS500 eingebunden, aber er wird nicht gefunden
Ist da was bekannt?

## Hardware
  * [x ] ESP32

### Version / Git SHA: 
Version: 0.8.83

lumapu, 2024-04-02:
ja, die neuesten haben Buchstaben in der Seriennummer und benötigen daher eine neuere Ahoy Version, einfach die neuste (0.8.103) ausprobieren.
https://fw.ahoydtu.de/fw/dev/

schnuller123, 2024-04-02:
sind aber nur Zahlen, wie bei den anderen!
Was komisch ist, das Netzkabel hat recht dünnen Querschnitt, ( max 1,5mm² )
Der HMS-350 hatte doppelt so dicken Draht, fake wechselrichter
der gar kein Funkmodul hat?

lumapu, 2024-04-02:
der Wechselrichter ist aber kein HMS 500W 1T oder? Man beachte das W
Diese Generation wird von Ahoy nicht unterstützt

schnuller123, 2024-04-03:
Hallo Lukas

 

gibt es überhaupt einen mit W , ich hab keinen gefunden 

meiner ist definitiv einer ohne W

mit W wäre mit Wlan funktion, klar dann geht Ahoy nicht

 

entweder der DTU teil im WR ist defekt oder nicht vorhanden  

 

MfG Dellinger
 

Gesendet: Dienstag, 02. April 2024 um 23:23 Uhr

lumapu, 2024-04-03:
das weiß ich nicht, war nur eine Vermutung mit dem 'W' und ist ein häufiger Fehler. (Hoymiles bringt immer wieder neue Wechselrichter auf den Markt, da bin ich nicht immer up-to-date)
Funkmodul wird das richtige sein, da du schon von HMS-350 schreibst.

schnuller123, 2024-04-04:
HalloHab nochmal ne Frage an den OberGuruanbri ein Log:17:43:21.205 I: Iv: 317:43:21.216 I: Ostseite/ch1/U_DC: 29.900 V17:43:21.216 I: Ostseite/ch1/I_DC: 0.600 A17:43:21.217 I: Ostseite/ch1/P_DC: 17.800 W17:43:21.218 I: Ostseite/ch1/YieldDay: 1386.000 Wh17:43:21.219 I: Ostseite/ch1/YieldTotal: 220.987 kWh17:43:21.230 I: Ostseite/ch1/Irradiation: 4.450 %17:43:21.231 I: Ostseite/ch1/MaxPower: 320.500 W17:43:21.232 I: Ostseite/ch2/U_DC: 0.800 V17:43:21.233 I: :43:21.114 I: webSerial, buffer overflow!webSerial, buffer overflow!Was bedeutet der overflow, macht nur 1 Wechselrichter von 3Und noch was17:43:12.343 I: (#2) Radio infos: -6 4 1 -4 -4 | t: 10888, s: 6343, f: 10, n: 4526 | p: 255Gibt es dafür eine Beschreibung,find leider meinen DTU nicht mehr und dachte ich könnt über signalstärke eingrenzenmfg Dellinger

schnuller123, 2024-04-04:
Hab die DTU mit neuster Version geflasht
Und WR nochmal neu eingefügt, da gibt es jetzt komischen Eintrag
Bei Funkmodul Frequenz und Sendeleistung, bei den anderen steht LOW drin
Was mach ich falsch?

schnuller123, 2024-04-04:
Kann es sein, das anhand der SNR ein WR Typ erkannt wird, der ein CMT2300A Funkmodul benötigt?
weil Feld aufgeht für Freqenz und Sende Leistung beim einrichten des WR

lumapu, 2024-04-05:
Die HMS Wechselrichter benötigen ein CMT2300A Funkmodul. Ich bin davon ausgegangen, dass du schon HMS350 im Einsatz hast und habe deshalb diese Frage garnicht gestellt.

kiu77, 2024-04-06:
Wie lumapu sagte, brauchst Du wohl eine DTU mit beiden Funkmodulen. nRF24L01+ für die alten HM-Inverter und ein CMT2300A für die neue HMS-Version. Falls Du eine passende Platine willst, geb Bescheid.