lumapu/ahoy #1561: Neuer Ahoy User - nach Einwahl in Heim-WLAN und NTP Settings verhält sich DTU wie ein Stein[Bug]

SebiWolze, 2024-04-02:
### Platform

ESP32

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.97

### Github Hash

abc1234

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Freshly installed latest stable / 0.8.97 / 0.8.103

### Error description

[image]
[image]

Ich hoffe deutsch ist okay, übersetze sonst gern noch auf Englisch.

Heute ESP32 "Wroom" erhalten, mit nrf24 und Display verkabelt.
Über Webinstallier die letzte stable geflasht.
Mit Wifi AP verbunden, eigene WLAN Daten + NTP (Fritzbox eingetragen).
Save.

Danach bekomme ich noch genau 2 Pings, danach ist Ende.
Nochmal geflasht, selbiges Fehlerbild.

Hier ein paar Issues gelesen, 0.8.97 geflasht.
Selbes Fehlerbild.
Mit 0.8.103 exakt das selbe.

2 Pings gehen durch, danach ist vorbei.

Hier noch eine Console Ausgabe mit latest Stable:
[image]

Arbeite im Bereich IT-Sec, Netzwerk/WLAN/ESP ist für mich täglich Brot, Antworten dürfen gerne technischer Natur sein.

Vielen Dank!

lumapu, 2024-04-02:
herzlich willkommen 😊

Die NTP Fehler waren eher so, dass hat keine Zeit geholt wurde, daher denke ich, dass das hier was anderes ist.
Wenn ich es richtig verstanden habe, ist kein Zugriff auf die web UI möglich, da nur zwei pings durchgehen.
Wurde per AP irgendwas neben dem lokalen WiFi konfiguriert?
Lese des öfteren in letzter Zeit Probleme beim ESP32, aber nicht beim letzten stable Release

knickohr, 2024-04-02:
Statische IP in Ahoy oder DHCP ?  Wenn Du statische IP hast, dann hoffentlich auch ein Gateway und DNS eingetragen ?

SebiWolze, 2024-04-02:
> @SebiWolze
> 
> > Arbeite im Bereich IT-Sec, Netzwerk/WLAN/ESP ist für mich täglich Brot, Antworten dürfen gerne technischer Natur sein.
> 
> sehr gut, wegen folgender procedure:
> 
> bitte flashe deinen esp32 mit einem anderen, non dtu related, projekt, zb. mit micropython, und richte damit das wlan ein. beobachte bitte dann das ping verhalten erneut.
> 
> reason: crosscheck der stabilen funktionalität des esp32 wifi modules dieses esp32 boards, mit alternativer, non related firmware.
> 
> alternative procedure: crosscheck der identischen firmware auf einem weiteren, ähnlichen (esp32 board eines anderen herstellers) oder baugleichen esp32 board (zweites board des gleichen herstellers)
> 
> thx & greetings, metachuh.

Installiere gerade ESP32 Home, gebe danach direkt Info.

SebiWolze, 2024-04-02:
> Statische IP in Ahoy oder DHCP ? Wenn Du statische IP hast, dann hoffentlich auch ein Gateway und DNS eingetragen ?

Wenn es an der IP liegen "würde", dann wäre diese ja nicht mal erreichbar und dann wieder nicht.
Im DHCP ist keine Reservierung eingetragen.
Ich habe es natürlich bereits mit einer static IP probiert, auch hier 1:1 selbes Fehlerbild.

Und ja, DNS und GW sind auch korrekt eingetragen.
GW ist ja erst einmal zweitrangig da sich ESP und meine Geräte im selben Netz/Subnetz befinden.

SebiWolze, 2024-04-02:
> herzlich willkommen 😊
>
> Die NTP Fehler waren eher so, dass hat keine Zeit geholt wurde, daher denke ich, dass das hier was anderes ist. Wenn ich es richtig verstanden habe, ist kein Zugriff auf die [...]

Danke fürs Willkommen :-)
Nein, nichts weiter konfiguriert.

In weiteren Versuchen dann NTP Server geändert.
Dann mal die IP static eingetragen. Dann 1. SSID, dann 2. SSID.
Ich habe wirklich alles probiert was mit dem Thema Netzwerk zusammenhängt.
Aber sobald die Settings drin sind und ein Reboot durchlief immer das beschriebene Fehlerbild.

SebiWolze, 2024-04-02:
> @SebiWolze
>
>
> sehr gut, wegen folgender procedure:
>
> bitte flashe deinen esp32 mit einem anderen, non dtu related, projekt, zb. mit micropython, und richte damit das wlan ein. beobachte bitte dann das [...]

Okay nun wirds unangenehm... (hätte es selbst testen müssen!):

[image]

ESP32 Home selbes Fehlerbild.
Nach Connect 2 Pings, dann Schicht im Schacht.

Also ESP32 im Eimer?

SebiWolze, 2024-04-02:
> @SebiWolze oje.
> 
> nicht zwingend komplett im eimer. probier mal im wifi router andere kanäle im 2.4ghz bereich aus. einige semi-faulty esp wifi module funktionieren dennoch auf spezifischen kanälen.
> 
> greetings, metachuh

Sind Ubiquiti Wifi 6 LR APs.
Ich spiele mal etwas rum.
Gibt Erfahrungen welche 2,4Ghz Kanäle besser funktionieren?

lumapu, 2024-04-02:
ich habe keine Erfahrung auf dem Gebiet, wieder daher sagen: ja kaputt
Finde gut, dass sich das Fehlerbild so eindeutig zeigt und damit klar wird wie es weiter geht.
Ist der erste Fall, wo ich von einem defekten ESP32 höre.
Du könntest den ESP noch 'nackt' machen und alles ablöten wie zb. NRF24, display und co. (Ich habe keine Idee was du schon dran gebaut hast)

SebiWolze, 2024-04-02:
Läuft nun auf Channel 6.
Läuft stabil.
Keinerlei Abbrüche mehr.

Ich werd verrückt...

Oh man oh man!

Danke euch allen!!!

knickohr, 2024-04-03:
Ohne jetzt klug zu scheißen !  Mit Ubiquiti gab es schon öfters Probleme hier, vor allem mit mehreren APs.

Und ich würde trotzdem das GW eintragen, auch wenn’s das gleiche Subnet ist.

Jumperm, 2024-04-03:
Hier hatten wir bereits das Thema der Mesh Systeme und Kanälen
https://github.com/lumapu/ahoy/issues/1307#issuecomment-1874609611

closed 2024-08-16 by lumapu