lumapu/ahoy #1562: [Bug] Crash/Bootloop beim Update von 0.8.103 -> 0.8.104

blueline13, 2024-04-04:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.104

### Github Hash

5a32288

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

CMT Konfig:
SCLK    GPIO12
SDIO    GPIO14
CSB    GPIO27
FCSB    GPIO26
GPIO3    GPIO34

### Debug Serial Log output

```shell
13:42:03: AP client connected
13:42:10: AP client connected
13:42:25: AP client connected
13:43:08: I: memory usage: 1495
I: capacity: 109556
I: max alloc: 44020
I: settings saved
13:43:11: I: Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
E (522) esp_core_dump_flash: No core dump[00][18]…ÉÑ¥Ñ¥½¹found!
E (522) esp_core_dump_flash: No core dump partition found!
I: Initializing FS ..
I:  .. done
I: Config Version: 11
I: Settings valid: true
I: Radio Config:
Channel			= 76 (~ 2476 MHz)
Model			= nRF24L01+
RF Data Rate		= 250 KBPS
RF Power Amplifier	= PA_LOW
RF Low Noise Amplifier	= Enabled
CRC Length		= 16 bits
Address Length		= 5 bytes
Static Payload Length	= 32 bytes
Auto Retry Delay	= 1000 microseconds
Auto Retry Attempts	= 15 maximum
Packets lost on
    current channel	= 0
Retry attempts made for
    last transmission	= 0
Multicast		= Disabled
Custom ACK Payload	= Disabled
Dynamic Payloads	= Enabled
Auto Acknowledgment	= Enabled
Primary Mode		= TX
TX address		= 0xe7e7e7e7e7
pipe 0 ( open ) bound	= 0xe7e7e7e7e7
pipe 1 ( open ) bound	= 0x2476638301
pipe 2 (closed) bound	= 0xc3
pipe 3 (closed) bound	= 0xc4
pipe 4 (closed) bound	= 0xc5
pipe 5 (closed) bound	= 0xc6
I: DTU_SN: 83637624
E (141) spi: spi_bus_initialize(756): SPI bus already initialized.
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x4008c980
file: "hms/esp32_3wSpi.h" line 58
func: void esp32_3wSpi::init(uint8_t, uint8_t, uint8_t, uint8_t)
expression: spi_bus_initialize(DEF_CMT_SPI_HOST, &buscfg, SPI_DMA_DISABLED)

abort() was called at PC 0x4008c983 on core 1

Backtrace: 0x40083cb1:0x3ffb2050 0x4008c98d:0x3ffb2070 0x40092b3d:0x3ffb2090 0x4008c983:0x3ffb2110 0x400d5fd8:0x3ffb2130 0x400f062d:0x3ffb21e0 0x400f0e4c:0x3ffb2270 0x40108a8a:0x3ffb2290

ELF file SHA256: 9fe42b891ad50f61

13:43:12: E (356) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
E (522) esp_core_dump_flash: No core dump[00][18]…ÉÑ¥Ñ¥½¹found!
E (522) esp_core_dump_flash: No core dump partition found!
I: Initializing FS ..
I:  .. done
I: Config Version: 11
I: Settings valid: true
I: Radio Config:
Channel			= 76 (~ 2476 MHz)
Model			= nRF24L01+
RF Data Rate		= 250 KBPS
RF Power Amplifier	= PA_LOW
RF Low Noise Amplifier	= Enabled
CRC Length		= 16 bits
Address Length		= 5 bytes
Static Payload Length	= 32 bytes
Auto Retry Delay	= 1000 microseconds
Auto Retry Attempts	= 15 maximum
Packets lost on
    current channel	= 0
Retry attempts made for
    last transmission	= 0
Multicast		= Disabled
Custom ACK Payload	= Disabled
Dynamic Payloads	= Enabled
Auto Acknowledgment	= Enabled
Primary Mode		= TX
TX address		= 0xe7e7e7e7e7
pipe 0 ( open ) bound	= 0xe7e7e7e7e7
pipe 1 ( open ) bound	= 0x2476638301
pipe 2 (closed) bound	= 0xc3
pipe 3 (closed) bound	= 0xc4
pipe 4 (closed) bound	= 0xc5
pipe 5 (closed) bound	= 0xc6
I: DTU_SN: 83637624
E (141) spi: spi_bus_initialize(756): SPI bus already initialized.
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x4008c980
file: "hms/esp32_3wSpi.h" line 58
func: void esp32_3wSpi::init(uint8_t, uint8_t, uint8_t, uint8_t)
expression: spi_bus_initialize(DEF_CMT_SPI_HOST, &buscfg, SPI_DMA_DISABLED)

abort() was called at PC 0x4008c983 on core 1

Backtrace: 0x40083cb1:0x3ffb2050 0x4008c98d:0x3ffb2070 0x40092b3d:0x3ffb2090 0x4008c983:0x3ffb2110 0x400d5fd8:0x3ffb2130 0x400f062d:0x3ffb21e0 0x400f0e4c:0x3ffb2270 0x40108a8a:0x3ffb2290

ELF file SHA256: 9fe42b891ad50f61

E (356) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
```

### Error description

- Mit einem laufenden System mit funktionierender Konfig und 0.8.103(240402_ahoy_0.8.103_7ee1f99_esp32-wroom32)
        das Update auf 104 gewagt(240404_ahoy_0.8.104_5a32288_esp32-wroom32)
- Nach Update ist der ESP nicht mehr zu erreichen per WEB
- ESP wieder mit der Webseite auf die 0.8.83 gebracht(inkl. erase)
- Alle Punkte von nun an nur im AP Modus
- Update auf 104 läuft und ESP startet
- ESP stromlos
- Factory Reset
- ESP stromlos
- Nur das CMT aktviert und gespeichert sofort Bootloop.

Das ganze ist reproduzierbar.

Ist etwas knapper gehalten, ich hoffe nicht zu knapp. Das Log ist ja sehr eindeutig. Ich reiche gerne nach falls euch etwas fehlt.

Wenn ich etwas testen kann lasst es mich wissen!

hrolofs, 2024-04-04:
Ich würde "vorsichtig" mit der 0.8.104 sein. 

Wenn du unter Actions nachschaust (https://github.com/lumapu/ahoy/actions/runs/8547014944) sind bei dem Durchlauf einige Fehler aufgetretten und ein Durchlaufzeit von mehr als 6h statt ein paar Minuten lässt auch drauf schliessen das da was schief geleufen ist.

slg60, 2024-04-04:
Ich bin da auch gerade gescheitert, 104 geflasht und DTU tot. ESP32!

slg60, 2024-04-04:
cmt2600a

lumapu, 2024-04-04:
> Ich würde "vorsichtig" mit der 0.8.104 sein.
> 
> Wenn du unter Actions nachschaust (https://github.com/lumapu/ahoy/actions/runs/8547014944) sind bei dem Durchlauf einige Fehler aufgetreten und eine Durchlaufzeit von mehr als 6h statt ein paar Minuten lässt auch drauf schliessen das da was schief gelaufen ist.

das lag an Github selbst, die hatten letzte Nacht ein Actions-Problem

blueline13, 2024-04-05:
@lumapu 
Moin zusammen, habe eben mal die 0.8.105(Normal Englisch) getestet und die zeigt noch immer das gleiche Problem. Vorgehensweise wie oben. Habe die Minimal Version probiert, auch die zeigt das gleiche Verhalten. Das Log unterscheidet sich nicht, wird noch immer der gleiche Fehler angezeigt. Auf wunsch kann ich das log auch gerne nachreichen.

lumapu, 2024-04-05:
Das Problem sehe ich bei mir auch, verstanden habe ich es noch nicht. Es liegt an SPI, der beim ESP32 doppelt verwendet werden muss (oder nicht?). Ich weiß an welcher Stelle es scheppert kann aber leider das Problem noch nicht genauer benennen.

blueline13, 2024-04-05:
Ich versuche mich mal an einer Interpretation des Fehler so wie ich es sehe. Ich habe mir den Code nicht angeschaut.

1) ESP32 hat 4 SPI Interfaces, 0+1 sind intern fürs Flash zuständig. 2+3 stehen dem Anwender zur Verfügung und sind eigenständig(HSPI & VSPI).
2) Die Pins für das SPI Interface sind frei Programmierbar.
3) im Log nach der DTU_SN steht das der SPI BUS für den CMT doppelt initialisiert wurde.

Ich lese ja in den Change Logs das gerade viel im Netzwerkbereich umgebaut wird. Kann es sein das etwas irgendwo stehen geblieben oder doppelt rein gerutscht ist für die Initialisierung des CMT SPI Busses. Denn nur das NRF Modul rennt ohne Probleme.

Es muss ja irgendetwas zwischen der 103 und 104 passiert sein. So habe mir noch das DTUFusionBoard genommen und das hat keine Probleme. Update auf 105 und rennt.

Also was nur mit dem ESP32 bis jetzt zu tun hat.

blueline13, 2024-04-06:
Bestätige gerne das das mit der 0.8.106 wieder läuft. Dann mache ich hier wieder zu.

Wie immer einen SUPER DANK!!!

closed 2024-04-06 by blueline13