lumapu/ahoy #1563: RSSI und RX No Answer/Loss rate

slg60, 2024-04-04:
## Hardware
  * [x] ESP32
  * [x] CMT2300A

### Antenna:
* [x] external antenna (SMA)
Version 0.8.101

Bin auf der Suche nach der richtigen Position der DTU. Die beiden HMS1600 sind etwa 30m auseinander, einer unter den Modulen, einer separat montiert. Aktuelle RSSI Werte sind bei dem separat montierten ziemlich konstant bei -56 dbm, der läuft unauffällig, bei dem unter den Modulen montierten schwanken sie stark zwischen -67 und -75 dbm. Bei letzterem kommt es ab etwa -70dbm zu RX no Answer von bis zu 50% und auch teilweise Inverter Loss. Teilweise antwortet der Inverter dann bis zu 3-5 Minuten gar nicht. Kann man da was optimieren in den Settings (z.B Frequenz?) oder muss man einfach eine bessere Position für die DTU finden? Ich würde die DTU halt gern irgendwo haben wo ich das Display sehen kann, Strom muss da natürlich auch vorhanden sein. Ich nutze parallel noch eine Homematic die auch auf 868mhz funkt. Gibt es da Erfahrungen ob sich das evtl. in die Quere kommt?
[WR_BAD.txt](https://github.com/lumapu/ahoy/files/14874017/WR_BAD.txt)

slg60, 2024-04-04:
Ich habe die kleine Antenne von Blinkyparts mit dazu gekauft, die scheint wohl nicht so dolle zu sein. Habe mir mal eine externe 7dbi 868Mhz Antenne bestellt. Wenn die besser ist, dann kommt die DTU mit in den Netzwerkschrank (momentan hängt sie daneben).

Das mit unterschiedlichen Frequenzen kann ich mal testen, will aber nicht zu weit hoch, dann steigt die Gefahr dass die Homematic sich nicht mehr wohlfühlt.

dtuuser, 2024-04-04:
> Hast du mal eine andere Antenne getestet? Du kannst auch den zweiten WR mal auf einer anderen Freq. laufen lassen - z.B. WR 1 865MHZ WR2: 863 MHZ - am besten abends umstellen wenn Licht aus ist. Ansonsten wenn du es tagsüber machst 15min DTU aus sonst ist das Funkmodul beleidigt.

Kann ich ich Freq. auch beim HM600 einstellen? Falls ja, wo geht das?

kiu77, 2024-04-05:
RSSI-Pegel von -56dBm sind schon nicht gut. aber < -67dBm ist echt schlecht.
Zum einen schirmen Solarmodule selbst ab, zum anderen ist wohl echt was nicht okay mit der Antenne, wie Dirk-Ahoy schon sagte. Ich habe Pegel um die -33dB mit richtiger Antenne. 
Zum Test kannst Du mal einen sehr dünnen Starrdraht in die SMD-Buchse stecken, der 8,6cm lang ist. Oder aber Du öffnest die Antenne und verlängerst die darin enthaltene Litze auf diese 8,6cm. Dann sollte der Pegel deutlich besser sein.

kiu77, 2024-04-05:
@slg60 
Ah, Du hast einen Inverter zwischen Modulen und Dachziegeln in einem Hohlraum <10cm Höhe?

slg60, 2024-04-05:
Der Inverter ist auf die Modulschiene geschraubt und befindet sich unterhalb des Moduls. Abstand zur Ziegel sind etwa 12-14cm. Klassische "Balkonkraftwerk Montage".

kiu77, 2024-04-05:
@slg60 
Naja, eher klassische Solaranlagenmontage. Ist ja schließlich kein Balkon ;-)
Falls Du bereit für eine kleine Anpassung bist: Verschiebe den Inverter soweit an den Rand Deiner Module, dass die kleine Stummelantenne grade rausschaut. Da es dann kabelmäßig im Unterschied zur Montage in der Mitte zwischen zwei Modulen passt, braucht es noch zwei Verlängerungen für das Solarkabel eines Moduls von ca 1,5m Länge.
Oder: Wenn die Schienen das zulassen, montiere die Module so, dass etwa 20cm Luft zwischen zwei Module ist und die Antenne des Inverters etwa da Luft sieht.

kiu77, 2024-04-05:
@slg60 
Bez. Verlängerungen überschätzt Du vielleicht die Auswirkungen. Die typischen 11A bei 2x1,5m und einem Querschnitt von 6mm2 ergeben zusätzliche 8,5mΩ = 94mV Spannungsabfall, ergo einen Leistungsverlust von etwa 1W = 0,22% bei einem 425Wp-Panel. Aber klar, wenn es ohne geht, ist das in Ordnung.
Bei klassischen Solaranlagen sind die Kabel zum Inverter in aller Regel sehr viel länger.

slg60, 2024-04-05:
Nutze 4mm² aufgrund der geringen Längen und wenn ich den WR von den Modulen weg gelegt hätte hätte ich Verlängerungen von je 8 Meter benötigt. Da wird das schon ein bissl mehr. Die zusätzlichen MC4 Verbindungen stellen auch noch einen Übergangswiderstand dar. Das ist alles nicht die Welt, aber macht sich bemerkbar. War aber auch nur ein weiterer Punkt der gegen eine Verlegung sprach und auch nicht der schwerwiegendste.

kiu77, 2024-04-05:
JA, das ergäben Verluste von 5,5W etwa. Immerhin ca. 1,2%.
Daher auch mein Vorschlag, die Sache so zu montieren, dass die Antenne des Inverters Luft sieht.
Aber eine richtige 860MHz-Antenne am CMT2300A dürfte auch reichen.

slg60, 2024-04-05:
So, DTU im Netzwerkschrank, Externe Antenne oben drauf, biss Zeit investiert in die Platzierung. Bin jetzt bei Inverter0 bei -37 und Inverter 1 bei -58. Besser bekomme ich es an dieser Stelle nicht hin. Bisher läuft das auch ganz gut. Mal sehen wie stabil das die nächsten Tage läuft.

DanielR92, 2024-06-29:
Da es hier keine weitere Diskussionen ergibt, close ich das hier mal.
Bitte neu öffnen wenn ich falsch gehandelt habe. :)

closed 2024-06-29 by DanielR92