lumapu/ahoy #1566: Version 0.8.105 geht bei mir überhaupt nicht.

kiu77, 2024-04-05:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.105

### Github Hash

1234

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

ESP32-32U auf Platine mit nRF24L01+ und CMT2300A, 3,3V mit eigenem Spannungsregler.
Update von 0.8.101 (die prima funktioniert)

### Debug Serial Log output

```shell
-
```

### Error description

Generell:
0.8.105 lässt sich nicht verwenden.

Ich habe verschiedene Methoden probiert:
1. das normale Online-Update mit der 0.8.105_...-wroom32-de. DTU war tot und ließ sich nicht mehr neu flashen mit der Funktion von AhoyDTU.de unter Chrome auf dem Mac. Auf einem PC mit Chrome ging es dann.
2. nach neu flashen mit Erase des EEPROMs ein Update von release 0.8.83 auf beta 0.8.101 mit anschließendem neu Hochladen der Settings (stammen von 0.8.97) ging es wieder. Aber anschließendes Update auf 0.8.105 siehe 1.
3. nach neu flashen mit Erase des EEPROMs ein Update von release 0.8.83 direkt auf beta 0.8.105 ebenfalls nix, siehe 1.

Übrigens war nach Update auf 0.8.105 egal ob mit direkt vorhergehenden Stromlosmachen und neuem booten oder ohne dies der ESP32 unter dem Mac nicht mehr ansprechbar.

blueline13, 2024-04-05:
Das hört sich nach dem gleiche Problem wie auch #1562 an.

kiu77, 2024-04-05:
@blueline13 
Stimmt. Ich hatte den Thread mit 0.8.104 in Erinnerung.
Hätte ich bloß gelesen VOR dem Versuch, au 105 anzuraten.

kiu77, 2024-04-05:
@MetaChuh 
Vielleicht doch, denn ich hab noch welche übrig. Die müssen vor einem Update wohl erst aufgebraucht sein  😉

blueline13, 2024-04-05:
@kiu77 
Die Tendenz geht klar zur "Dritt" DTU oder gar zu einem ganzen Zoo, wenn ich so auf meine Schreibtisch gucke :grin: Also nicht alle Boards verschenken :wink:

hrolofs, 2024-04-05:
OTA update von 0.8.97 auf 0.8.105 => Dauer reboot (opendtufusion board)

hrolofs, 2024-04-06:
@MetaChuh 

> verspreche (zumindest ich dir) deinen avatar für 2 wochen nicht mehr zu erwähnen 😉

Bloss nicht, hab die Kommentare lieb gewonnen. 😉

hrolofs, 2024-04-06:
Gerade OTA Update von 0.8.97 auf 0.8.106 gibt wieder Dauer Reboot. Das taucht auf der Putty Console auf.`Nur ein Wechselrichter per NRF angeschlossen.
 Werde nachher ein folles flash Update versuchen.

``
Attempting to boot anyway...
entry 0x403c98d4
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x42049005
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
SHA-256 comparison failed:
Calculated: 3e3cea51273ddabf24003323ac0eb1044c5ebfaa93fe02d9876c9b3f8f8a585b
Expected: 3f6c8f5984f141d8075c85bb4d14f5d9211389c3e2f5992b3ae03ff85649cdf8
Attempting to boot anyway...
entry 0x403c98d4
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x42048c58
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
SHA-256 comparison failed:
Calculated: 3e3cea51273ddabf24003323ac0eb1044c5ebfaa93fe02d9876c9b3f8f8a585b
Expected: 3f6c8f5984f141d8075c85bb4d14f5d9211389c3e2f5992b3ae03ff85649cdf8
Attempting to boot anyway...
entry 0x403c98d4
``

hrolofs, 2024-04-06:
gerade versucht meine blinkyparts (ESP32 NRF) auf 0.8.106 zu heben, gleiches Ergebnis: Dauer-Reboot.

blinkiparts läuft jetzt auf 0.8.97, so hab ich das opendtufusion board für Tests frei.

frisch erased und geflashed mit 
ahoy_v0.8.36\ESP32-S3\bootloader.bin
ahoy_v0.8.36\ESP32-S3\partitions.bin
ahoy_v0.8.36\ESP32-S3\ota.bin
ahoy_v0.8.106\ESP32-S3\240405_ahoy_0.8.106_435aa17_opendtufusion.bin

wieder Dauerreboot. Manchmnal schaff ich es auf den AP und sehe den Wizard. Das wars dann wieder.

Über Putty bekomme ich auch nichts ausgelesen

kiu77, 2024-04-06:
Habe 0.8.106 gerade auf einen ESP32 geflachst, der frei ist, also an nichts angeschlossen. Das geht. Lediglich die Menüs sind etwas träge.
Jetzt kommt eine komplette DTU an die Reihe.

kiu77, 2024-04-06:
Sodele, auch die "richtige" DTU mit nRF24 und CMT2300A tut mit 0.8.106. Also success!

hrolofs, 2024-04-07:
Bei mir läuft die 0.8.106 jetzt auch. Problem war die Option "System Config/Reboot Ahoy at midnight". Wenn die aktiv ist begibt sich die DTU in einen Dauer Reboot. Ohne läuft sie bislang stabil.

hrolofs, 2024-04-08:
Ich hab mir jetzt bei meinen Nachbarn ein esp8266 Board geliehen. 
240405_ahoy_0.8.106_435aa17_esp8266-all.bin aufgespielt. DTU in mein WLAN gehangen, Display konfiguriert, kein Inverter. Läuft  wunderbar.

"Reboot Ahoy at midnight" aktiviert & SAVE/Reboot und das Teil geht in den Dauer Reboot.

kiu77, 2024-05-05:
0.8.115 ist responsiv und funktioniert gut bei mir.

lumapu, 2024-05-05:
die 0.8115 hat leider einen Fehler, der die DTU regelmäßig abstürzen lässt, daher die 0.8.116 verwenden

kiu77, 2024-05-05:
@lumapu 
Done, thanks a, lot!

closed 2024-08-16 by lumapu