lumapu/ahoy #1567: [Bug] 0.8.103 - Reboot at midnight verursacht bootloop bei ESP8266

Elarion245, 2024-04-06:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.103

### Github Hash

7ee1f99

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Interval: 3s, ansonsten standard

### Error description

Die 103 lief soweit eigentlich gut, bis auf die Tatsache, dass beim 8266 nur 1-2 Tage ohne reboot möglich sind. Ich habe daher mal versucht den Haken bei reboot at midnight zu setzen um tagsüber keine Unterbrechungen zu haben. War leider nichts...
Danach ein permanenter Bootloop. 
Nachdem der Haken raus ist, ist es wieder stabil. Interessanterweise sind aber meine Pinout-Settings beim Speichern verloren gegangen (vermutlich weil ich durch den Bootloop nur ca 5-10s hatte um Speichern zu drücken und der Vorgang vielleicht nicht perfekt beendet wurde).

hpd96, 2024-06-02:
me too
**esp32**-wroom32
0.8.**106**
Radio CMT

hpd96, 2024-06-02:
release 0.8.107 - 2024-04-08
duplicate of : ???
fix boot loop on reboot on midnight feature #1542, #1599, #1566, #1571

lumapu, 2024-06-09:
ist der issue damit noch nicht gefixt? dann entferne ich mal das "resolved dev" flag

stefan123t, 2024-10-31:
@Elarion245 @hpd96 könnt Ihr das Problem mit der aktuellen 0.8.152 Version noch bestätigen ?

hpd96, 2024-10-31:
> @Elarion245 @hpd96 könnt Ihr das Problem mit der aktuellen 0.8.152 Version noch bestätigen ?

ist gelöst mit 0.8.108.

hatte nur mit #1571 verlinkt nachdem ich den Fehler auch gefunden habe

stefan123t, 2024-11-02:
@Elarion245 kannst Du das Issue bitte als gelöst schließen, wenn Du das Problem mit der aktuellen Version 0.8.152 bzw. seit 0.8.108 nicht mehr gesehen hast ?