lumapu/ahoy #1568: How about increase CPU frequency?

DB1BMN-collection, 2024-04-06:
## Hardware
  * [X] ESP8266

Hi,
since it seems that more and more people report stability issues and reboots I propose to increase (doubling) CPU frequency.

Background: A long time ago I experimented with some WS2812 LEDs on a WeMos D1 mini and had to set CPU clock from 80 MHz to 160 MHz to make it work well.

Perhaps this would also be helpful to improve performance of Ahoy DTU?

kiu77, 2024-04-06:
Doubling? Really?
I can't believe that this act will bring more stability. I think that the MCU doesn't have tis headroom unless the MCU has been stripped down for the sake of low electric power consumption.
Would never do that. Maximum 10...20% more clock.

knickohr, 2024-04-06:
How about to change the hardware ?

DB1BMN-collection, 2024-04-06:
I don't see any problem.
The D1 mini is capcable of 160 MHz: 
[image]

As I said: Even for a simple program with the neopixel WS2812 I had to increase from 80 to 160 MHz because the timings were wrong and the LEDs did not react or showed the wrong colors.

I don't think that an other hardware will cure it because there also people with ESP32's who have trouble,
I think the scheduler is messed up or loosing some ticks.

kiu77, 2024-04-06:
In case the MCU at your ESP8266 board is set to 80MHz you may change that to 160MHz, yes. Because the Tensilica LX106 MCU used there has a max clock of 160MHz. So here you'll do nothing wrong - just the MCU uses more electrical power.
So I have to correct myself ;-)