lumapu/ahoy #1569: WiFi-Wizzard auf dem Mac, Umlautfehler im Button.

kiu77, 2024-04-06:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

≥ 0.8.97

### Github Hash

123

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

standard

### Debug Serial Log output

```shell
-
```

### Error description

Nur was Minimales:
Auf dem Mac gibt es einen Umlautfehler im Button "Prüfen" des Wizards.

[image]

Gubi2023, 2024-04-06:
Fehler ist beim Windows-PC genauso vorhanden...

kiu77, 2024-04-06:
Des is halt wegen des Inderned, gell?

lumapu, 2024-04-08:
> zur ergänzung:
> 
> ```diff
> --- AhoyDTU versucht eine Verindung mit deinem Netzwerk herzustellen
> +++ AhoyDTU versucht eine Verbindung mit Deinem Netzwerk herzustellen
> ```

ich musste jetzt 3x Mal lesen um den "groben" Fehler zu sehen 😊

lumapu, 2024-04-08:
der Umlautfehler ist mir noch unklar, aber wahrscheinlich kommt der von der durchgehenden Großschreibung.

Gubi2023, 2024-04-09:
> > zur ergänzung:
> > ```diff
> > --- AhoyDTU versucht eine Verindung mit deinem Netzwerk herzustellen
> > +++ AhoyDTU versucht eine Verbindung mit Deinem Netzwerk herzustellen
> > ```
> 
> ich musste jetzt 3x Mal lesen um den "groben" Fehler zu sehen 😊
Tippfehler ist auch in"  _Verbindung_ mit Deinem Netzwerk hergestellt" so drin....

lumapu, 2024-04-09:
@Gubi2023 verstehe nicht was du meinst. Ich habe zwei Fehler in dem Satz gesehen und behoben

Gubi2023, 2024-04-09:
@lumapu : das Wort "Verbindung" ist in der Maske "Verbindung herstellen" und "Verbindung hergestellt" vorhanden.
(Kann die Masken leider nicht reproduzieren)

lumapu, 2024-04-09:
danke für Aufklärung, jetzt hab ich's verstanden

Gubi2023, 2024-04-10:
V8.108: in dieser Maske scheint der Tippfehler noch drin zu sein.    
[image]

kiu77, 2024-04-12:
0.8.110 läuft den ganzen tag problemlos.

closed 2024-08-16 by lumapu