lumapu/ahoy #1570: Feature Request: Übertragung des unlimitierten P_AC-Wertes

Tjossul, 2024-04-06:
Hallo,

ist es möglich, den oben genannten Wert ebenfalls zu übermitteln?

Bsp.: Der WR hat gerade1000 W AC. Ich muß auf 600 W limitieren. Neu wird jetzt P_AC natürlich limitiert angezeigt - 600 W. Ich würde aber unabhängig davon auch gern wissen (ohne aktiv die Limitierung aufheben zu müssen), was unlimitiert zur Verfügung stünde.

Vielleicht gibt es den Wert a auch schon irgendwo, und ich habe ihn nur übersehen?!

Vielen Dank und viele Grüße.
T.

knickohr, 2024-04-06:
Das ist technisch nicht möglich weil der WR ja überhaupt nicht weiß was das Solarmudul gerade unlimitiert liefern könnte. Dazu müßte der WR ja die Leistung auch ausgeben und der Verbraucher es abnehmen.

Tjossul, 2024-04-06:
Oh, super - das hilft mir (bei meinen Steuerungen) tatsächlich sehr!

Danke, Mann!

Grüße!
T.

closed 2024-04-07 by Tjossul