lumapu/ahoy #1571: [Bug] 0.8.106 Dauer-Reboot nach Aktivierung "System Config/Reboot Ahoy at midnight"

hrolofs, 2024-04-07:
### Platform

ESP32

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.106

### Github Hash

435aa17

### Build & Flash Method

ESP Tools (flash)

### Setup

einen Inverter und MQTT

### Error description

Erfolgreiches Update auf 0.8.106 meiner blinkyparts DTU (esp32) und opendtufusion (esp32-S3). Wichtig war das "Reboot Ahoy at midnight" nickt aktiviert ist.

Beide DTUs zeigen das gleiche Verhalten.
Nach aktivieren von "Reboot Ahoy at midnight" und Save/Reboot gehen beide DTUs mit der 0.8.106 in einen Dauer-Reboot.

Mit der 0.8.97 laufen die DTUs auch problemlos mit aktiviertem "Reboot Ahoy at midnight".

lumapu, 2024-04-08:
das klingt nach einfach nachvollziehbarem Issue, ich werde mal versuchen das nachzustellen.

lumapu, 2024-04-08:
related to #1566

hrolofs, 2024-04-10:
Läuft beides mit "System Config/Reboot Ahoy at midnight"

240409_ahoy_0.8.108_58c5752_esp32-wroom32.bin
240409_ahoy_0.8.108_58c5752_opendtufusion.bin

Danke :-)

hrolofs, 2024-04-10:
Keine Reboots mehr mit aktiviertem "Reboot Ahoy at midnight"

closed 2024-04-10 by hrolofs