lumapu/ahoy #1572: Feature Request: Messwerte auf SD-Karte schreiben

MaDill, 2024-04-07:
Für die Zukunft würde ich mir wünschen, dass die Messwerte auf eine SD Karte geschrieben werden. Vielleicht wird pro Monat eine CSV Datei angelegt und nach jeder Messung die Werte da hinzugefügt. Nach X Monate wird die älteste Datei gelöscht.

Hintergrund ist, dass ich dann nicht noch mehr Hardware/Software (MQTT, Datenbank, Visum) ständig am laufen habe und pflegen muss.

Über eine einfache HTML Seite könnten dann die aktuellen Dateien auf der SD Karte aufgelistet und herunterladen werden.

Im Anschluss kann man dann Auswertungen mit Excel oder ähnlichem machen.

lumapu, 2024-04-08:
schaue bitte auch mal in #1199 vorbei, evtl. haben wir diesen Wunsch dort auch schon aufgenommen. Neu ist die Idee nicht, es gab schon mehrere Issues mit diesem Wunsch 😊

lumapu, 2024-04-08:
in #1063 haben wir schonmal darüber diskutiert - ich habe sogar schon Adapter hier, den hier:
https://www.az-delivery.de/products/copy-of-spi-reader-micro-speicherkartenmodul-fur-arduino

technics42, 2024-04-09:
Actually you don't need any adapter.
You can directly connect sd-cards to ESPs.
Adapter just is needed if you want to have possibility to connect without soldering.
I connected SD cards with soldering Micro-SF-adapters to ESP (not Ahoy-DTU), by this you can also change your micro-SD card.

lumapu, 2024-04-09:
what do you mean by connecting directly? My ESP does not have a slot for cards.

technics42, 2024-04-09:
You have to solder/wire your SD card to your ESP, just like the adapter. The adapter just passes through tge signals. E.g. take a look at: https://alexlubbock.com/micro-sd-adapter-esp8266-esp32

lumapu, 2024-04-09:
ok, I won't do that to a SD card 🙈. For me it's like destroying it.

technics42, 2024-04-09:
You don't have to solder your card directly,  just solder your micro-SD-adapter and just can exchange your micro-SD like usual.
I just wanted to clarify, that those adapters you were referring, are not needed at all.

lumapu, 2024-04-09:
I got your point, it's clever. I didn't thought about these adapters. It's the first time I know their intentional use 😊

stefan123t, 2024-09-10:
@lumapu hier haben wir das Thema SD Card mit einigen der zu beachtenden Besonderheiten in OpenDTU besprochen:

https://github.com/tbnobody/OpenDTU/pull/2125