lumapu/ahoy #1573: [Bug]

Edibo63, 2024-04-07:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.103

### Github Hash

?

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Update von 0.8.103 auf 0.8.106 kein NFR

### Error description

An 3 DTUs einen Update von 0.8.103 auf 0.8.106 gemacht.
Bei 2 war nachher kein NRF connected. Beim 3. schon.
Unterschiede der DTUs:
2 mit NRF24L01  - nach Update NRF nicht verbunden - zurück auf 0.8.103, alles funktioniert.
beim 3. mit  ebyte NRF24L01 läuft auch die Vers. 0.8.106
Besorge mir noch ein ebyte NRF24L01 und werde mal umstecken und testen.

lumapu, 2024-04-08:
ist das Pinout bei allen drei Setups identisch? Von der 103 zur 106 wurde was an SPI geändert, was durchaus mit dem Problem zu tun hat. Mir ist allerdings noch nicht klar wo genau der Fehler passiert.

Edibo63, 2024-04-09:
Hab erst nach Lesen von #1580 darauf geachtet.
Die 2 mit nicht connected NRF24 haben epaper.
Das mit  OLED display funtioniert.

lumapu, 2024-04-09:
ok, es gibt noch einen ePaper issue hier.
Aktuell fehlt leider noch ein hal (Hardware Access layer) um den SPI Bus mit den Funkmodulen zu teilen.

closed 2024-04-09 by Edibo63