lumapu/ahoy #1574: [Bug] Haken bei Zugriffsschutz lassen sich teilweise nicht löschten

lille2000, 2024-04-07:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.97

### Github Hash

?

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

de-all, Display, Inverter, kein MQTT 

### Error description

Ich kann die Haken bei „Zugriffsscutz“ nicht alle löschen. 
Man kann alle Haken löschen und nach dem Reboot ist es wie vorher eingestellt, bis auf: 
Wenn ich versuche, den letzten Haken bei „Index“ zu entfernen, werden nach den Reboot wieder alle angehakt bis auf „Daten“ und „Verlauf“.

Im json Export fehlen auch alle Einstellungen zum „Zugriffsschutz“

Peertje-62, 2024-04-07:
Hi,
the work around would be (for now) to set no admin password.
regards

lille2000, 2024-04-08:
Danke, dann habe ich das tatsächlich übersehen.

closed 2024-04-08 by lille2000