lumapu/ahoy #1575: Ahoy update 0.8.83 (rel) ro 0.8.106 (dev) - WiFi disconnect

nexulm, 2024-04-08:
After upgrading from 0.8.83 due to the HomeAssistant MQTT discovery issue to the latest development version 0.8.106 my AhoyDTU (based on ESP8266) isn't reachable anymore.
DTU was in AP mode afterwards, connection via SSID AHOY-DTU was pssoible and the Installation Wizard Welcome with setup for WiFi network occurs.
Stopping the wizard shows the old working NRF24 configuration and setup including MQTT but the WiFi "home" connection isn't working anymore. "Next" button to configure up the WiFi network again doesn't seem to wrok for me at all. 
Any suggestion w/o reflashing the device?

Maybe I'll give it a try later to reflash it via the AHOY-DTU network on 192.168.4.1 before doing it from scratch via USB-Flash. :-(

nexulm, 2024-04-08:
After downgrading to 0.8.84 via AHOY-DTU SSID to have the HA MQTT discovery fix available the DTU is working again w/o any further configuration steps.
...BUT...
Updating to the latest dev (here: 0.8.106) isn't working for me out of the box.

Gubi2023, 2024-04-08:
i also have always problems with connecting to home-Wlan by OTA. Maybe all three input boxes must be filled correctly, that transformation to home-Wlan works....

lumapu, 2024-04-08:
Thank you for reporting, I will look into that. The itention is to fill only the first and last field.

Gubi2023, 2024-04-13:
same behaviour at me with V0.8.110:
every morning the DTU only ir reachable over AHOY-DTU AP. After Power off/on it is visible in my home-Network and works!
(Es nervt aber, wenn man jeden morgen das Teil áus undd wieder einstecken muss) Bin ich der einzige mit dem Problem??

siehe auch https://discord.com/channels/984173303147155506/1029761098381017098/1228075094904143954

Jawada74, 2024-04-13:
> same behaviour at me with V0.8.110: every morning the DTU only is reachable over AHOY-DTU AP. After Power off/on it is visible in my home-Network and works! (Es nervt aber, wenn man jeden morgen das Teil áus und wieder einstecken muss) Bin ich der einzige mit dem Problem??
> 
> siehe auch https://discord.com/channels/984173303147155506/1029761098381017098/1228075094904143954

I have the same behavior even during daytime now. DTU looses wlan connect.  After  reboot via unpluggin WLAN  reapppears  without any problem.
( Gleiches Problem hier mit der V0.8.110 kam von einer V0.8.97  seltsam ist, das es jetzt auch tagsüber passiert.)

technics42, 2024-04-16:
Just fyi, I am using v0.8.101 and don't have this issue, although disabling wifi at night.

Jawada74, 2024-04-16:
I reverte from 08.110 to 08.97 and ervthing is fine again.

nexulm, 2024-04-19:
like to confirm that 0.8.111 fixes this issue for me. 0.8.84 was running before (= stable v0.8.83 + HA MW
DTU was directly accessible for me as expected after updating and is now running for mor than 2h.
---
You are using development version 0.8.111. In case of issues you may want to try the current stable release: 0.8.83

Gubi2023, 2024-04-20:
tja, bei mir hat die DTU auch mit 0.8.111 das heimische WLAN verloren, nur über AHOY-DTU-AP erreichbar. 2x in 24 Std.
Ursache unklar.

Jawada74, 2024-04-20:
Habe jetzt auch das update von 0.8.97  auf 0.8.111 gem,acht und bei mir ist die erste Nacht gut verlaufen. WLAN-Verbindung war morgens noch da. Ich beobachte weiter...

closed 2024-04-22 by nexulm