lumapu/ahoy #1576: [Bug] einzelnen Wechselrichter aus der Berechnung ausschließen

PelzViech, 2024-04-08:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a

### Build & Flash Method

was already installed

### Setup

normales Setup

### Error description

Früher gab es den Menüpunkt dass man einen einzelnen Wechselrichter aus der Berechnung ausschließen kann. Gibt es den Menüpunkt nicht mehr oder finde ich diesen einfach nur nicht?

PelzViech, 2024-04-08:
Ich meine mich zu erinnern, dass es bei den Invertern einen Menüpunkt gab, wo man einen "nimm diesen Inverter aus der Berechnung raus" Button gab.

knickohr, 2024-04-08:
Es gab mal so eine Option. Die war aber nur MQTT-relevant :
https://github.com/lumapu/ahoy/blob/development03/src/CHANGES.md#0871---2024-02-03

lumapu, 2024-04-08:
diese Berechnung kann man selbst in der Hausautomatisierung erledigen, ich sehe aktuell keinen Grund dies in Ahoy zu machen. Man kann einzelen Wechselrichter deaktivieren, d.h. diese werden dann nicht ausgelesen.

PelzViech, 2024-04-09:
OK, danke. Vielleicht habe ich mir diese Einstellungen bei den Invertern auch nur eingebildet. Dachte es wäre recht spannend gewisse Live Daten sehen zu können, die dann aber nicht in die Berechnung einfließen.