lumapu/ahoy #1578: ESP-Time läuft weiter ohne Strom, ist das normal?

Loetnase, 2024-04-09:
## Hardware
  * [ ] ESP8266
  * [x ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x ] external antenna (SMA)

### Power Stabilization:
* [x ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version:0.8.106__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Wenn ich zum Kaltstart den USB Stecker ziehe läuft die ESP-Time munter weiter.
Die Uptime wird aber nicht mehr aktuallisiert.
Erst mir refresh der Seite verschwindet mit der nichterreichbarkeit der Seite (Zeitüberschreitung)
Ist das so normal, der ESP ist ja tot/stromlos. 

[image]

Peertje-62, 2024-04-09:
Hi,
refresh your browser, CTRL F5
regards

kiu77, 2024-04-09:
> der ESP ist ja tot/stromlos.

Peertje-62, 2024-04-09:
Hi, https://de.wikipedia.org/wiki/Browser-Cache

Browser-Cache ist ein [Puffer-Speicher] des [Webbrowsers], in dem bereits abgerufene Ressourcen des Benutzers (lokal) als Kopie aufbewahrt werden. Wird eine Ressource später erneut benötigt, ist sie aus dem Cache schneller abrufbar, als wenn sie erneut aus dem untergeladen werden müsste.
regards

Loetnase, 2024-04-09:
Aber dann ist aber die ESP-Time eigentlich Browser-Time, oder wie wird der Bezug zum ESP hergestellt? 
Wird da einmal die Zeit vom ESP geholt und die läuft dann im Browser weiter?

kiu77, 2024-04-09:
> Wird da einmal die Zeit vom ESP geholt und die läuft dann im Browser weiter?

Ich vermute, so ist es. Wissen würde ich es nur, wenn ich im Code danach suchen würde.

lumapu, 2024-04-09:
alle 10s wird die Zeit versucht vom ESP zu holen, den Rest läuft sie selbst weiter. Hintergrund ist die Reduktion der Anfragen

Loetnase, 2024-04-09:
Okay, dann habe ich das alles soweit verstanden, 
bedanke mich bei allen für die Antworten 
und mach das issue artig wieder zu

closed 2024-04-09 by Loetnase